Phil1988/FreeDi #173: [Bug] Display dont go into *Print screen

cptcl, 2025-02-28:
**Describe the bug**
After starting a print, the display does not change into the printing screen. 
It just stays in the same screen like when u startet up the printer. 
And I can't get into the print screen . 

I remembered this happened also on stock qidi firmware sometimes but wheen u push a few buttons on the display it will change to the printing screen at some point. 

But here it won't. 

**Expected behavior**
After starting a print display should go into printing screen. 

**To Reproduce**
Does not happen every time but I think it happens if u try to print a file with a huge file size.. 

See pictures 
This happened twice for this exact file. 

But worked as usual for all other prints before. 

**Screenshots or picture**

[image]
[image]
[image]
[image]

**Console output
None. Like nothing unusual

**Base (please complete the following information):**
 - Printer model: x-max 3 rev2
 - latest FreeDi 
 - no modification on Mainboard, or tool head board.

pmbroth, 2025-02-28:
just curious if that happens if you Orcaslicer with the same file

cptcl, 2025-02-28:
> just curious if that happens if you Orcaslicer with the same file

This is... Orcaslicer.

Phil1988, 2025-02-28:
You showed a STL file but we all know that a printer can't execute that.

Please show us the gcode and a debug output:
Start the debug console as described on the wiki.
https://github.com/Phil1988/FreeDi/wiki/Bugs!
Then upload/start the print via Orcaslicer.
Show us the debug output.

cptcl, 2025-02-28:
> You showed a STL file but we all know that a printer can't execute that.
> 
> Please show us the gcode and a debug output: Start the debug console as described on the wiki. [How to report bugs](https://github.com/Phil1988/FreeDi/wiki/Bugs!) Then upload/start the print via Orcaslicer. Show us the debug output.
> 
> Additionally: We need to know the orca slicer version. Nightlies and Dev versions are known for bugs and causes multiple issues.

[image]

**Can i input the debug command while its printing?** if not i will finish the print first.

Also, wanted to download the gcode file from the printer but it wont let me. so i will just save it as file and upload it

[13x13 Multiboard Corner Tile_PLA+_13h57m.zip](https://github.com/user-attachments/files/19031946/13x13.Multiboard.Corner.Tile_PLA%2B_13h57m.zip)

Phil1988, 2025-02-28:
If it's currently printing, just let it finish.
No hurry.

But when finishes please give us the debug output.
This really help us understanding the cause

rosenrot00, 2025-02-28:
Could you run cd ~ && rm -rf FreeDi && git clone --filter=blob:none --sparse https://github.com/Phil1988/FreeDi ~/FreeDi && cd ~/FreeDi && ./install.sh

Let me know if it works.

00stag, 2025-02-28:
The gcode file is 122 MB!? Isn't that a bit too much?

cptcl, 2025-02-28:
> The gcode file is 122 MB!? Isn't that a bit too much?

So this file is directly from https://www.multiboard.io/ (the 13x13 Multiboard Corner Tile) and yes, orca tells me that it have many triangles and i should maybe simplify it.. but thats not the point.
As is said i had the sampe thing with stock qidi firmware (there it took like 3 minutes and if u touch a few things on the display it would eventually switch) and so my guess is, that if the file is too large the "print screen" maybe cant handle it and so it cant switch to it.

i mean i can controll my printer over web gui.. this is just a thing i noticed yesterday (But as i said, sometimes this happened with stock firmware too, so i didnt mind it) i started the print (but it failed cause the edges where lifting a bit, so i stoped it.) Today i resliced it with some more material on the outside and started the new print and noticed this again.

@rosenrot00 Can i run this command while its printing? my guess is no. 
@Phil1988 
Since this is a ~14h print i will report back tomorrow. Will run the command after finish and post it here.
Do you need me to do something else more?

rosenrot00, 2025-02-28:
You could but never mess with a 14h print. Try it once it is done printing.

breathless19, 2025-03-01:
What you are describing is definitely a bug in the stock firmware, and I have experienced it many times myself with all of my X-Max 3's.   It's possibly related (at least for me) to the size of the file.   I don't remember this happening on small files, though it may have and I just didn't take note of it.   To make the file pop up on the printer display on the stock firmware, you just tap into the gcode files menu as if you were going to go to print a file that was already uploaded to the printer and it will instantly pop up as it should and show the currently printing file as normal. 

I believe this has happened with the FreeDi firmware too, but if it did, I probably just wrote it off and didn't think twice about it.  If it happens again, I'll be sure to give the debug info in this thread.

cptcl, 2025-03-01:
> What you are describing is definitely a bug in the stock firmware, and I have experienced it many times myself with all of my X-Max 3's. It's possibly related (at least for me) to the size of the [...]
@breathless19 
Yes.. as i already said.. on stock firmware, this is the way to get the display switch to printscreen. And it happens there from time to time. Sometimes just with a delay and it switches to printscreen after a certain time. But it does not work like this with the FreeDi firmware. 
Only thing i did not try, since i want my print to finish, is ..
On Display "Go to gcode files, select the file it is already printing and press "print" button while its printing"

Also,.. yes. Printer prints normaly and u can just access and control the print over webgui. 
Its not that big of a deal but i just want to mention it and maybe helps other ppl who got the same "problem".

Not a very big deal indeed for me, cause i have a KlipperScreen over wifi running too. so i could just use this too to control the print without PC. 
(Yes, i have this Klipperscreen the whole time, no, it does not interfer with anything, I only have this cause my printer sits around 25cm above the ground and i dont want to bend over everytime to read the display, so i mounted a Klipperscreen on a bit higher level just for comfort. 
I even dont like klipperscreen that much estaticly, but the comfort too not have to knee or bend over) 

@Phil1988 i did
``` sudo systemctl stop FreeDi.service ```
and
``` cd ~/FreeDi/FreeDiLCD && ~/klippy-env/bin/python3 start.py -debug ```
Result is
```  
Connecting to Moonraker at ws://127.0.0.1:80/websocket
Using selector: EpollSelector
TFT baudrate detection...Websocket connected
Moonraker connection opened at ws://127.0.0.1:80/websocket
Detected baudrate: 921600
Display response: comok 1,1281-0,TJC8048X250_011R,165,10201,CEB3340100FBAE72,33554432-0
G-Code file path is: ~/printer_data/gcodes
Display firmware: v1.40
Printer model: x-max3
Printer volume: 328x326x322
Klipper version: v0.12.0-289-g08a1c9f12-dirty
Software version: v1.40 stable
FreeDiLCD start
FreeDi sw-version sent to screen!
self.lcd.data_update()
printjob.name.txt="Multiboard/13x13 Multiboard Corner Tile_PLA+_13h57m.gcode"
data.klipper_state != self.printer.klipper_state
klippy ready
Printer state: complete
Printjob completed
extract_printjob_values: self.printer.file_name: Multiboard/13x13 Multiboard Corner Tile_PLA+_13h57m.gcode
Starting new HTTP connection (1): 127.0.0.1:80
http://127.0.0.1:80 "GET /server/files/metadata?filename=Multiboard/13x13%20Multiboard%20Corner%20Tile_PLA%2B_13h57m.gcode HTTP/11" 200 None
Filament Type: PLA+
Estimated Time: 13h 57m
Filament Used (mm): 62746.74
Filament Used (cm): 62.74 m
Filament Used (g): 181.1 g
22 ms: Loading GCODE
show_thumbnail: self.printer.file_name: Multiboard/13x13 Multiboard Corner Tile_PLA+_13h57m.gcode
file: Multiboard/13x13%20Multiboard%20Corner%20Tile_PLA%2B_13h57m.gcode
http://127.0.0.1:80 "GET /server/files/thumbnails?filename=Multiboard/13x13%20Multiboard%20Corner%20Tile_PLA%2B_13h57m.gcode HTTP/11" 200 None
http://127.0.0.1:80 "GET /server/files/gcodes/Multiboard/.thumbs/13x13%20Multiboard%20Corner%20Tile_PLA+_13h57m-300x300.png HTTP/11" 200 19083
54 ms: Thumbnail Processing
Sending thumbnail to screen
Write thumbnail to LCD
Unexpected header received: 0x24
Unexpected header received: 0xff
Unexpected header received: 0xff
Unexpected header received: 0xff
Unexpected header received: 0x24
Unexpected header received: 0xff
Unexpected header received: 0xff
Unexpected header received: 0xff
Unexpected header received: 0x24
Unexpected header received: 0xff
Unexpected header received: 0xff
Unexpected header received: 0xff
Unexpected header received: 0x24
Unexpected header received: 0xff
Unexpected header received: 0xff
Unexpected header received: 0xff
Write thumbnail to LCD
Unexpected header received: 0x24
Unexpected header received: 0xff
Unexpected header received: 0xff
Unexpected header received: 0xff
Unexpected header received: 0x2e
Unexpected header received: 0xff
Unexpected header received: 0xff
Unexpected header received: 0x24
Unexpected header received: 0xff
Unexpected header received: 0xff
Unexpected header received: 0xff
Unexpected header received: 0x24
Unexpected header received: 0xff
Unexpected header received: 0xff
Unexpected header received: 0xff
Write thumbnail to LCD
...
```
Following part repeats it self like ... while i wrote this whole text.. it repeas and still repeating...  feels like forever. will let it run and see if there is something.. new or if something changes, but i feel this could take a while
```
Write thumbnail to LCD
Unexpected header received: 0x24
Unexpected header received: 0xff
Unexpected header received: 0xff
Unexpected header received: 0xff
Unexpected header received: 0x64
Unexpected header received: 0xff
Unexpected header received: 0xff
Unexpected header received: 0xff
Unexpected header received: 0x24
Unexpected header received: 0xff
Unexpected header received: 0xff
Unexpected header received: 0xff
Unexpected header received: 0x24
Unexpected header received: 0xff
Unexpected header received: 0xff
```

PS: while i was in preview of my post here, the log showed me

[image]

and the printer display is now stuck on bootscreen for like +5 minutes. it says (connecting to klipper...)
Have to turn off and turn back on (Powercycle with a wait of 30 seconds)

rosenrot00, 2025-03-01:
Did you run the command and checked if the print screen comes up?

cptcl, 2025-03-01:
> Did you run the command and checked if the print screen comes up?

How to enter a command when connection is lost? so no.
i simply could not run 
``` cd ~ && rm -rf FreeDi && git clone --filter=blob:none --sparse https://github.com/Phil1988/FreeDi ~/FreeDi && cd ~/FreeDi && ./install.sh ```
while printer was running. I had to power cycle it

I just run Phil1988´s command for debug output

Yea, my guess that klipper hung in the cycle of this debug error and thats why connection lost.

PS: I just edited the post above.. printer display showed me the "Print finished" screen. and also debug error shows, it just want to display the thumbnail and the print screen, but cant. Maybe it realy has to do with the size of an gcode file. idk. 
Lets see what Phil1988 has in mind

rosenrot00, 2025-03-01:
I fixed the underyling issue which you posted here originally. This is why I asked to run this command to push the bugfix on your printer.

The error you are describing here is something I'm aware of and has nothing to do with the original bug and is an additional bug that I'm working on.

Filesize of the gcode doesn't matter until it crashes klipper itself.

cptcl, 2025-03-01:
> I fixed the underyling issue which you posted here originally. This is why I asked to run this command to push the bugfix on your printer.
>
> The error you are describing here is something I'm aware [...]

Ok.. now i started another print.
Display does NOT show "print screen" again.
This time i tested on display, while the printbed was heating up, go into gcode files, select the gcode i already start to print, and press the print button. It worked and the display goes into the "Print Screen".
Now i will cancel the print (in the heat up phase) and will run your command.
```
cd ~ && rm -rf FreeDi && git clone --filter=blob:none --sparse https://github.com/Phil1988/FreeDi ~/FreeDi && cd ~/FreeDi && ./install.sh
```
After i will restart the printer and restart the print i wanted to do.
Will edit this post when ive done it

rosenrot00, 2025-03-01:
Thanks :)

No it won't it will just replace the FreeDi files.

cptcl, 2025-03-01:
> Thanks :)
> 
> No it won't, it will just replace the FreeDi files. It shouldn't even require a restart.
> 
> If this fixes the bug your printer will show the print screen and everything will work for you as expected.

Printer rebooted by itself.
Homing all axes
Resliced and print start over orcaslicer

Display shows printscreen right away with thumbnail as it should

Phil1988, 2025-03-01:
Can you please check it with the exact same file that didnt work before?
There are multiple sources that could cause errors like this so we need to drop down the variables as best as possible.

If the error with "13x13 Multiboard Corner Tile_PLA+_13h57m.gcode" persists and it doesnt forward you to the print screen, then please:
1. abort/cancel the print
2. stop freedi
    `sudo systemctl stop FreeDi.service`
3. Run the debug command:
    `cd ~/FreeDi/FreeDiLCD && ~/klippy-env/bin/python3 start.py -debug`
4. Start "13x13 Multiboard Corner Tile_PLA+_13h57m.gcode" again
5. See and report back what the console reports

pmbroth, 2025-03-01:
> > just curious if that happens if you Orcaslicer with the same file
> 
> This is... Orcaslicer.
Sorry, was looking in another thread, also Orcaslicer 2.3 beta is out.

Phil1988, 2025-03-01:
Should be fixed now.

Please update to the latest version and check if it still persists:
https://github.com/Phil1988/FreeDi/releases/tag/v1.41

I'm closing this to get it off my radar, but please give me feedback in every case :)

closed 2025-03-01 by Phil1988

cptcl, 2025-03-02:
@Phil1988 
Updated to 1.41

Starting the print from the display is getting me into the "print screen". Tryed it 3 times.
Starting a new sliced file from Orcaslicer to print is getting me also into the "print screen". Tryed 2 times.

Seems to be fixed, or it was just a hickup like in the stock firmware. if fixed,.. thx for fast fixing.

Phil1988, 2025-03-02:
@cptcl Thanks for the feedback.
This is how it should be.

I'm sure it was no hickup, but fixed by @rosenrot00 ❤