Phil1988/FreeDi #174: FreeDi v1.41 released

Phil1988, 2025-03-01:
### v1.41 has just been released!

This is a rather small update to wipe out some minor bugs and add a few small features.

## Instructions to Update:

**Click update** 😉
[image]

### For the 3 macros ( PRINT_END, CANCEL_PRINT and M107):
Go to the [config_section](https://github.com/Phil1988/FreeDi/tree/master/config_section) and get the matching macro for your printer (copy the macro and replace yours)

### Changelog and more information can be found on the release page:
https://github.com/Phil1988/FreeDi/releases/tag/v1.41

<br>

pmbroth, 2025-03-02:
@Phil1988   Thank you Phil for all of the hard work.  Looking forward to the discord session tomorrow.  Also when should we see the update pop up?

Thank you again,  Brian

breathless19, 2025-03-02:
Just a heads up to anyone else that notices that no amount of refreshing will make the new version show, I needed to run: 

`cd ~ && rm -rf FreeDi && git clone --filter=blob:none --sparse https://github.com/Phil1988/FreeDi ~/FreeDi && cd ~/FreeDi && ./install.sh`

rmx-it, 2025-03-02:
Hi,
I - since I beta tested some intermediate changes - get a different symptom:

[image]

Is there an intended solution besides `git pull; ./install.sh` if I want to test the new version ... ?

schichtdienst38, 2025-03-02:
> This is great, thanks!
> 
> Just a heads up to anyone else that notices that no amount of refreshing will make the new version show, I needed to run:
> 
> `cd ~ && rm -rf FreeDi && git clone --filter=blob:none --sparse https://github.com/Phil1988/FreeDi ~/FreeDi && cd ~/FreeDi && ./install.sh`

So there are others who are missing showing the new version within the automated update option?
Is your code the only way at the moment to get the version 1.41?

Phil1988, 2025-03-02:
Before cross posting blows up more, can anyone with this problem share the ouput of this?:

```
cd ~/FreeDi
git remote -v
git status
git fetch origin
git status
```

I think that should be enough to ask git to fetch the new version.
Then Update Manager should do.

Phil1988, 2025-03-02:
> So there are others who are missing showing the new version within the automated update option? Is your code the only way at the moment to get the version 1.41?

There is likely some git sync problem.
Im not sure about the cause yet because on my dev machine it worked - but ofc my printer is not on the same page as yours as I am constantly developing on it.

The command from breathless19 will clone the repository and install everything.
Its a workaround, but I would rather like to understand what is causing this issue for you.
Maybe the installation of 1.40 stoped the git sync.

rmx-it, 2025-03-02:
mks@mkspi:~/FreeDi$ git status
On branch master
You are in a sparse checkout with 28% of tracked files present.

nothing to commit, working tree clean
mks@mkspi:~/FreeDi$ git fetch origin
From https://github.com/Phil1988/FreeDi
 * branch            master     -> FETCH_HEAD
mks@mkspi:~/FreeDi$ git status
On branch master
You are in a sparse checkout with 28% of tracked files present.

nothing to commit, working tree clean

rmx-it, 2025-03-02:
BTW:
After doing a `git pull` manually I get the INVALID on refresh as shown above...

rmx-it, 2025-03-02:
Maybe - in my case - some symptoms are due to the installation/upgrade problem I was involved in.
See https://github.com/Phil1988/FreeDi/issues/146#issuecomment-2685241901

There I was advised to get a fresh FreeDi checkout and some `git pull`.

Now I can either switch to this FreeDi or to the original FreeDi checkout from the 1.4 image.

Here is the info from the original FreeDi checkout:

mks@mkspi:~/FreeDi$ git remote -v
git branch -vv
origin	https://github.com/Phil1988/FreeDi (fetch) [blob:none]
origin	https://github.com/Phil1988/FreeDi (push)
* master 64e22fd Update macros.cfg

The refresh shows no new version. But why is version 1.30 shown?

[image]

breathless19, 2025-03-02:
Here's what I get when I run the command on my machine that I haven't forced the upgrade to 1.41 yet: 

[image]

Phil1988, 2025-03-03:
Thanks everyone!

I think I found the reason for this:
My project on github is relatively large.
Since you only need files for your printer, it only downloads/updates the files you need.
Git seems to have a problem with this and then no longer synchronizes properly.
(It basically breaks Update Manager functionality).

This is why the update is currently not displayed in Mainsail.

There are several ways to solve this:
- Automatically check the status on github periodically (via cron)
- Check the status on github every time the printer/freedi is started
- Fundamentally change this repository on github (split into several) so that it works normally via git (but this may require a quite bit of work and some structural changes)
- ... and other ideas

I have not yet made a decision on this.
Since there are no critical errors, I ask everyone to be patient, because there are solutions, but I'm still undecided about the way forward.

For all the impatient ones ;)
You can update via the terminal:
```
cd ~ && rm -rf FreeDi && git clone --filter=blob:none --sparse https://github.com/Phil1988/FreeDi ~/FreeDi && cd ~/FreeDi && ./install.sh
```

Everyone seeing v1.30 even if they have updated to 1.40 previously:
A bug in v1.40 made KAF not executable and I deactivated this version.
Everyone on this is "seeing" v1.30 (fallback) as you are actually on v1.40 with this minor bug.

schichtdienst38, 2025-03-03:
Thank you for your options and thoughts. For me because of impatience;-) it was the terminal way doing very fine. I am now on 1.41

But in this context I tried to find out which is the actual better say last (in)-official version number of toolhead rp2040? Maybe it is my fault or I have a wrong view of it...but your update version log is saying something about new toolhead firmware, but after update both (MCU and Rp2040) say something about v0.12.0.289 as before. 

Thank you for your great work of course

breathless19, 2025-03-04:
Yes, please let us know if there is an easy way to see if the rp2040 toolhead is on the latest version or not.   I did notice my machine go through a firmware update on the LCD after the update, but I'd like to confirm the version number too :)

Phil1988, 2025-03-04:
>  your update version log is saying something about new toolhead firmware

It doesnt affect you, as the previous one is working flawless for you.
It's "the same" klipper firmware with a different clock divider to support lower performing toolheads.
So also the other firmware would show v0.12.0.289.

So to both of you @breathless19 @schichtdienst38 :
If you dont experience a disconnected toolhead, you are good and I dont recommend changing anything.
If you experience that, you need to flash the toolhead via KAF.

schichtdienst38, 2025-03-04:
Now I have understood the contexts and can understand the version log better for myself. Thank you @Phil1988 for such a great work

TopheC, 2025-03-09:
I was in v1.30 on my Xmax3 screen, fallback as you explained, but in v1.40 in fact, since early February.
I make some production since, everyday, and I experienced some small bugs like erroneous print duration reported on screen, and sometimes, screen stay on 98-99% job while print was entirely finished and OK on Mainsail....
Annoying but not critical ;)

I was waiting for seeing notification of upgrade on Machine menu from Mainsail, but as yesterday nothing but OS updates was notified... I do all those upgrades flawlessly though.
Yesterday, on my maintenance schedule, I decided to upgrade to v1.41 with the "impatient" way from SSH terminal, as you suggested @Phil1988 :)

Everything went OK on console, some questions/passwords to enter, that's all.
BUT, on the end of the upgrade, it is recommended to restart/reboot, as usual I supposed, so I do... 
AND, as I was at some distance from the printer, I finally saw on the screen that the firmware download was not finished, and then stuck at ~25%, because of my  early reboot of the host.
I frankly think i was bricking my printer....

I turn Off then printer, then On, the host was OK on SSH terminal, but then screen stuck on same ~25% downloading.
Then I redo all the "impatient" procedure, and then the download restarted on screen, till the end, and the normal restart Klipper/Firmware message appears. 
Pfffiuuu :)

Since, I was able to verify that v1.41 is installed, up and running !
- 10 prints without problems at all,
- no more erroneous jobs duration, on screen,
- every print was finishing at 100% OK, on screen.

As you mention it in the Installation Guide (https://github.com/Phil1988/FreeDi/wiki/Updating-FreeDi) the screen needs time to upgrade, but I think I zap it...  
Can I suggest to put a _Disclaimer_ and/or _Warning_ about not to be so "Impatient" when upgrading, the screen may take is time to download/install is firmware, slightly more than ~4mn for my case, and one must wait until the end of the process, the "restart" screen.

An sorry for my poor "frenglish" and the novel to read, I wanted to share and be exhaustive....
So, once again, what a good you made @Phil1988 !!!

TopheC, 2025-03-09:
> The update of the screen firmware took "about" 231,943s - 237,465s based on 40 upload measurements on my side - so I still claim its less then 4min 

You're certainly right, as I not took precise measures... 

>  I hope (pray 🙏) to not see any freezes of your systems in the future while updating the screen.

I will test it as soon as possible, at future upgrade.

>  Shutting a system down while doing an update will cause problems 😄

As old Sysops I know that from a long time, my bad was of my remote operation (printer is in other room)...
I felt so stupid when I saw the printer screen with the progress stuck at 25%....
Mea culpa ! 

And thank you for the Warning added to the guide 🙏

Phil1988, 2025-03-09:
I think it wasnt your fault or only "partially".
Even if you interrupted the firmware update, my goal is to make FreeDi robust enough to make it recover the display.
This happened successfully, but it may have run into the "freeze" while updating it (that is hopefully fixed now).

breathless19, 2025-03-10:
>     * The print job duration is calculated by klipper/moonraker and I just take those values. If it now shows them correctly its a nice coincidence, but its a globaly known thing with klipper calculating them imprecise :(
> 

I was going to add a feature request or bug fix request for this very thing.   It seems to me that the best way to do this instead of relying on Klippers calculation is to just do a simple math difference between the files stated completion time, and the time already past.   So the remaining time would be a simple countdown difference between the two and not a realtime calculated type of thing...

Phil1988, 2025-03-11:
@breathless19 I completely understand your intention to clarify things and I appreciate it. However, I kindly ask you not to mix different topics.  

This thread should not become a collection of unresolved issues, questions or feature requests.  

If you have something specifically related to the 1.41 update, this is the right place to discuss it.  

For everything else, please post it in the appropriate issue or discussion section based on the topic. 🙂
Thanks mate!

Phil1988, 2025-03-11:
Hey everyone!

I just released v1.42 to hopefully solved the Update Manager issue.
I also did a few changes to improve screen and MCU update reliability.
Please head over and [update to v1.42](https://github.com/Phil1988/FreeDi/issues/187).

We will see if it works with the next (v1.43+) update 😉

closed 2025-03-11 by Phil1988