Phil1988/FreeDi #185: [Help]

jocoka73, 2025-03-10:
Hello
I have a little problem, unfortunately I did something wrong and now my printer won't start.
Unfortunately I don't speak English well, I wrote this with a translator.
I'm attaching a picture.
Thank you very much

[image]

colin-small, 2025-03-10:
@jocoka73 This is not enough information to help you. Please provide a detailed description of your issue and information about your hardware and software implementation.

At this point all anyone can see is that you have an X Max 3 and Klipper won't start.

jocoka73, 2025-03-10:
Greetings
The following error message is on the printer's network interface.

Klipper reports: ERROR

mcu 'mcu': Unable to connect
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Error configuring printer

Phil1988, 2025-03-11:
@jocoka73 
Unfortunately, this request is far from ideal. It lacks a meaningful title and many relevant details. 
A screenshot of Mainsail and the Machine section would be really helpful here.   

Since you struggle with english, I’ll make an exception and help you anyway.  
However, in the future, I kindly ask you to put in more effort (e.g.by using a translator and collecting relevant information) to save everyone's time. 🙂  

I just used my crystal ball and it told me this might solve your problem ;)

```
cd ~/FreeDi/mainboard_and_toolhead_firmwares/ sudo service klipper start
sleep 3 echo "FIRMWARE_RESTART" | sudo tee /home/mks/printer_data/comms/klippy.serial sudo service klipper stop
./hid-flash v0.12.0-289/X_4.bin ttyS0
```

Please give this a try in the terminal (enter this complete command and confirm with the password makerbase).

If it doesn't work, I guess I'll have to send my crystal ball in for repairs...
In the meantime, please provide me with more details so I can help you better. 😉

jocoka73, 2025-03-11:
Hi
I just uploaded printer.cfg that I found on your site, so I got this error.

[image]

jocoka73, 2025-03-11:
Hello
I did what you described, two pictures were received, first the blue one, then the red one.
I am attaching another picture where I was looking for the head adjustment.
Thank you very much.

[image]

[image]

[image]

jocoka73, 2025-03-11:
Hello
I am attaching the picture.

Thank you very much.

[image]

Phil1988, 2025-03-11:
This is the 3rd time now I am asking for that 😔:

> And have you dont [that](https://github.com/Phil1988/FreeDi/issues/185#issuecomment-2712157687) ?

jocoka73, 2025-03-11:
hello
sorry but I don't understand this right now.
I'm not very familiar with the linux system.
I'm just trying to find a solution to the problem.  🤕

pmbroth, 2025-03-11:
I know english is not your first language.  Phil is asking did you try this below:

cd ~/FreeDi/mainboard_and_toolhead_firmwares/
sudo service klipper start
sleep 3 echo "FIRMWARE_RESTART" | sudo tee /home/mks/printer_data/comms/klippy.serial
sudo service klipper stop
./hid-flash v0.12.0-289/X_4.bin ttyS0

jocoka73, 2025-03-11:
Hi
Sorry, I just realized.
I'm attaching a picture.
Thank you very much.

[image]

jocoka73, 2025-03-11:
Hi
Still an error. 🤕

[image]

jocoka73, 2025-03-11:
Hi
I don't know if this is interesting, but the tool head identifier in the klipper.log file is incorrect.

[image]

[image]

jocoka73, 2025-03-11:
Hello
I found a test on this website to check if the head is in boot mode, but unfortunately it says it is not.
Maybe the electronics in the head are broken. 🤕

jocoka73, 2025-03-11:
Hello
unfortunately no change I entered the identified in the printer.cfg file but no result.

[image]
[image]

jocoka73, 2025-03-11:
Hello
Is this okay?

[image]

jocoka73, 2025-03-11:
Hello
I did it

[image]

Phil1988, 2025-03-11:
Download the X_4.bin from here:
https://github.com/Phil1988/FreeDi/blob/master/mainboard_and_toolhead_firmwares/v0.12.0-289/X_4.bin

Like this:
[image]

Then do this:
https://youtu.be/_ZmSTJBWUus?feature=shared&t=1036

jocoka73, 2025-03-11:
Hello
I am very happy that my head finally has life.

[image]

Phil1988, 2025-03-11:
So what did solve it for you?

Did you only use the terminal command?
Or did you use the sd card?

jocoka73, 2025-03-11:
Inserting the SD card was fine.
Now how do I proceed?

Phil1988, 2025-03-11:
Have you ever had problems with your tool head before?

First thing I would to is to power the printer off and power it on again.
Is it still persistent?

If yes I tend to say it might be a hardware failure.

rosenrot00, 2025-03-11:
@jocoka73 open the toolhead and send us a picture of the electronics. I have an idea what it could be.

jocoka73, 2025-03-12:
Hello
Sorry I'm only writing now but unfortunately I've been sick for 3 days, I was able to pull myself together today.
So the thing started when I bought a Chinese printhead.
It printed perfectly, but for some reason it accidentally found the print on the table and it bent.
Then I took the head apart and took out the nozzle, adjusted it and put it back.
The machine started but the temperature at the head was 0C, no matter how much I raised the temperature nothing happened.
Then I checked the mini bead thermistor and its value was 0.Ohm. This indicated a short circuit.
So that's how the thing started.
I might be thinking that the IC connected to it is broken, so it can't provide data, but of course this is just a thought.

[image]
[image]
[image]

Phil1988, 2025-03-12:
Your thermocouple is defective.
Please replace it with a new one.

I unplugged my thermocouple and got the same error as you:
[image]

When I measure the thermocouple, I can read 1.7-1.9 Ohm at ambient temp.
[image]

Summery:
MCU flash was not possible via serial (sd worked)
Toolhead firmware was wrong
ID of the toolhead was wrong
Thermocouple is broken

I am closing it now as all software and/or FreeDi related issues have been solved.

When you replaced your thermocouple it should work properly 😉

closed 2025-03-12 by Phil1988

jocoka73, 2025-03-12:
Hello
Thank you so much for your help so far.
You are a fantastic helpful person.
Thanks again.

Phil1988, 2025-03-13:
Thanks for your kind feedback @jocoka73 
I hope you can solve the hardware issues soon to completely enjoy FreeDi 😊