Phil1988/FreeDi #186: [Bug] "Section 'freedi' is not a valid config section" after some days of power off

rmx-it, 2025-03-11:
**Describe the bug**
A clear and concise description of what the bug is.
After some days beeing powered off after power on I get
the following error from klipper:

"Section 'freedi' is not a valid config section"

Usually this seems to indicate a syntax error in the printer.cfg
But in this case I'm quite confident that at least during the last few times of power offs/reboots
no changes to the printer.cfg occured.
A diff to the last changes to the printer.cfg only shows differences in the dynamic sections for the bed mesh leveling section and the z_offset....

Any clue how this can happen?

Config file is attached.

**Expected behavior**
Normal startup.

**To Reproduce**
Steps to reproduce the behavior:
1. Power off
2. Power on

**Screenshots or picture**
[image]

**Console output**
If applicable, add the debug console output here (here is a [how to](https://github.com/Phil1988/FreeDi/wiki/Bugs!))

....
=======================
Extruder max_extrude_ratio=207.875844
Config error
Traceback (most recent call last):
  File "/home/mks/klipper/klippy/klippy.py", line 130, in _connect
    self._read_config()
  File "/home/mks/klipper/klippy/klippy.py", line 127, in _read_config
    pconfig.check_unused_options(config)
  File "/home/mks/klipper/klippy/configfile.py", line 305, in check_unused_options
    raise error("Section '%s' is not a valid config section"
configparser.Error: Section 'freedi' is not a valid config section
webhooks client 281473780404112: New connection
webhooks client 281473780404112: Client info {'program': 'Moonraker', 'version': 'v0.9.3-0-g71f9e67'}
....

**Base (please complete the following information):**
 - Printer model: X-Plus 3
 - X3seriesLCD firmware version: v1.41-0
 - Source: v1.41-0
 - Mainboard: stock Mainboard X-4

[config-20250311-122735.zip](https://github.com/user-attachments/files/19181918/config-20250311-122735.zip)

Phil1988, 2025-03-11:
> Any clue how this can happen?

Yes. 
You failed to mention that you updated/reinstalled klipper on a secret mission, but CSI FreeDi has good reason to suspect that this is exactly what happened.

Please dont open a "bug" report if you messed with the system ;)

Now recover by doing:
`cd ~/FreeDi && ./install.sh`

closed 2025-03-11 by Phil1988

rmx-it, 2025-03-11:
> > Any clue how this can happen?
> 
> Yes. You failed to mention that you updated/reinstalled klipper on a secret mission, but CSI FreeDi has good reason to suspect that this is exactly what happened.
> 

Of course not (intentionally)!
After upgrading to 1.41 I did not touch klipper et. al.!
The only commands I used where ` poweroff`,  `diff ...`, `tar ...`, `mkdir old; mv ... old/` and ` nano macro.cfg` 

Is there a way possible the klipper changes during poweroff/poweron?!?

> Please dont open a "bug" report if you messed with the system ;)
> 

Yes. If messing up more or less intentionally.
But in this case, I did not mess with the system at all ?!?

> Now recover by doing: `cd ~/FreeDi && ./install.sh`
This did help indeed!
Thanks!

How can I see if the FreeDi specific extensions are in place?
~/klipper/klippy/extras/freedi.py ?

Phil1988, 2025-03-11:
I cant say what exactly you did, but I'm pretty sure that you reinstalled / soft reset / hard reset it, because the symlink is gone.

For checking only, you can `ls -la ~/klipper/klippy/extras/`.
But that doesnt really matter.
Executing the install.sh will recreate the symlink and everything is fine :)