Phil1988/FreeDi #187: FreeDi v1.42 released

Phil1988, 2025-03-11:
### v1.42 has just been released!

This update integrates the new project structure on GitHub.
Hopefully this will completely enable seamless **Moonraker Updates** for better maintainability. 

### Update instructions:
Please do:
```
cd ~ && rm -rf FreeDi && git clone https://github.com/Phil1988/FreeDi ~/FreeDi && cd ~/FreeDi && ./install.sh
```

This will bring you to v1.42
[image]

The next release will show if the update functionallity via Update Manager is now completely working :)

### Changelog and more information can be found on the release page:
https://github.com/Phil1988/FreeDi/releases/tag/v1.42

pmbroth, 2025-03-11:
I just ran the script,  the system goes through the update, and when it gets to Freedi Service is starting, the printer reboots automatically and restarts klipper, while the updating package list is running.  It reboots fine, but still on older version.

pmbroth, 2025-03-11:
I tried it again, same issue.  stays on 1.41, and reboots right before updating package after freedi Service starts

pmbroth, 2025-03-11:
Screen contents
mks@mkspi:~$ cd ~ && rm -rf FreeDi && git clone https://github.com/Phil1988/FreeDi ~/FreeDi && cd ~/FreeDi && ./install.sh
Cloning into '/home/mks/FreeDi'...
remote: Enumerating objects: 1494, done.
remote: Counting objects: 100% (153/153), done.
remote: Compressing objects: 100% (103/103), done.
Receiving objects:  27% (404/1494), 21.16 MiB | 5.29 MiB/s
Message from syslogd@mkspi at Mar 11 07:52:21 ...
 kernel:[   66.116193] Disabling IRQ #42
remote: Total 1494 (delta 105), reused 74 (delta 50), pack-reused 1341 (from 3)
Receiving objects: 100% (1494/1494), 76.22 MiB | 5.63 MiB/s, done.
Resolving deltas: 100% (800/800), done.
Do you use the stock Mainboard? (y/n)
Y
Starting the installation...
Sparse checkout only the required folders...
fatal: no sparse-checkout to add to
fatal: no sparse-checkout to add to
fatal: no sparse-checkout to add to
fatal: no sparse-checkout to add to
fatal: no sparse-checkout to add to
Configuring git to fetch tags automatically...
Removing freedi_update.sh from git index because it's already tracked...
Disabling freedi_update.sh git tracking for future modifications...
Successfully ignoring freedi_update.sh
Local ignore setup completed. The file freedi_update.sh will now be ignored locally by git.
Creating a symbolic link for freedi.py from /home/mks/FreeDi/klipper_module to /home/mks/klipper/klippy/extras...
Successfully installed freedi.py to /home/mks/klipper/klippy/extras.
Restarting Klipper service...
[sudo] password for mks:
Klipper service restarted successfully.
Installation complete.
Setup dtbo for serial communication...
dtbo install done!
Customize the armbianEnv.txt file for serial communication...
Line found. Replacing the line.
armbianEnv.txt file modified successfully!
Setup the toolhead serial path in printer.cfg...
Found serial device: /dev/serial/by-id/usb-Klipper_rp2040_D5D44D9552A25058-if00
Modifying printer.cfg
Updated serial path for the toolhead in /home/mks/printer_data/config/printer.cfg
Activating Klipper virtual environment and installing Python packages...
Klipper environment python version: 3.11.2
Arranging Python requirements...
Requirement already satisfied: pip in /home/mks/klippy-env/lib/python3.11/site-packages (25.0.1)
Requirement already satisfied: configparser in /home/mks/klippy-env/lib/python3.11/site-packages (from -r /home/mks/FreeDi/requirements.txt (line 1)) (7.1.0)
Requirement already satisfied: requests in /home/mks/klippy-env/lib/python3.11/site-packages (from -r /home/mks/FreeDi/requirements.txt (line 2)) (2.32.3)
Requirement already satisfied: pyserial>=3.5 in /home/mks/klippy-env/lib/python3.11/site-packages (from -r /home/mks/FreeDi/requirements.txt (line 3)) (3.5)
Requirement already satisfied: Pillow in /home/mks/klippy-env/lib/python3.11/site-packages (from -r /home/mks/FreeDi/requirements.txt (line 4)) (11.0.0)
Requirement already satisfied: numpy in /home/mks/klippy-env/lib/python3.11/site-packages (from -r /home/mks/FreeDi/requirements.txt (line 5)) (1.26.2)
Requirement already satisfied: matplotlib in /home/mks/klippy-env/lib/python3.11/site-packages (from -r /home/mks/FreeDi/requirements.txt (line 6)) (3.8.2)
Requirement already satisfied: psutil in /home/mks/klippy-env/lib/python3.11/site-packages (from -r /home/mks/FreeDi/requirements.txt (line 7)) (6.1.1)
Requirement already satisfied: websocket-client in /home/mks/klippy-env/lib/python3.11/site-packages (from -r /home/mks/FreeDi/requirements.txt (line 8)) (1.8.0)
Requirement already satisfied: charset-normalizer<4,>=2 in /home/mks/klippy-env/lib/python3.11/site-packages (from requests->-r /home/mks/FreeDi/requirements.txt (line 2)) (3.4.0)
Requirement already satisfied: idna<4,>=2.5 in /home/mks/klippy-env/lib/python3.11/site-packages (from requests->-r /home/mks/FreeDi/requirements.txt (line 2)) (3.10)
Requirement already satisfied: urllib3<3,>=1.21.1 in /home/mks/klippy-env/lib/python3.11/site-packages (from requests->-r /home/mks/FreeDi/requirements.txt (line 2)) (2.2.3)
Requirement already satisfied: certifi>=2017.4.17 in /home/mks/klippy-env/lib/python3.11/site-packages (from requests->-r /home/mks/FreeDi/requirements.txt (line 2)) (2024.12.14)
Requirement already satisfied: contourpy>=1.0.1 in /home/mks/klippy-env/lib/python3.11/site-packages (from matplotlib->-r /home/mks/FreeDi/requirements.txt (line 6)) (1.3.1)
Requirement already satisfied: cycler>=0.10 in /home/mks/klippy-env/lib/python3.11/site-packages (from matplotlib->-r /home/mks/FreeDi/requirements.txt (line 6)) (0.12.1)
Requirement already satisfied: fonttools>=4.22.0 in /home/mks/klippy-env/lib/python3.11/site-packages (from matplotlib->-r /home/mks/FreeDi/requirements.txt (line 6)) (4.55.3)
Requirement already satisfied: kiwisolver>=1.3.1 in /home/mks/klippy-env/lib/python3.11/site-packages (from matplotlib->-r /home/mks/FreeDi/requirements.txt (line 6)) (1.4.7)
Requirement already satisfied: packaging>=20.0 in /home/mks/klippy-env/lib/python3.11/site-packages (from matplotlib->-r /home/mks/FreeDi/requirements.txt (line 6)) (24.2)
Requirement already satisfied: pyparsing>=2.3.1 in /home/mks/klippy-env/lib/python3.11/site-packages (from matplotlib->-r /home/mks/FreeDi/requirements.txt (line 6)) (3.2.0)
Requirement already satisfied: python-dateutil>=2.7 in /home/mks/klippy-env/lib/python3.11/site-packages (from matplotlib->-r /home/mks/FreeDi/requirements.txt (line 6)) (2.9.0.post0)
[... cut ...]
Hit:2 http://deb.debian.org/debian bookworm-updates InRelease
Hit:3 http://deb.debian.org/debian bookworm-backports InRelease
Hit:4 http://security.debian.org bookworm-security InRelease
Hit:6 https://download.docker.com/linux/debian bookworm InRelease
Hit:5 http://netcup.armbian.com/apt bookworm InRelease
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
9 packages can be upgraded. Run 'apt list --upgradable' to see them.
Setup complete!
Please restart your system for the changes to take effect.

pmbroth, 2025-03-11:
I rebooted, and retried the update for a 3rd time  now..  :-)

pmbroth, 2025-03-11:
I reloaded and checked for the 4th time, for updates now I get repo is dirty.

pmbroth, 2025-03-11:
first two times, the machine rebooted at the same point as I mentioned, on the third time running the script, it did not reboot automatically. (Yes, I saw it reboot, as I lost connection via terminal software, and system was on reboot screen).  3rd time I ran the script, I had to reboot system, manually. But as mentioned, dirty repo, and 1,42 displayed, but on tft 1.41 for tft and software 1.41.

Note: Prior to running the script each time, I clean booted before running the script.

pmbroth, 2025-03-11:
I tried it a 4th time, and right about the time the screen is saying autoflasherr service is enabled my machine restarts ...and the script on the screen is still running while the machine is restarting

pmbroth, 2025-03-11:
After I ran the update for the 4th time, the  Dirty repo issue has been resolved, only issue is on display of machine, tft still 1.41 and software version 1.41.  

Mainsail and fluid, see as below:

[image]

Phil1988, 2025-03-11:
Now thats an impressive spam wall 😄
@pmbroth please hold your horses dude 😅

Please show me:
`ls /home/mks/FreeDi/screen_firmwares`

Then we walk through it together. slowly. one by one :D 

I'm still not home yet. ETA is 1h.

pmbroth, 2025-03-11:
mks@mkspi:~$ ls /home/mks/FreeDi/screen_firmwares
x-plus3_x-max3_beta_v1.420.tft  x-plus3_x-max3_stable_v1.42.tft  x-smart3_beta_v1.420.tft  x-smart3_stable_v1.42.tft

pmbroth, 2025-03-11:
I type like I talk....   ;-)

billkenney, 2025-03-11:
I updated (delete FreeDi, git clone, install.sh) and ran the install script without issue and rebooted afterwards with sudo reboot. When I initially checked the version in fluidd it said 1.41. I clicked check for updates and it refreshed the version to 1.42 (showing that it had been updated). The about section of the screen still said 1.41 for the software and screen firmware. I rebooting again after checking for updates in fluidd, and the about screen still said 1.41 for the software and screen. I put the 1.42 tft file on a micro sd card and updated that way and now the about screen shows version 1.42 for the screen, but it still shows version 1.41 for the software. I have the same files in screen_firmwares as pmbroth.

RustysGarage, 2025-03-11:
same as above, tft screen shows 1.41 software on boot and, in mainsail do a refresh with update mngr shows 1.42 but screen shows 1.41  and on about page on screen same thing 1.41 for both

Phil1988, 2025-03-11:
I'm back home and I'll check and fix this in the next... 10-15min

Phil1988, 2025-03-11:
Ok, I just pushed the fix, please redo the update command everyone and be prepaired, that the display update may fail.
If this happens, please power off and turn on again.

pmbroth, 2025-03-11:
ok, I will let you know how it goes, Thank you Phil...

pmbroth, 2025-03-11:
For me as soon as it does the auto flasher service, my printer reboots, and  now right after it rebooted, the screen flash started. And while  the machine rebooted,  the script kept running.   And yes the machine had the reboot screen on the display, while the script on the computer was still running.   If you want after this is completed, I can rerun the script and try to take a video and see if it does the  same thing?

pmbroth, 2025-03-11:
Machine is up and running, all on 1.42, I can run the script and take a video, if you want me to see if it will do the same behavior

Phil1988, 2025-03-11:
> Machine is up and running, all on 1.42, I can run the script and take a video, if you want me to see if it will do the same behavior

Yes please do a screen capture video, that would help 👍

pmbroth, 2025-03-11:
I did it again, all is up and running , I have a video, showing the reboots and the script still running.  

Here is a link to the video:  ping me if you don't have access at my email address with an email address for you to access:  

https://www.dropbox.com/scl/fi/1sxpl6rqrxrdim1owa032/IMG_1654.mov?rlkey=apnvhu0dawb8pvswd29x6y9yp&st=5xqmw7xy&dl=0

Phil1988, 2025-03-11:
> did 3 machines so far and all 3 worked perfect no hang ups with the update

That's awesome! 🎉

But careful - starting from 2 printers, the **FreeDi Licensing Fee** kicks in! 

The cost grows exponentially with each additional machine. 
At 3 printers, you owe me a fresh self brewed tea, and at 5… well, let's just say I hope you like expensive dinners! 😆
(It's a joke... I mention that to make sure to not freak you out 😉)

pmbroth, 2025-03-11:
Phil - let me know if you have access to the video.  Sorry for the crazy movements

Phil1988, 2025-03-11:
> Here is a link to the video

Thanks for the video this was really helpfull.
I could see it and it was totaly not confusing 👍 

Everything is allright!

While installing, it kills the FreeDi.service at some point. 
When this is happening, the screen kind of resets its state and thinks its rebooting - while the host (your printer) is actually not.
So everything is allright 😇

pmbroth, 2025-03-11:
Ok, so as you see, I'm not nuts, well I guess it is who you ask..  :-)  Thank you.  Brian

Phil1988, 2025-03-11:
All good, you just worried too much and took wrong assumption because you didnt had the insides.
Thanks for reporting it tho (also if it in this case was totally normal).

While you did that I found a uncritical side catch and fixed it already 👍

pmbroth, 2025-03-11:
I was not aware that the host should be rebooting during the update process, and that it can still receive data while it was updating.  Thank you for this explanation, makes sense.  This may be something for the Wiki, that the host will reboot during the update process.  .  Brian

rmx-it, 2025-03-11:
Hi,
did work.
First download froze at 5%.
After power-cycle everything got well.
Distplay FW is on 1.42 now.

Phil1988, 2025-03-11:
> This may be something for the Wiki, that the host will reboot during the update process.

It wont. But it is not a bad idea to reboot the host manually.

> Hi, did work. First download froze at 5%. After power-cycle everything got well. Distplay FW is on 1.42 now.

Yeah that its still not entirely solved and super hard for me to fix.
At least you were prepared and it did work on the 2nd attempt 👍

RustysGarage, 2025-03-11:
i got ya Phil!!!  i only have 6 Qidi,  the other 2 are plus 4 and q1 pro, but they are waiting hahahah

jawzx2, 2025-03-11:
Update seemed to go smoothly after reboot and flash of the tool head and board mcu. everything says it's on FreeDi 1.42!  just a quick question: as I am upgrading from 1.4 should I be getting those updated macros manually like the 1.41 update asked for?

Phil1988, 2025-03-11:
> i got ya Phil!!! i only have 6 Qidi, the other 2 are plus 4 and q1 pro, but they are waiting hahahah

That will get very pricey mate ;)

> Update seemed to go smoothly [...].  should I be getting those updated macros manually like the 1.41 update asked for?

I'm happy to hear that 👍 
Yes, I do recommend getting the updated macros, but they are not super crucial if you use Orca.
If you are using QIDISlicer I highly recommend it tho.

Ludn, 2025-03-12:
Glad to hear about update :)

Done via ssh: coming from 1.40, initiated an other display update, which was successful on 2nd try.

main difference of log from above:

`remote: Enumerating objects: 1498, done.
remote: Counting objects: 100% (155/155), done.
remote: Compressing objects: 100% (102/102), done.
...
remote: Total 1498 (delta 107), reused 78 (delta 53), pack-reused 1343 (from 3)
Receiving objects: 100% (1498/1498), 76.22 MiB | 5.00 MiB/s, done.
Resolving deltas: 100% (803/803), done.
...
Arranging Python requirements...
Requirement already satisfied: pip in /home/mks/klippy-env/lib/python3.11/site-packages (25.0)
Collecting pip
  Downloading pip-25.0.1-py3-none-any.whl.metadata (3.7 kB)
Downloading pip-25.0.1-py3-none-any.whl (1.8 MB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 1.8/1.8 MB 4.3 MB/s eta 0:00:00
Installing collected packages: pip
  Attempting uninstall: pip
    Found existing installation: pip 25.0
    Uninstalling pip-25.0:
      Successfully uninstalled pip-25.0
Successfully installed pip-25.0.1

Requirement already satisfied: numpy in /home/mks/klippy-env/lib/python3.11/site-packages (from -r /home/mks/FreeDi/requirements.txt (line 5)) (2.2.0)
Requirement already satisfied: matplotlib in /home/mks/klippy-env/lib/python3.11/site-packages (from -r /home/mks/FreeDi/requirements.txt (line 6)) (3.10.0)

Detecting WiFi chip...
RTL8188GU detected.
vendor_id: 0bda, product_id: b711
Look for default devices ...
 Found devices in default mode (1)
Access device 003 on bus 002`

Reboot screen is 1.42, but Update Manager stays on 'FreeDi v1.30-85-g64e22fd0',
 so was mismatched before, but seems to run smooth.

[but freedi_update.sh is for all < v1.30 users, right?]

Phil1988, 2025-03-13:
It's for all <= v1.20 users :)

Hit the refresh button on update manager.
Currently this is the most reported "bug" which I guess it's a pretty good sign :D

Ludn, 2025-03-13:
Confirmed 👍 -last time, during LCD update, was a bit too early for that :D

cptcl, 2025-03-13:
@Phil1988 
Did start the update.. From 1.41 like in the install instructions.
Printer rebooted and startet display update.
But is it normal to be only on  29% after like 15 minutes? 
I waited 3-4 minutes and its still on 29% now.
1.41 was updated in like 10 minutes

PS: X-Max 3 Rev2 here

MaZZly, 2025-03-15:
@Phil1988 is it possible to go straight from `v1.40-0-g99a27099` to `1.42`, or should I do `1.41` first before continuing to latest? :)

Phil1988, 2025-03-15:
@MaZZly You can go directly.
But I recommend you have a look at the [macro changes](https://github.com/Phil1988/FreeDi/issues/174#issue-2889220921) to include what has been improved (but its not a must).

TopheC, 2025-03-18:
Hi all.

Did work. 
BUT, first update froze at 80%. And this time, I wait a loooooong time, before switching off. Wait another 10-15mn for heat to dissipate. After power-cycle on update restart from 0% to 100%, flowlessly. Everything is OK now.
Screen/display says v1.42 now. Mainsail too.
Must only make a "Full Home", each axis at a time, just because first print after upgrade threw a clear "positionning" error of the head.
Printing now....

Good job @Phil1988

Phil1988, 2025-03-18:
@TopheC 
No need to wait before switching off.
When the screen updated stops, the system is frozen and can not be recovered.
Still investigating this problem.

About the individual homing of each axis: this happens sometimes and is related to klipper.
A klipper restart should fix it.
If it persists, open a discussion/bug report/join on discord :)

I'm happy that beside this inconvinience everything works :)

jimwhitelaw, 2025-03-26:
I’ve just done a fresh install of 1.42 by installing 1.40 and then running the update. It is mostly working now, except that it shows the mcu is still on v0.10 and mainsail indicates kipper is “invalid”. I’m not sure how to proceed from here. I had to run the update (as posted at the top of this topic) a few times to get the WiFi to be re-enabled, and I’m not sure if the message about the mcu fw is accurate, as I did see the fw update screen at least twice during the install & update process. What can I do to get the update completed?

[image]

souriceaux06, 2025-04-04:
Hello from Menton, France,
Your display management interface is truly superb.
I'd like to update your V1.40 version with the image you created.
But my printer isn't connected via the internet.
Is it possible to easily update to V1.42 using a USB drive?
Thank you for your help and response.
Sincerely,

Phil1988, 2025-04-05:
> Is it possible to easily update to V1.42 using a USB drive?

Not fully automatically.
You can download this project to a USB drive/ notebook and replace the ~/FreeDi folder with it.
Then you need the install command I showed above.
Both of these steps need some kind of ssh terminal to do this - so a computer or connected phone is required.

But: I recommend not updating on your case.
v2.0 is about to come this month.

souriceaux06, 2025-04-05:
Hello Phill1988,
Thank you so much for your help and response. 
I will follow your advice and wait for version V2. 
Thank you again.

pmbroth, 2025-04-15:
DId you try Orca Slicer?

dirtydolph, 2025-04-15:
> DId you try Orca Slicer?

I'm trying now, because didn't configured as Qidi Studio, and got issues with the SFS Filament Sensor config not work as @Phil1988 post

Also you can see that error: Must home axis first: 10.000 10.000 2.000 [0.000] 
Is an error or not?

[image]

Now is printing using Qidi Slicer, but I don't like as Qidi Studio

Phil1988, 2025-04-22:
Hello @dirtydolph and welcome on board!
Please dont hijack this, as its not a support thread ;)

About Slicers: https://github.com/Phil1988/FreeDi/discussions/193

About your "Must home axis first": This is somehow common of klipperin certain cenarios and happens if a restart is needed.

About the SFS filament sensor: Its more or less up to you to make your custom mods work, but many do use it with freedi.
I ask you to open a discussion here on git or ask at the discord channel.

This thread is not for questions/issues like this.

Ferropluss, 2025-04-24:
Hello. How do I fix this?

[image]

jensbodal, 2025-05-02:
FWIW it looks like the script asks for a sudo password after the initial run, and if you aren't watching it will time out and skip past the part that runs after. 

```
Creating udev rule...
Udev rule created: /etc/udev/rules.d/99-usb-thumb.rules
Creating systemd service file...
Systemd service file created: /etc/systemd/system/usb-mount@.service
Making the usbmounter.sh executable...
Script /home/mks/FreeDi/helpers/usbmounter.sh is now executable.
==== AUTHENTICATING FOR org.freedesktop.systemd1.reload-daemon ====
Authentication is required to reload the systemd state.
Authenticating as: MKS,,, (mks)
Password:
==== AUTHENTICATION COMPLETE ====
Udev and systemd have been reloaded.
USB mounter service created successfully!
Installing FreeDi service...
Stopping FreeDi service...
FreeDi service stopped.
Moving FreeDi.service to /etc/systemd/system/
FreeDi.service moved to /etc/systemd/system/
Setting user to mks in FreeDi.service
Enabling FreeDi.service to start at boot...
```

This portion right here, the first time it ran this happened:

```
Creating udev rule...
Udev rule created: /etc/udev/rules.d/99-usb-thumb.rules
Creating systemd service file...
Systemd service file created: /etc/systemd/system/usb-mount@.service
Making the usbmounter.sh executable...
Script /home/mks/FreeDi/helpers/usbmounter.sh is now executable.
==== AUTHENTICATING FOR org.freedesktop.systemd1.reload-daemon ====
Authentication is required to reload the systemd state.
Authenticating as: MKS,,, (mks)
Password: Reload daemon failed: Method call timed out
polkit-agent-helper-1: pam_authenticate failed: Authentication failure
Udev and systemd have been reloaded.
USB mounter service created successfully!
Installing FreeDi service...
Stopping FreeDi service...
FreeDi service stopped.
Moving FreeDi.service to /etc/systemd/system/
FreeDi.service moved to /etc/systemd/system/
Setting user to mks in FreeDi.service
Enabling FreeDi.service to start at boot...
```

I have no idea if that matters, sorry if it's just extra noise. 

A few notes for me:

* I went from stock qidi 32gb emmc to 1.40
* on first boot the screen froze (as mentioned somewhere it's a known issue), I rebooted to next step
* I believe after the first screen freeze it booted just fine and then I didn't really know what to do
* Guide mentions to access printer via webui, would be nice if at this stage the ip address was displayed (if possible) since I believe it just shows 4 reboot options. At any rate I just found the ip address on my router
* Still going through everything else but have one successful so far :)

Thanks for all the work it's been an awesome experience so far.  I have more details and notes but just haven't organized it all yet

Phil1988, 2025-05-02:
@jensbodal 
Yes that is normal - I mentioned it already a couple of times, but it wont hurt to let others know here :)

About the IP-address:
Yes! The next major version has this and much more already integrated :)

Regarding your feedback: You can post it at the discussions section or even better - get in touch with me and others via [discord](https://github.com/Phil1988/FreeDi/wiki/Community)

potens1, 2025-06-06:
Oooops, I followed the `cd ~ && rm -rf FreeDi && git clone https://github.com/Phil1988/FreeDi ~/FreeDi && cd ~/FreeDi && ./install.sh` command, and I ended with 

[image]

and now the screen is stuck flashing at 38%, for an hour already...

Phil1988, 2025-06-07:
No worries mate.
I think you dont loose anything if your system now looks a bit... weird :D
Most likely its at the latest stable release state but git and moonraker doesnt reflect that correctly.

I suggest you keep it like this at the time.
There isn't too much that have changed from 1.40 to 1.42.

D3FCONau, 2025-06-21:
odd issue, flashed 1.40, Shows v1.40-7-g64e22fd0 in machine, When i click check for updates it shows version v1.30-85-g64e22fd0 "Up To Date" So my bigger issue is If I run cd ~ && rm -rf FreeDi && git clone https://github.com/Phil1988/FreeDi ~/FreeDi && cd ~/FreeDi && ./install.sh it will then show Version 2.0 and this is bad. it's unuseable. I just want to get to 1.4.2, Pehaps you need to update the url for the update command so it does not process to 2.0

D3FCONau, 2025-06-22:
Thank you for your response, Unuesable it the issue i have faced with my X-plus 3 for a while now. No Z offsets are saving, I have to baby step each print. 

Issue started about 3 months ago. out of the blue. Revcerted back to stock and seems issue is consistant with factory firmware also now so thats a bit odd. 

If i restart / shutdown or save_config nothing is changed on reboot. Ill do a Z off set via screen "LCD" or in clipper. set it and then try to run a print is is about 1 mil or more off the plate.

closed 2025-06-23 by Phil1988