Phil1988/FreeDi #189: [Bug] XPlus3 software image shows picture and text of X-Max3 on Display screen after install.

Ramon-Santini, 2025-03-13:
**Describe the bug**
Not sure if this is because the image(s) are all made off of a standard template based off of the X-Max 3 but when installing the image for the X-Plus3 the start up display shows and the text on the display makes reference to the X-Max3 instead of the X-Plus3. :(

**Expected behavior**
Software Image for X-Plus3 should show a picture of the X-Plus3 on the display.

**Base (please complete the following information):**
 - Printer model: X-Plus 3
 - Source: Latest FreeDi image installed and then updated with the Klipper Update Manager to 1.4

Phil1988, 2025-03-13:
Hey @Ramon-Santini 
That's not a bug, it's you ;)

See also here:
https://github.com/Phil1988/FreeDi/issues/155

Check the menu and set the printer model to plus3.
I showed it in the video and I hope you find it now.

closed 2025-03-13 by Phil1988

Ramon-Santini, 2025-03-13:
I would have sworn I went through all of those screens and somehow I missed that! DOH!

Phil1988, 2025-03-14:
I'm happy that you found it now