Phil1988/FreeDi #190: Support both klipper [fan_generic] and [output_pin] config sections

savardd, 2025-03-13:
**Describe the feature**
Klipper supports [fan_generic] section that defines fans instead of output pins.  By using them to define filterfan and partfan (and sidefan for x-max 3 and x-plus 3) the Mainsail interface will display fan icons that will rotate when the fans are turned on.

This mostly works but the event listener from the screen.  It will need to register both objects for each fan (filterfan, partfan and sidefan):
```
{"id": 123, "method": "objects/subscribe", "params": {"objects":{"output_pin filterfan": ["value"], "webhooks": ["state"]}, "response_template":{}}}
```
and
```
{"id": 123, "method": "objects/subscribe", "params": {"objects":{"fan_generic filterfan": ["speed"], "webhooks": ["state"]}, "response_template":{}}}
```

As only one of the hooks may exists at the same time (a pin may be defined as output_pin or fan_generic but not both at same time), the screen listener only have to update itself when it receive an update on one of them.

With this change, the screen will support both, output_pin section and fan_generic section.  Users will have the choice between the two way of running their fans.  And, in Mailsail, the fan sliders will have the beautiful fan icon (that will rotate if running):

[image]

---------------
**More information on the change needed**

The configuration of the printer will/may change from:
```
## Part cooling fan 
[output_pin partfan]
pin: MKS_THR:gpio2
pwm: True
cycle_time: 0.0100
hardware_pwm: false
value: 0
scale: 255
shutdown_value: 0.0
```

To (for each supported fan):
```
## Part cooling fan 
[fan_generic partfan]
pin: MKS_THR:gpio2
cycle_time: 0.0100
hardware_pwm: false
```
-----------------------

And the M106 macro will/may change from (just a small sample showed below):
```
########################################
# Fan Macros
########################################
# Set Fan Speed macro
# Mainly name castings for nicer web interface names and display functionality
[gcode_macro M106]
gcode:  
    {% if params.P is defined %}
      {% if params.S is defined %}
        {% if (params.P|int) == 0 %}
          SET_PIN PIN=partfan VALUE={params.S|int}
        {% elif (params.P|int) == 3 %}
          SET_PIN PIN=filterfan VALUE={params.S|int}   
        {% elif (params.P|int) == 2 %}
          # fix for X-Smart 3 "missing fan2" -> dont do anything                  
        {% else %}
          SET_PIN PIN=fan{params.P|int} VALUE={params.S|int}
        {% endif %}
      {% else %}
        {% if (params.P|int) == 0 %}
          SET_PIN PIN=partfan VALUE=255
        {% elif (params.P|int) == 3 %}
          SET_PIN PIN=filterfan VALUE=255    
        {% elif (params.P|int) == 2 %}
          # fix for X-Smart 3 "missing fan2" -> dont do anything
        {% else %}
          SET_PIN PIN=fan{params.P|int} VALUE=255
        {% endif %}
      {% endif %}
    {% endif %}
```

To:
```
########################################
# Fan Macros
########################################
# Set Fan Speed macro
# Mainly name castings for nicer web interface names and display functionality
[gcode_macro M106]
gcode:  
    {% if params.P is defined %}
      {% if params.S is defined %}
        {% set speed = 0.003921*params.S|int %}
        {% if (params.P|int) == 0 %}
          SET_FAN_SPEED FAN=partfan SPEED={speed}
        {% elif (params.P|int) == 3 %}
          SET_FAN_SPEED FAN=filterfan SPEED={speed}
        {% elif (params.P|int) == 2 %}
          # fix for X-Smart 3 "missing fan2" -> dont do anything                  
        {% else %}
          SET_FAN_SPEED FAN=fan{params.P|int} SPEED={speed}
        {% endif %}
      {% else %}
        {% if (params.P|int) == 0 %}
          SET_FAN_SPEED FAN=partfan SPEED=1
        {% elif (params.P|int) == 3 %}
          SET_FAN_SPEED FAN=filterfan SPEED=1
        {% elif (params.P|int) == 2 %}
          # fix for X-Smart 3 "missing fan2" -> dont do anything
        {% else %}
          SET_FAN_SPEED FAN=fan{params.P|int} SPEED=1
        {% endif %}
      {% endif %}
    {% endif %}
```

**A little hack**

On my side, to further test, I made a small hack on the klipper's webhook.py file.  It is not be needed if the screen listener is updated to listen to both: output_pin and fan_generic at same time.  This is to *cheat* on the current subscription on output_pin by faking generating a fake output_pin section for the event response.  With the above modification to printer.cfg and macro.cfg and the hack below, I can have the beautiful fan running.  *Note: this will mark the klipper repository dirty*

Here is the modification I made (patch file) in the ~/klipper directory:
```
diff --git a/klippy/webhooks.py b/klippy/webhooks.py
index bccc5aace..3ccaf9ee8 100644
--- a/klippy/webhooks.py
+++ b/klippy/webhooks.py
@@ -506,6 +506,10 @@ class QueryStatusHelper:
                         cres[ri] = rd
                 if cres or is_query:
                     cquery[obj_name] = cres
+                    if obj_name.startswith('fan_generic'):
+                        alt_name = obj_name.replace('fan_generic','output_pin')
+                        cres['value'] = cres['speed'] or 0.0
+                        cquery[alt_name] = cres
             # Send data
             if cquery or is_query:
                 tmp = dict(template)
```

savardd, 2025-03-13:
Also, is it possible to access the underlying python code of the screen listener?  
It seems that it is compiled to executable libraries.
Thanks for your wonderful jobs!

Phil1988, 2025-03-13:
Thanks for your detailed report, the effort you did for finding a solution and for sharing.

Your solution will work and it's already on my list to change that (on a low prio task).
My approach it to change this entirely to generic fans.

I'm happy to have you in board and I'm looking forward for the optimizations you might come up with :)

SmileyMan, 2025-03-19:
I have been working on the same thing. And come to the same conclusions that the screen code needs to be modified to fully support. 

I used this logic to convert the byte into decimal:
```
{% set speed = (params.S | int) / 255 | round(2) %}
```

Also found changing the names to like part_fan (adding space) caused issues.

For me moving fans to generic fans was a secondary requirement. I am using Kalico and MPC rather than PID for the hot end.

Config:
```
[extruder]
#control: mpc
heater_power: 115  
cooling_fan: fan
filament_diameter: 1.75
filament_density: 1.20
filament_heat_capacity: 1.8 
```

cooling_fan must be a the part fan. You can not use generic_fan here I found.

I ended up using:
```
[duplicate_pin_override]
pins: MKS_THR:gpio2
```

Then dual declaring the Part Fan to allow me to calibrate MPC heating:
```
## Part cooling fan 
[fan]
pin: MKS_THR:gpio2
max_power: 1.0
shutdown_speed: 0.0
cycle_time: 0.0100
hardware_pwm: false

## Part cooling fan pin
[output_pin partfan]
pin: MKS_THR:gpio2
pwm: True
cycle_time: 0.0100
hardware_pwm: false
value: 0
scale: 255
shutdown_value: 0.0
```

Maybe the Part Fan declared as [fan] could be supported on the screen in the future?

A change to use gerneric_fan would be great.

Phil1988, 2025-03-19:
Here we go @savardd and @SmileyMan :

I worked on this and every possiibility will be supported for maximum freedom and compatibility:
You can use [output_pin]
You can use [fan_generic] 
You can use [fan]

I tested every combination and its working 😊
It will be rolled out with the next release, which will probably take some time.
I'll notify you here once the new version is released :)

SmileyMan, 2025-03-19:
> Here we go [@savardd](https://github.com/savardd) and [@SmileyMan](https://github.com/SmileyMan) :
>
> I worked on this and every possiibility will be supported for maximum freedom and compatibility: [...]

Thank you.