Phil1988/FreeDi #194: [Bug] Qidi Camera No Signal

alexkipar, 2025-03-15:
After the FreeDI installation, Crowsnest installation, and setup of the camera in the mainsail, it is not working. It shows either "No Signal" or "disconnected". It worked fine before.

The only change I've made to USB connections is moving the Wi-Fi dongle from the motherboard to the printer's top location. 
The second strange thing the scan shows that there are 2 camera devices (any of them in crowsnest.conf leads to same result):
usb-SYX-230524-J_HD_Camera-video-index0
usb-SYX-230524-J_HD_Camera-video-index1

X-Max 3 and the latest firmware

I've tried various setup for crowsnest.conf, current one:

```
[crowsnest]
log_path: /home/mks/printer_data/logs/crowsnest.log
log_level: verbose                      # Valid Options are quiet/verbose/debug
delete_log: false                       # Deletes log on every restart, if set to true
no_proxy: false                         # If set to true, no reverse proxy is required. Only change this, if you know what you are doing.

[cam 1]
mode: ustreamer                         # ustreamer - Provides MJPG and snapshots. (All devices)
                                        # camera-streamer - Provides WebRTC, MJPG and snapshots. (only RPiOS + RPi 0/1/2/3/4)
enable_rtsp: false                      # If camera-streamer is used, this also enables usage of an RTSP server
rtsp_port: 8554                         # Set different ports for each device!
port: 8080                              # HTTP/MJPG stream/snapshot port
device: usb-SYX-230524-J_HD_Camera-video-index0                     # See log for available devices
resolution: 640x480                     # <width>x<height> format
max_fps: 5                             # If hardware supports it, it will be forced, otherwise ignored/coerced.
```

alexkipar, 2025-03-15:
Crowsnest shows that device has been lost somehow

[03/15/25 15:30:27] crowsnest: crowsnest - A webcam Service for multiple Cams and Stream Services.
[03/15/25 15:30:27] crowsnest: Version: v4.1.12-1-g9623b5f
[03/15/25 15:30:27] crowsnest: Prepare Startup ...
[03/15/25 15:30:27] crowsnest: INFO: Host information:
[03/15/25 15:30:27] crowsnest: Host Info: Distribution: Armbian-unofficial 24.2.0-trunk bookworm
Armbian-unofficial 24.2.0-trunk bookworm
[03/15/25 15:30:27] crowsnest: Host Info: Kernel: Linux 6.6.17-current-rockchip64 aarch64
[03/15/25 15:30:27] crowsnest: Host Info: Available CPU Cores: 4
[03/15/25 15:30:27] crowsnest: Host Info: Available Memory: 932920 kB
[03/15/25 15:30:27] crowsnest: Host Info: Diskspace (avail. / total): 23G / 29G
[03/15/25 15:30:27] crowsnest: INFO: Checking Dependencies
[03/15/25 15:30:27] crowsnest: Dependency: 'crudini' found in /usr/bin/crudini.
[03/15/25 15:30:27] crowsnest: Dependency: 'find' found in /usr/bin/find.
[03/15/25 15:30:27] crowsnest: Dependency: 'xargs' found in /usr/bin/xargs.
[03/15/25 15:30:27] crowsnest: Dependency: 'ustreamer' found in bin/ustreamer/src/ustreamer.bin.
[03/15/25 15:30:27] crowsnest: Version Control: ustreamer is up to date. (v6.31)
[03/15/25 15:30:27] crowsnest: INFO: Print Configfile: '/home/mks/printer_data/config/crowsnest.conf'
[03/15/25 15:30:27] crowsnest:		[crowsnest]
[03/15/25 15:30:27] crowsnest:		log_path: /home/mks/printer_data/logs/crowsnest.log
[03/15/25 15:30:27] crowsnest:		log_level: verbose
[03/15/25 15:30:27] crowsnest:		delete_log: false
[03/15/25 15:30:27] crowsnest:		no_proxy: false
[03/15/25 15:30:27] crowsnest:		
[03/15/25 15:30:27] crowsnest:		[cam 1]
[03/15/25 15:30:27] crowsnest:		mode: ustreamer
[03/15/25 15:30:27] crowsnest:		
[03/15/25 15:30:27] crowsnest:		enable_rtsp: false
[03/15/25 15:30:27] crowsnest:		rtsp_port: 8554
[03/15/25 15:30:27] crowsnest:		port: 8080
[03/15/25 15:30:27] crowsnest:		device: usb-SYX-230524-J_HD_Camera-video-index0
[03/15/25 15:30:27] crowsnest:		resolution: 640x480
[03/15/25 15:30:27] crowsnest:		max_fps: 5
[03/15/25 15:30:27] crowsnest: INFO: Detect available Devices
[03/15/25 15:30:27] crowsnest: INFO: Found 1 total available Device(s)
[03/15/25 15:30:27] crowsnest: INFO: Found 1 available v4l2 (UVC) camera(s)
[03/15/25 15:30:27] crowsnest: /dev/v4l/by-id/usb-SYX-230524-J_HD_Camera-video-index0 -> /dev/video3
[03/15/25 15:30:27] crowsnest: Supported Formats:
[03/15/25 15:30:27] crowsnest:		[0]: 'MJPG' (Motion-JPEG, compressed)
[03/15/25 15:30:27] crowsnest:		Size: Discrete 1280x720
[03/15/25 15:30:27] crowsnest:		Interval: Discrete 0.033s (30.000 fps)
[03/15/25 15:30:27] crowsnest:		Size: Discrete 1920x1080
[03/15/25 15:30:27] crowsnest:		Interval: Discrete 0.033s (30.000 fps)
[03/15/25 15:30:27] crowsnest:		Size: Discrete 640x480
[03/15/25 15:30:27] crowsnest:		Interval: Discrete 0.033s (30.000 fps)
[03/15/25 15:30:27] crowsnest:		[1]: 'YUYV' (YUYV 4:2:2)
[03/15/25 15:30:27] crowsnest:		Size: Discrete 1280x720
[03/15/25 15:30:27] crowsnest:		Interval: Discrete 0.100s (10.000 fps)
[03/15/25 15:30:27] crowsnest:		Size: Discrete 1920x1080
[03/15/25 15:30:27] crowsnest:		Interval: Discrete 0.200s (5.000 fps)
[03/15/25 15:30:27] crowsnest:		Size: Discrete 640x480
[03/15/25 15:30:27] crowsnest:		Interval: Discrete 0.033s (30.000 fps)
[03/15/25 15:30:27] crowsnest: Supported Controls:
[03/15/25 15:30:27] crowsnest:		
[03/15/25 15:30:27] crowsnest:		User Controls
[03/15/25 15:30:27] crowsnest:		
[03/15/25 15:30:27] crowsnest:		brightness 0x00980900 (int)    : min=-64 max=64 step=1 default=0 value=0
[03/15/25 15:30:27] crowsnest:		contrast 0x00980901 (int)    : min=0 max=95 step=1 default=0 value=0
[03/15/25 15:30:27] crowsnest:		saturation 0x00980902 (int)    : min=0 max=100 step=1 default=80 value=80
[03/15/25 15:30:27] crowsnest:		hue 0x00980903 (int)    : min=-2000 max=2000 step=1 default=0 value=0
[03/15/25 15:30:27] crowsnest:		white_balance_automatic 0x0098090c (bool)   : default=1 value=1
[03/15/25 15:30:27] crowsnest:		gamma 0x00980910 (int)    : min=64 max=300 step=1 default=84 value=84
[03/15/25 15:30:27] crowsnest:		gain 0x00980913 (int)    : min=1 max=8 step=1 default=1 value=1
[03/15/25 15:30:27] crowsnest:		power_line_frequency 0x00980918 (menu)   : min=0 max=2 default=1 value=1 (50 Hz)
[03/15/25 15:30:27] crowsnest:		0: Disabled
[03/15/25 15:30:27] crowsnest:		1: 50 Hz
[03/15/25 15:30:27] crowsnest:		2: 60 Hz
[03/15/25 15:30:27] crowsnest:		white_balance_temperature 0x0098091a (int)    : min=2800 max=6500 step=1 default=3980 value=3980 flags=inactive
[03/15/25 15:30:27] crowsnest:		sharpness 0x0098091b (int)    : min=1 max=7 step=1 default=2 value=2
[03/15/25 15:30:27] crowsnest:		backlight_compensation 0x0098091c (int)    : min=0 max=128 step=0 default=0 value=0
[03/15/25 15:30:27] crowsnest:		
[03/15/25 15:30:27] crowsnest:		Camera Controls
[03/15/25 15:30:27] crowsnest:		
[03/15/25 15:30:27] crowsnest:		auto_exposure 0x009a0901 (menu)   : min=0 max=3 default=3 value=3 (Aperture Priority Mode)
[03/15/25 15:30:27] crowsnest:		1: Manual Mode
[03/15/25 15:30:27] crowsnest:		3: Aperture Priority Mode
[03/15/25 15:30:27] crowsnest:		exposure_time_absolute 0x009a0902 (int)    : min=10 max=626 step=1 default=156 value=156 flags=inactive
[03/15/25 15:30:27] crowsnest: Try to start configured Cams / Services...
[03/15/25 15:30:29] crowsnest: INFO: Configuration of Section [cam 1] looks good. Continue ...
[03/15/25 15:30:30] crowsnest: V4L2 Control: No parameters set for [cam 1]. Skipped.
[03/15/25 15:30:33] crowsnest: Starting ustreamer with Device usb-SYX-230524-J_HD_Camera-video-index0 ...
[03/15/25 15:30:34] crowsnest:  ... Done!
[03/15/25 15:30:35] crowsnest: WATCHDOG: Lost Device: 'usb-SYX-230524-J_HD_Camera-video-index0'

dleffler, 2025-03-15:
I'm no expert, but I think the device name needs to be the entire path according to the wiki here.
```
device: /dev/v4l/by-id/usb-SYX-230524-J_HD_Camera-video-index0

alexkipar, 2025-03-15:
@dleffler thanks, changing this, and restart finally enabled camera

closed 2025-03-15 by alexkipar