Phil1988/FreeDi #197: Error when updating from 1.03 to 1.42

hypdal, 2025-03-15:
Just a little hint, when updating from 1.03 to version 1.42 I noticed that the file FreeDi.service is located in the helpers directory. So the command “sudo cp ${FREEDI_LCD_DIR}/FreeDi.service /etc/systemd/system/FreeDi.service” and the following in the update.sh script fails:

...
Installing the service to starts this program automatically at boot time...
Moving new FreeDi.service to /etc/systemd/system/
cp: cannot stat '/home/mks/FreeDi/FreeDiLCD/FreeDi.service': No such file or directory
FreeDi.service moved to /etc/systemd/system/
Setting user to mks in FreeDi.service
sed: can't read /etc/systemd/system/FreeDi.service: No such file or directory
Reloading systemd manager configuration...
systemd manager configuration reloaded!
Enabling FreeDi.service to start at boot...
Failed to enable unit: Unit file FreeDi.service does not exist.
FreeDi.service enabled to start at boot!
Starting FreeDi.service...
Failed to start FreeDi.service: Unit FreeDi.service not found.
FreeDiLCD.service started!
...

I downloaded the file manually, replaced the {{USER}}, registered the service and started it. The LCD was then reflashed as described. Everything works fine for me, as far as I have tested it...

Phil1988, 2025-03-15:
How did you update?
You can't simply go from 1.03 to 1.42.

hypdal, 2025-03-16:
Hmm, 

I used the instructions at https://github.com/Phil1988/FreeDi/wiki/Updating-FreeDi and compared the macros.cfg and moonraker.conf after the display was running and updated them if necessary.

Is there anything else I need to consider?

Phil1988, 2025-03-16:
No, but you just remembered me, that I need to update the instructions and the script :D

closed 2025-03-16 by Phil1988