Phil1988/FreeDi #198: FreeDi Logo Contest

Phil1988, 2025-03-16:
# 🎨 FreeDi Logo Contest – Calling All Creatives!  

I am developing the next version of FreeDi, and I would like to have a **nice logo** for it.  

While I was creating one myself, I thought:  
*"Why don’t you involve the community? There are surely some creative and talented people who can do this just as well - if not better!"*
And I also think this can be fun 😊

**Everyone can participate** (and I’ll probably submit a design myself 😉).  

## How It Works:  
1. **Design one or multiple logos** for FreeDi.  
2. **Submit your designs** as a comment or link in this thread. 
3. **End of this event*** will be **March 30, 2025**
4. **Community Vote**: On **March 31, 2025**, a poll will be posted in the **Discussions** section, where the community will choose the best design. 

### Important:  
By submitting a logo, you **grant FreeDi the unrestricted right** to use, modify, and distribute the design without limitations.  

Let’s make FreeDi’s identity unique together! 
I'm looking forward to see your amazing designs! 🚀

chrashing, 2025-03-17:
I asked Google Gemini AI to 'sketch of a bird being printed on a 3D printer', and then added print FreeDI underneath. It didn't get the lettering case right, but did a great job for a start.

[image]

leckert123, 2025-03-17:
Some further inspiration

[image]

[image]

[image]

[image]

pmbroth, 2025-03-17:
[image]

[image]

[image]

Lucianius77, 2025-03-17:
My quick contribution - non KI. 
If you like it, I could refine or adapt it and offer SVG  :)

[image]

Phil1988, 2025-03-17:
Wow! I didnt expected such a lively participation.
First of all, **a huge thank you** to everyone who has already submitted a logo! 🙌  
Some of the ideas look absolutely fantastic, and I really appreciate the effort and creativity you're putting into this.

### A little note on the design direction and I feel like I missed to be more precise: 
**I personally think the logo should be simple and clean so that it looks great on displays** and it should really only be a logo.
The FreeDi text will be added as plain text.
Some submissions are already heading in that direction, and I’d love to see more of that!

Also, I’ve noticed that we're truly living in the **age of AI** - some logos clearly have that "AI magic" going on.  
...and I also noticed FreeDi users seems to be bird lovers 😜😄

But remember, **quality over quantity**. 
It's better to have one solid design than five rushed ones. 
Keep up the great work!  

Now, it’s time for **my own contribution**! 
(And nope, no AI was involved here. Just good ancient human creativity 😉)  

<p align="center">
  [image]
</p>

Looking forward to more amazing designs! 🚀

leckert123, 2025-03-17:
For the record, I was just looking for imagery that suggested "freedom" to spur those with greater ability to create a logo.  It seems that freedom is often associated with birds flying free, or with chains / handcuffs / shackles being removed.

No AI was used in at least what I submitted, Phil. I literally used Google Images and looked for things not copyrighted and free to use or modify.

pmbroth, 2025-03-17:
[image]

[image]

Lucianius77, 2025-03-17:
another suggestion, less playful, clear and simple - as requested standard font ;-)

[image]
[image]

breathless19, 2025-03-18:
> 
> ![FreeDi Logo Submission](https://private-user-images.githubusercontent.com/6054234/423626398-18f293e1-fb6a-4323-a4b8-014e73f02d51.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDIzMzY5NDksIm5iZiI6MTc0MjMzNjY0OSwicGF0aCI6Ii82MDU0MjM0LzQyMzYyNjM5OC0xOGYyOTNlMS1mYjZhLTQzMjMtYTRiOC0wMTRlNzNmMDJkNTEucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDMxOCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTAzMThUMjIyNDA5WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZWYzN2RlMDIyYjQyMTcwZTFjNzhlNGE0M2FiOTU2NTMxM2YwNmZiY2RiZDM5ZTk4MmU0ZDAzZDYxOTNmZWNiNCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.iFq6nz5_-x6FTjo4M2XB84jYSXyPetwrxZr9zs5z-_s)
> 
> on screen:
> 
> ![FreeDi Logo Submission](https://private-user-images.githubusercontent.com/6054234/423633105-cda2c3fa-7afd-4337-a946-91c7effc0f3d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDIzMzY5NDksIm5iZiI6MTc0MjMzNjY0OSwicGF0aCI6Ii82MDU0MjM0LzQyMzYzMzEwNS1jZGEyYzNmYS03YWZkLTQzMzctYTk0Ni05MWM3ZWZmYzBmM2QucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDMxOCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTAzMThUMjIyNDA5WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NTdmYzAwMmUzOTQ3MjdhZjNmMGRhMDUxY2E1NDIyNDc1NWJkZmE2ZDllYzQ2YWEwNzI5ZTIwNjBkZGY2YTRiMCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.zIyLu3MrFx6gMXI5HXEvVLAQ-j4ym9uPeqkKz54IBsA)
> 
> Looking forward to more amazing designs! 🚀

So far, I'm liking these the best, though I think it would look better if the "extruder" here were made narrower and slightly reshaped to make the leftmost portion look more like an "F", and the rightmost portion look more like a "D".    They already sort of do, but it would look better in my view if it were more noticeable.

Lucianius77, 2025-03-19:
> So far, I'm liking these the best, though I think it would look better if the "extruder" here were made narrower and slightly reshaped to make the leftmost portion look more like an "F", and the [...]
---
A nice idea to make a combination of letters and tool head. However, I personally miss the aspect of "liberation" compared to Qidi Stock.

SmileyMan, 2025-03-20:
[image]

Bonnyfede00, 2025-03-24:
Created with ChatGPT
[image]

Phil1988, 2025-03-30:
Thanks @chrashing @leckert123 @pmbroth @Lucianius77 @SmileyMan for your contributions.
Some of them are pretty nice, so thanks to every designer and all the AI's involved 😊

Lets start the voting for the logo here:
https://github.com/Phil1988/FreeDi/discussions/211

closed 2025-04-02 by Phil1988

Phil1988, 2025-04-08:
> Unfortunately, I didn't discover this till this week. But I'd love to throw this one out there.

Thanks for your logo :)
It was too late, but I appreciate that you took your time to provide your suggestion :)