Phil1988/FreeDi #204: [Bug] Case Light button on screen display doesn't work if value is not 0 or 1

savardd, 2025-03-19:
**Describe the bug**
If we use a dimmed value for the case light output_pin, the case light toggle button on the screen doesn't work. 

**Expected behavior**
If the light is ON (even if dimmed), it should be possible to turn it OFF via the case light button on the display.
If it is off, it is OK to turn it ON at max power instead of last dimmed value.

**To Reproduce**
I am using this section in my printer.cfg to be able to dim the case light via Mainsail:
```
## caselight LEDs (dimmable)
[output_pin caselight]
pin: PC7
pwm: true
cycle_time: 0.0100
shutdown_value:0
value:1
```

For reference, here is the original section:
```
## caselight LEDs
[output_pin caselight]
pin: PC7
pwm: false
shutdown_value:0
value:1
```

Steps to reproduce the behavior:
1. Using the output_pin configuration for dimmable caselight
2. Choose a value greater than 0 but lower than 1 from the Mailsail slider, for sample, choose 0.3 (30% dimmed)
3. Try to toggle the case light via the display button on the screen
4. The button stay same and doesn't turn off the light

If I choose 0 (OFF) or 1 (ON full light) as the value from the Mailsail slider, the case light button works.  

**Base (please complete the following information):**
 - Printer model: X-Smart 3
 -  FreeDI 1.42
 - Mainboard: stock Mainboard X-Smart 3

Phil1988, 2025-03-19:
It's not a bug as the light is not dimmable.
Nor on the stock system, neither on FreeDi.

This would be a feature request.

Phil1988, 2025-03-19:
It's now implemented and will be released with the next version 😉