Phil1988/FreeDi #209: [Bug] mcu 'MKS_THR' configured for 12Mhz but running at 5MHZ!

Ramon-Santini, 2025-03-22:
**Describe the bug**
Klipper runtime warning of >mcu 'MKS_THR' configured for 12Mhz but running at 5MHZ!<

**Expected behavior**
Incorrect configuration of the FreeDI MKS_THR?? 

**To Reproduce**
Notification Bell showing "1" message. Message: >mcu 'MKS_THR' configured for 12Mhz but running at 5MHZ!<

Just received warning after various prints without issue(?)

**Base (please complete the following information):**
 - Printer model: X-Plus 3
 - mcu(stm32f401xc)
Version: v0.12.0-289-g08a1c9f1
 - mcu MKS_THR(rp2040)
Version: v0.12.0-289-g08a1c9f12
 - Host(aarch64, 64bit)
Version: v0.12.0-289-g08a1c9f12-dirty
OS: Armbian-unofficial 24.2.0-trunk bookworm
Distro: armbian
 - X3seriesLCD firmware version: [e.g. v1.00 stable]
 - Source: FreeDi image
 - Mainboard: stock Mainboard X-4
Software 
 - crowsnest
v4.1.12-1-g9623b5fe
 - fluidd (installed but using Mainsail)
v1.32.6 > v1.33.0
 - fluidd-config (installed but using Mainsail)
v1.2.1-0-g807175d7
 - FreeDi
v1.42-1-g1b6b6054
 - klipper
v0.12.0-289-g08a1c9f1
 - mainsail
v2.13.2
 - moonraker
v0.9.3-0-g71f9e677

Phil1988, 2025-03-23:
I currently dont have the time to have a deep insight of this.
I think it is related to the [fix](https://github.com/Phil1988/FreeDi/issues/165#issuecomment-2680969290) we had to apply for others loosing availability of the toolhead on certain toolhead boards.

Am I understanding it correctly, that your printer is working flawless without any issues and only drops this as a hint?

Ramon-Santini, 2025-03-23:
It did it the first time I brought up the printer yesterday to do some fiddling with the centering of the print bed. After looking up the code and writing this "Bug Report" I restarted and Update Manager said there was a "System" update for libxslt1.1 and that my FreeDI Repo is dirty, (which I can't seem to clear with a soft or hard recovery). and FLUIDD has been updated. Once all the updates were done...

mcu(stm32f401xc)
Version: v0.12.0-289-g08a1c9f1
Load: 0.01, Awake: 0.00, Freq: 84 MHz, Temp: 30°C
1
mcu MKS_THR(rp2040)
Version: v0.12.0-289-g08a1c9f12
Load: 0.01, Awake: 0.00, **Freq: 12 MHz,** Temp: 21°C
1
Host(aarch64, 64bit)
Version: v0.12.0-289-g08a1c9f12-dirty
OS: Armbian-unofficial 24.2.0-trunk bookworm
Distro: armbian
Load: 1.27, 

So I would call this intermittent as I can't reproduce it and the printer is working fine.

Phil1988, 2025-03-23:
Please can you explain what you did at the "fiddling" ? :D

FreeDi repo dirty is not really a problem, bit it leads to the assumption that something is not as it should be.
Can you expand the info icon to see whats the reason for the dirty repo?

You can also direct to FreeDi via terminal and check `git status`

Ramon-Santini, 2025-03-23:
"Fiddling" with the X & Y stepper settings in printer.cfg to see if I could get the plate to center, (It didn't), by changing the position_min and position_endstop parameters. It didn't work so I set them back to the original parameters. :P

>mks@mkspi:~/FreeDi$ git status
>On branch master
>Your branch is behind 'origin/master' by 1 commit, and can be fast-forwarded.
 > (use "git pull" to update your local branch)

>Changes not staged for commit:
 >(use "git add/rm <file>..." to update what will be committed)
 >(use "git restore <file>..." to discard changes in working directory)
 >     deleted:    FreeDiLCD/freedi_update.sh
>
>no changes added to commit (use "git add" and/or "git commit -a")

I did a "git pull" and now it says I am up to date with 'origin/master' and

mcu(stm32f401xc)
Version: v0.12.0-289-g08a1c9f1
Load: 0.01, Awake: 0.00, Freq: 84 MHz, Temp: 33°C
1
mcu MKS_THR(rp2040)
Version: v0.12.0-289-g08a1c9f12
Load: 0.01, Awake: 0.00, Freq: 12 MHz, Temp: 24°C
1
Host(aarch64, 64bit)
Version: v0.12.0-289-g08a1c9f12-dirty
OS: Armbian-unofficial 24.2.0-trunk bookworm
Distro: armbian
Load: 0.52, 

BUT...

[image]

rosenrot00, 2025-03-23:
> **Describe the bug** Klipper runtime warning of >mcu 'MKS_THR' configured for 12Mhz but running at 5MHZ!<
>
> **Expected behavior** Incorrect configuration of the FreeDI MKS_THR??
>
> **To Reproduce** [...]

What you describe has nothing to do with FreeDi as the mcu and mks_thr firmwares are just compiled using Klipper. The reported software timer frequency of usually 12MHz is untouched by FreeDi.

How often did that happen until now?

Ramon-Santini, 2025-03-23:
> How often did that happen until now? <

Just once.

rosenrot00, 2025-03-24:
> > How often did that happen until now? <
> 
> Just once.

Thanks for the information. I would say nothing to worry about until it happens more often.

In any case not a issue caused by FreeDi.

closed 2025-03-24 by Phil1988