Phil1988/FreeDi #212: [Help] stuck at Klipper reports error, no network connection

mlevenston1, 2025-04-01:
I'm attempting to install 1.40 using the rar file, from a USB drive. Everything **seemed** to be working, the display flashed properly, but I'm stuck at the Klipper reports error screen because my machine isn't connecting to the network.  Before installation, I verified that I could connect via ethernet connection from my PC (Fluidd interface and ssh from terminal), but the mks board is not connecting to the internet so I can't ssh to complete the process. I've scanned the network (arp-a from terminal, plus others) and the device isn't connected.

I used USB because I don't currently have an EMMC reader; am I stuck until one arrives next week and I can re-flash?  Any ideas?

I've rebooted a half dozen or more times, no difference.

Thanks.

Marc

mlevenston1, 2025-04-01:
PS: X-Plus 3

Phil1988, 2025-04-01:
You "used USB" in what matter?
Tried the danger method or trying to boot from USB?
How exactly did flash the display?

... so many questions :)

mlevenston1, 2025-04-01:
Tried to boot from USB-- in the external (top) USB port.  Slow, but seemed to go ok. Display eventually switched to white w/ black text, slowly ramped from 0% to 100%, then brought up boot screen.  After a while, it began "Connecting to Klipper."  I have to let it sit for about 20 minutes, then it eventually gives me the Klipper reports error screen.  The touchscreen is responsive (I can restart Klipper/firmware, reboot the system, etc.).  

I have the ethernet cable from the printer connected to my PC LAN port, but the lights aren't flashing (indicating no connection). It appears that the mks board can't access the network. Before flashing, I had it connected this way, checked the hardwired IP address on the Qidi network screen, and was able to connect to Fluidd using that IP address (as well as the normal wifi IP address) and also was able to ssh into the mkspi from the terminal.

Phil1988, 2025-04-02:
Then you simply have to get the printers IP address.
Your PC will give random IP addresses as its not a real DHCP server.
Check this and you will get access to the printer.
Its only a network issue on your side :)

mlevenston1, 2025-04-02:
Sorry if I'm being dense-- I've tried various ways to aaaign an address.  Network bridge, assigning static ip to Ethernet, using a rpi as a bridge, assigning up to eth0 on rpi.  Nothing works, it seems like the board in the Qidi isn't requesting or recognizing a network connection.

Am I missing something?

MarkusKgit, 2025-04-02:
Easiest option would be to connect the printer to a router with DHCP enabled. As ours at work has it disabled and all devices need to enter manual IP (don't ask...) I had to do a little work around and quickly found that its a bit tricky with Windows and DHCP. In the end I used this [program](https://www.hanewin.net/dhcp-e.htm) (just download the second file called Application, you don't need the service)

First assign your network card on the PC a static IP like 192.168.1.1 and then launch this program after you unzipped it (dhcpsrv-64.exe)
Then go to Options-Manage Profiles and add a new profile with a name of your liking. Choose your network interface and fill the settings for IP range etc like this 

[image]

Theoretically your printer should now connect and the IP will be shown in the list. It wasn't working for me initially because my Antivirus was blocking the port so be aware.

After you SSH into it you can do `sudo armbian-config` and then set a permanent IP in the Network settings.

mlevenston1, 2025-04-02:
Thanks-- much appreciated.  Will try this tonight.

Phil1988, 2025-04-02:
I would be carefull with settin a permanent IP as it could collide with other devices and result in an unreachable printer.
If the PC directly connected you can also do a cmd batch.
I dont know what exactly I used but something like this (depending on your system language) will work:
```
for /L %i in (1,1,254) do @ping -n 1 -w 1 192.168.1.%i | find "Reply from" > nul && echo 192.168.1.%i is reachable
```

mlevenston1, 2025-04-04:
Thanks-- this worked and I was able to finish setup, although the mkspi still couldn't access the outside world for updates. I gave up and moved a wifi node to the printer so I could directly plug in ethernet.  Although I could update everything, I couldn't get the wifi setup to work using the script, so I had to download and install a driver manually.  Everything is fully updated and working great now.

Phil1988, 2025-04-05:
Do you use a stock wifi module?
To the date I am only aware of 2 different modules from Qidi and both are supported.
Maybe you have another one.

Can you expand and describe the wifi installation process you did?
This way I could integrate it to install it for others that might be in the same position.

Phil1988, 2025-04-08:
So is this now working?
I cant implement auto driver installations for every wifi module.
So if you have a custom module, I kindly ask you to install the driver on your own and check if you can connect to your wifi using NetworkManager/nmcli etc.

mlevenston1, 2025-04-08:
> So is this now working for you?

I didn’t expect you to change the script, just telling you how I got it to work.  Maybe put a note in the instructions that if the wifi module has been changed it would be better to look up how to get the driver before installing FreeDi?  Wasn’t a problem for me once I got the ethernet issue straightened out.

Marc

Phil1988, 2025-04-09:
I'm glad you could solve it and its working now for you :)
Custom hardware modifications often require adjustments from the user. 
Thank you for the suggestion; I'll consider adding a note in the instructions.

closed 2025-04-09 by Phil1988