Phil1988/FreeDi #214: [Bug] mcu 'MKS_THR': Unable to open serial port

martin-badin, 2025-04-05:
**Describe the bug**
Toolhead is not recognise. RP2040 is not in the list of connected devices. Sometimes it works, sometimes not.

**Expected behavior**
Toolhead should be recognise. RP2040 should be in the list of connected devices. It should works everytime.

**To Reproduce**
Steps to reproduce the behavior:
1. Turn on printer

**Screenshots or picture**

[image]

[image]

[image]

[image]

**Console output**
```
Connecting to Moonraker at ws://127.0.0.1:80/websocket
Using selector: EpollSelector
TFT baudrate detection...Websocket connected
Moonraker connection opened at ws://127.0.0.1:80/websocket
G-Code file path is: ~/printer_data/gcodes
Found z_offset = 0.0 in variable "z_offset" in section "probe"
Detected baudrate: 921600
Display response: comok 1,1281-0,TJC8048X250_011R,165,10201,E9D83401A80FEE65,33554432-0
Display firmware: v1.42
Printer model: x-max3
Printer volume: 328x326x322
Klipper version: v0.12.0-289-g08a1c9f12-dirty
Software version: v1.41 stable
FreeDiLCD start
FreeDi sw-version sent to screen!
pushing data to tft. Last update: (0.99 sec)
data.klipper_state != self.printer.klipper_state
klippy error
pushing data to tft. Last update: (0.2 sec)
_NewConnectionLostPage: len: 1 data[0]: ee
ActiveScreen: = ConnectionLostPage
```

**Base (please complete the following information):**
 - Printer model: X-Max 3
 - X3seriesLCD firmware version: v1.42 stable
 - Source: FreeDi image
 - Mainboard: stock Mainboard X-4

[klippy.log](https://github.com/user-attachments/files/19613816/klippy.log)

Phil1988, 2025-04-05:
Sometimes it finds your toolhead?
If yes - its likely a hardware failure.

If it only finds the toolhead after flashing klipper to it - and after a power cycle it is persistently not availible then your might check this toolhead firmware and flash it:
https://github.com/Phil1988/FreeDi/blob/master/mainboard_and_toolhead_firmwares/v0.12.0-289/Toolhead_RP2040.uf2

boydenus, 2025-04-05:
Sounds like an issue with your USB-C cable that connects the main board to the toolhead board. It's a common issue with QIDI printers that QIDI issued replacement cables for.

-Tim

martin-badin, 2025-04-05:
> Sounds like an issue with your USB-C cable that connects the main board to the toolhead board. It's a common issue with QIDI printers that QIDI issued replacement cables for.
> 
> -Tim

Thank you. I haven’t had this issue before. This issue started after upgrading to FreeDi. So I will try to contact Qidi support. Thank you

martin-badin, 2025-04-05:
> Sometimes it finds your toolhead? If yes - its likely a hardware failure.
>
> If it only finds the toolhead after flashing klipper to it - and after a power cycle it is persistently not availible then [...]

Thank you. I will try to reflash toolhead.

martin-badin, 2025-04-06:
> Thank you. I will try to reflash toolhead.

I did it. It seems it helps. Currently I see different version on toolhead but I am still in the testing phase. See attached screenshot. So each part of printer has the same klipper version now.

[image]

martin-badin, 2025-04-06:
@Phil1988 another question. Why am I see dirty version on Host? Is it ok, or It should be better to reset that version?

Phil1988, 2025-04-08:
> Currently I see different version on toolhead 

On your screenshot I do see the same versions v0.12.0-289 for all your devices.

The dirty host is nothing to worry about - its something from this release of klipper and git but I dont want to go into detail as you dont have to deal with it ;)

I am closing this as its a dublicate https://github.com/Phil1988/FreeDi/issues/165 and solved.
Happy printing!

closed 2025-04-08 by Phil1988