Phil1988/FreeDi #215: [Bug] Z-Offset - X-Max 3 - FreeDi 1.4.2

boydenus, 2025-04-05:
This may be a Klipper issue but I am not sure, so throwing it out there to see what you all think since you've been into the guts of the system.

I have FreeDi installed, updated to 1.4.2, calibrated, and have done a first layer calibration test (or a series of them per this bug). So, everything is functional. When I was doing the first layer calibration, I tried to set the z-offset both through the panel, in Main Sail, and in printer.cfg in this section of the configuration:

`
#####################################################################
# Probe
#####################################################################
#!# Pay attention to this section and compare it with your stock backup config
[probe]
pin: ^!MKS_THR:gpio21
x_offset: 28
y_offset: 4.4
z_offset: -1.80
speed: 5
samples: 2
samples_result: average
sample_retract_dist: 3
samples_tolerance: 0.08
samples_tolerance_retries:3
`
The printer seems to ignore any of those values. 

If I babystep during the print it will adjust the nozzle height. If I set the z-offset in the slicer under printer settings, that will adjust the nozzle height (which I am currently using for my prints).

What I notice that is different now is that in Mainsail during a print, the current z-offset is no longer visible in the Z-Offset section. It only shows 0.000 until you babystep it. With the stock firmware, it would show the current Z-Offset value and let you step from there. I don't know if this related, the cause, or other. Unfortunately, I don't have a standard Klipper printer to compare against, so not something I can easily troubleshoot other than the work-arounds I have already found.

Thanks,

Tim

baderas, 2025-04-06:
Can also confirm that on my X-Smart 3, I just flashed FreeDI 1.4.2 (was on stock Qidi firmware before, so I have no experience how is was on FreeDi < 1.4.2) and my X-Smart 3 is unusable now due to the z-offset being ignored or somehow calculated wrong. 

Also compared to stock software, the x/y position of the z-calibration is off and not in the bed center.

Phil1988, 2025-04-08:
Sorry for the delay - I am developing the new release and didnt check it here.

I need more information to help you with this like:
1. How exactly do you adjust and save the new offset?
2. Can you see the new (verified it is indeed new?) offset in the auto generated section at the end of your printer.cfg?
3. Do you leave it as it is or do you do any modifications at the printer.cfg when doing so?
4. Is this new value still persistent after a reboot
5. Have you done any modifications to your slicers start GCODE section?
6. Do you have any non stock hardware installed (especially beacon/cartographer)?
This way I know what you are doing, if the value is properly changed and if the value is persistent.
These information being important to track the cause if this issue and solve it :)

I am also not sure if it is a user error or a bug, because what @breathless19 describes is the total normal way how klipper works and how new values to the printer.cfg has to be applied.

I dont recommend workarounds.
I have seen a user setting the offset in the slicers start gcode to a set z-offset value (because he didnt knew better) and then adding the z-offset delta to this config like @boydenus does.
I mean it is your printer and you are free to do what you like, but I would rather fix it properly :)

About the macro and SAVE_CONFIG :
A probe calibration via Mainsail does not automatically do a `SAVE_CONFIG` and this is for reasons (testing/tinkering/verification...) I like to keep. 
So I did not include the SAVE_CONFIG to that macro.
When you are doing it via Mainsail, the CONSOLE does report back, that you need to perform a SAVE_CONFIG or click the popping up button at the top to make this new value persistent. You might have overlooked it @breathless19.
When you do a probe calibration via the touch display, it is doing a `SAVE_CONFIG` at the end as you dont have any other option to tinker with the machine and likely just want to calibrate. If that is not the case, then this is a bug on my side.

General mechanics of klipper:
I dont know who "Grok" is and I want to put some wordings right.
The way that @breathless19 describes is normal for klipper.
The bottom of the printer.cfg is not a SAVE_CONFIG section, but the auto-generated-section. 
SAVE_CONFIG is just the command to calculate and save the current config and restarting klipper.
Reason is, that klipper reads the values from your config once at starting. 
When you modify values, they dont apply untill you restart klipper.
Additionally, in some cases (eg. while printing) modifying the printer.cfg cant save the new value directly (restarting klipper at this point will cause the print to fail). 
In these cases it will only temporarly save the new values and you have to perform a SAVE_CONFIG after the print finished.
This is all covered by FreeDi. 
By the end of a print, a button will pop up in case you have adjusted (babystepping) the z-offset to save the new value.

boydenus, 2025-04-08:
So, not to confuse what issue/issues you are trying to address, and because I am the owner of the original request, let's keep the analysis to my specifics. :-)

First attempt at setting the z-offset was via the panel, just as if you were setting up the printer for the first time. It runs through the basic "paper test" and says it saved my z-offset value. Klipper restarts and a check of printer.cfg shows the z-offset value under probe is commented out and the commented section at the end shows the panel-based z-offset. However, when I go to print, Mainsail reports in the z-offset section that the z-offset is zero and indeed my print nozzle is at least 2mm above the bed, resulting in no adhesion and the start of a blob of death.

Second attempt was setting it through Mainsail. I removed the values from the probe section and the commented section at the bottom of the printer.cfg file. I saved the config and restarted Klipper. Then I used the interface in Mainsail to do the same z-offset "paper test" to set the z-offset. Again, printer.cfg was updated with the z-offset at the bottom of the printer.cfg file and I did a manual save of the configuration and restart of Klipper. I did a test print again and had the same result, no offset showing in the Z-Offset section of Mainsail and my printer's nozzle was again at least 2mm above the print bed and no adhesion and the start of a blob of death.

My test print is a simple 100x100x0.4 cube, essentially a two-layer first layer test.

I again removed the z-offset settings, saved the config, and restarted Klipper. This is when I started trying work-arounds and set the z-offset in the slicer (Orca) under printer settings. I am using the standard Qidi start and end g-code and my z-offset change is not in the start and end g-code. The printer settings z-offset adds/subtracts the specified z-offset from all the g-code commands when the g-code file is generated, and the file is sent to the printer already adjusted. This works because the printer's offset is set at the default of 0.

Restarting the printer has no effect on the process because the z-offset value is being saved to the config file, but the printer is ignoring the value.

Phil1988, 2025-04-09:
>  the z-offset value under probe is commented out and the commented section at the end shows the panel-based z-offset. However, when I go to print, Mainsail reports in the z-offset section that the z-offset is zero and indeed my print nozzle is at least 2mm above the bed

This leads to my assumption that there is something else wrong with your specific setup:
Its commented out at the [probe] section and added to the auto generated section at the end: This is correct.
Mainsail reports z-offset is 0.0: This is correct. It shows the temporarly relative offset based from your config (which is 0)
The nozzle is 2mm aboce the bed: This is not correct.
Please attach your printer.cfg, the gcode you are printing and the start-gcode section at the current state.

baderas, 2025-04-09:
I can only confirm the experienced behavior that @boydenus describes. 
On my X-Smart 3 it could only be the weirdy placed z-offset measurement (not centered), but you can basically enter any given z-offset value, it seems to be just ignored. Although its writte to the autoconfig section of printer.cfg.

baderas, 2025-04-09:
To answer you questions:
```
   1.How exactly do you adjust and save the new offset?
   2. Can you see the new (verified it is indeed new?) offset in the auto generated section at the end of your printer.cfg?
   3. Do you leave it as it is or do you do any modifications at the printer.cfg when doing so?
   4. Is this new value still persistent after a reboot
   5. Have you done any modifications to your slicers start GCODE section? (please show me what you have there)
   6. Do you have any non stock hardware installed (especially beacon/cartographer)?
```

1. I tried it via Display and via Mainsail Macro. With the Displayversion it always starts around 6,9, I bring it down to 0,1 - 0,2 until the paper has light friction.  Position of the z-offset measurement is strange: center of the right edge, sensor over the edge of the bed
2. Yes, but a different value than I saw on the display, e.g. it begins at 6,9 I set it to 0,2 and in printer.cfg it sais 1,9 (could not figure out how that translates) - I would however assume a negative value in printer.cfg
3. No special modification
4. yes
5. Just the defaults of orcaslicer:
```
PRINT_START
G28
G0 Z50 F600
M190 S[hot_plate_temp_initial_layer]
G28 Z
G29; mesh bed leveling ,comment this code to close it
G0 X0 Y0 Z50 F6000

M109 S[nozzle_temperature_initial_layer]
M106 P3 S255
M83
G4 P3000
G0 X{max((min(print_bed_max[0], first_layer_print_min[0] + 80) - 85),0)} Y{max((min(print_bed_max[1], first_layer_print_min[1] + 80) - 85),0)} Z5 F6000
G0 Z[initial_layer_print_height] F600
G1 E3 F1800
G1 X{(min(print_bed_max[0], first_layer_print_min[0] + 80))} E{85 * 0.5 * initial_layer_print_height * nozzle_diameter[0]} F3000
G1 Y{max((min(print_bed_max[1], first_layer_print_min[1] + 80) - 85),0) + 2} E{2 * 0.5 * initial_layer_print_height * nozzle_diameter[0]} F3000
G1 X{max((min(print_bed_max[0], first_layer_print_min[0] + 80) - 85),0)} E{85 * 0.5 * initial_layer_print_height * nozzle_diameter[0]} F3000
G1 Y{max((min(print_bed_max[1], first_layer_print_min[1] + 80) - 85),0) + 85} E{83 * 0.5 * initial_layer_print_height * nozzle_diameter[0]} F3000
G1 X{max((min(print_bed_max[0], first_layer_print_min[0] + 80) - 85),0) + 2} E{2 * 0.5 * initial_layer_print_height * nozzle_diameter[0]} F3000
G1 Y{max((min(print_bed_max[1], first_layer_print_min[1] + 80) - 85),0) + 3} E{82 * 0.5 * initial_layer_print_height * nozzle_diameter[0]} F3000
G1 X{max((min(print_bed_max[0], first_layer_print_min[0] + 80) - 85),0) + 12} E{-10 * 0.5 * initial_layer_print_height * nozzle_diameter[0]} F3000
G1 E{10 * 0.5 * initial_layer_print_height * nozzle_diameter[0]} F3000
```
6. no, stock QIDI X-Smart 3 with QIDI wecbam - worked flawless on stock fw. First FreeDi Version was 1.40 which was directly updated to 1.42 (due to 1.42 not having an image on Github)

baderas, 2025-04-09:
Details z-calibration workflow:

1. start z-calibration via Display:
[image]
[image]

2. finished calibration:

[image]

3. Values in printer.cfg
```
[probe]
pin: ^!MKS_THR:gpio21
x_offset:27
y_offset:3.5
#z_offset: 0.0

....

#*# <---------------------- SAVE_CONFIG ---------------------->
#*# DO NOT EDIT THIS BLOCK OR BELOW. The contents are auto-generated.
#*#
#*# [probe]
#*# z_offset = 1.500
```

martin-badin, 2025-04-10:
@Phil1988 Z-offset works for me, but I don't see the default value from the config on the screen. There is still 0.000. I would expect the number from the config to be displayed on the screen.

baderas, 2025-04-10:
When using Mainsail the result in printer.cfg is close to the same: #*# z_offset = 1.600

The position is only center on the bed because I use in [gcode_macro G28]:
```
{% set home_all_final_position_y = printer.toolhead.axis_maximum.y / 2 %}
```
Otherwise it would be set behind the center with:
```
{% set home_all_final_position_y = printer.toolhead.axis_maximum.y - 30 %}
```

With babystepping the offset is around 0.4 to 0.5 off --> with around 2.0 it is working.
The difference seems quite high to me.

boydenus, 2025-04-11:
> [@boydenus](https://github.com/boydenus) I appreciate your response, but I believe there may be a misunderstanding...

No, we are good. Just been busy with work this week and haven't gotten back to this project yet. Probably won't until Saturday. Will upload requested files then.

-Tim

Phil1988, 2025-04-11:
> With babystepping the offset is around 0.4 to 0.5 off --> with around 2.0 it is working. The difference seems quite high to me.

Yes its quite high, but not important, if babystepping adjusts it correctly and the z-offset is right for you.

So right now, only @boydenus is affected by this.
Lets wait and see what information he give and what the cause might be.

boydenus, 2025-04-13:
I am going to go ahead and close this issue based on further testing I did this morning. Part of the issue is variable names in the Start g-code that I copied from Qidi Slicer into Orca. The Orca slicer default profile for the Qidi X-Max 3 uses different variable names. Based on my testing the answer is the Klipper z-offset is only part of the z-offset configuration. It relies on code in the Start g-code in the printer profile in the slicer to fully implement the z-offset setting. That's why my setting the z-offset under Printer Settings in the slicer works, because it is just another way vs dynamic adjustment of z-offset via the Start g-code, to achieve the same result. So, it isn't that Klipper isn't saving the z-offset, it just isn't the only piece that determines z-offset. The dynamic z-offset wasn't being set in the Start g-code due to differences in the variable names between Orca and Qidi Slicer.

closed 2025-04-13 by boydenus

baderas, 2025-04-21:
> The dynamic z-offset wasn't being set in the Start g-code due to differences in the variable names between Orca and Qidi  Slicer.

Can you explain a bit more what you did or how that can be fixed?
I have to fix the dynamic z-offset in my x-smart 3D after every boot, it seems to be always changing (sometime its 1.6, then 2.0, and so on). Despite that, the value itself in printer.cfg changes sometimes after a reboot without any reason.

There seems to be some deeper issue within FreeDi or at my setup - I'm trying to figure out why its always changing with every reboot.

D3FCONau, 2025-05-31:
THE ISSUE WAS DRIVING ME CRAZY, 

I printed this to babystep it on the screen to get it right https://www.printables.com/model/114982-rectangular-spiral-z-offsetbed-level-test/files#preview.kAv2k

dialed it in over several goes using the front screen / settings and up down z whilst printing. Then  clicking save in fluid in the z offest section after each print