Phil1988/FreeDi #216: Calling my Alpha-Squad 2.0

Phil1988, 2025-04-10:
Hey everyone! 👋

I’m currently developing **FreeDi v2.00**, and I’m thrilled to let you know that it's packed with **a lot** of new features and various improvements have been made. 
However, the extensive changes are taking longer than I anticipated, and I could really use your help to speed things up! ⏩

Many of you have already shown great expertise and a willingness to support this project (thanks for that again).
That’s why I’m reaching out to you to lighten my load and accelerate the development of the next major version. 🚀

## What’s in it for you?
- **Early access** to the latest version (complete - with new bugs! 😉)
- Possibly a working software with new features - or maybe just a ~~mess~~ work in progress! 🛠️ (it really can be a wonky alpha for some machines)

## Requirements:
- You need **Discord** for direct communication, so we can keep things moving quickly.
- Familiarity and experience with the **terminal**, **SFTP**, and **Linux**.
- Knowledge and experience with **Klipper** (I can't teach/cover the basics).
- **Patience!** I’ll do my best, but I’m not always available 24/7 or have immediate solutions for everything.
- No **urgent printing needs!** Interrupting tests mid-way or reverting everything for printing purposes costs time, which is the opposite of why I’m making this call. 😊
- The ability to restore your printer to a functional state (I might not be able to help with urgent repairs, as I need to avoid spending too much time on individual issues).
- Your willingness to test **everything thoroughly** and summarize results clearly. Many short messages jumping between topics confuse me more than they help!

Please only check in if these requirements are met.
I understand that you would like to test the latest version, but it doesnt help anyone if I'm getting blocked by helping on individual issues. 

## Who am I looking for?
- **Test Stage 1:** Users with an **X-Plus 3** or **X-Max 3** (about 2 testers)
Current state: Slots open.
- **Test Stage 2:** Users with an **X-Smart 3** (about 2 testers)
Current state: Slots open.
- **Test Stage 3:** Users with a **Plus4** (about 2 testers)
Current state: Slots open.
- **Test Stage 4:** Users with a **Q1 Pro** (about 2 testers)
Current state: Slots open.

I’ll be on vacation from **April 13th to April 20th**. There will be some testing before, maybe during, and definitely after this period. My availability until April 20th is still uncertain.

Please refrain from spreading any **leaks** during the testing period, as I’m not sure which features will make it to the stable version and I don’t want to raise any false expectations! 🙈

Thanks so much for considering this! 
Looking forward to working with you! 💪

If you are interessted, write here or directly via discord :)

daja63, 2025-04-10:
Hi Phil1988, I am a seasoned X-MAX 3 user, hardware/software tech person and proficient in problems solving, I repair electronics and have knowledge in Linux, networking and some basic programming skills. Let me know if I can help with the alpha tests. Thanks for the work you do.

DragonzaS, 2025-04-10:
Already wrote in discord channel, but write here also

I can help with alpha testing for Q1 Pro, my machine is free for that

Phil1988, 2025-04-10:
@DragonzaS @IAmSHOCK please DM me in discord so I know who you are there ^^

pmbroth, 2025-04-10:
Anything, you need, I cam test.  I'm not an expert, buit can test as needed. Brian

billkenney, 2025-04-11:
I'd be happy to test it out on my max 3. I just got the cartographer working perfectly and could use another part-time project. I also ordered a bunch of filament before the tariffs went into effect so I have a lot of printing to do...

Phil1988, 2025-04-11:
Thank you all very much for your great interest and your willingness to support 😊👍

Currently, there is only one slot available for X-Smart 3 and one slot for X-Plus 3.

Phil1988, 2025-04-12:
@billkenney Please get in touch with me via discord :)

Phil1988, 2025-04-12:
@billkenney it says you dont accept invitations and DMs, so please add me :)
coco.33

billkenney, 2025-04-12:
> @billkenney it says you dont accept invitations and DMs, so please add me :)
> coco.33

That's what I searched for last night, but nothing came up. I just sent you a friend request. Sorry I don't use discord all that much...

jamincollins, 2025-04-18:
🤚 I'll take an XS3 slot

Phil1988, 2025-04-22:
> 🤚 I'll take an XS3 slot

Cool!
Please contact me via discors, so I can send you something ;)

pmbroth, 2025-05-04:
Just curious,  how is the alpha test going?

Phil1988, 2025-05-05:
It's going pretty decent now.
Having trouble with WiFi and flashing the Mainboard MCU.
Besides that it's going smooth at the moment.
Only updated/working/optimized configs under construction.

Developing on April was rough though.
Many things were harder and more complex then I hoped.

roo0991, 2025-05-07:
Hey mate just wanna see how it's going with the q1 part so far? Been very interesting on this 

Help to be a tester if you need extra help

pmbroth, 2025-05-09:
When will you be releasing a sneak peek for what the new release will contain?

Phil1988, 2025-05-11:
> When will you be releasing a sneak peek for what the new release will contain?

The most important change will be support for new printers, full on-screen installation.
But also a bunch of new features, changes at the UI, more functions and latest software.

More as soon as I have everything working.

Right now wifi ist not working and serial flashing the onboard MCU is broken.
These are the last big puzzles to get the new release out.

pmbroth, 2025-05-13:
Phil - Thank you for the update.  I"m ready for beta testing, when you get to that point.  Brian

liberodark, 2025-05-14:
Hi,

If you need testers for qidi plus 4 im available.

Best Regards

iltis2010, 2025-05-20:
x plus 3 köönte ich anbieten

cwayne1989, 2025-05-31:
I got you on another Smart 3 test if you need it Phil or if someone ends up backing out.

Phil1988, 2025-05-31:
@pmbroth @iltis2010 @liberodark @Arsoth @cwayne1989
Sorry for the delay guys!

I was 100% into developing v2.00 with every free minute (and to be honest even above the "free time").
The alpha tests were good as long as the testers had time/the possibility to test.
Right now the most problematic model is the Q1 Pro where each tester has its individual issues for not being able to test :D

I lost too much time already circling around this while I needed/should have focused on developing the sofware further.
So I decided to release an open beta for all of you.

It is working pretty good for me and the alpha testers, but especially the install process needs to be tested further and also every function and display menue in detail.

If you are interessted, please have a look at the [Open Beta Release of FreeDi v2.00](https://github.com/Phil1988/FreeDi/issues/243)

closed 2025-05-31 by Phil1988