Phil1988/FreeDi #219: [Help] Klipper 13

cptcl, 2025-04-13:
Hey

Qidi X-Max 3 Rev 2 User here

[image]

Is it save to do this?
I was told that i can click every update that shows up in the GUI.

Do i have to to something extra? like flash something or so?

Thx in adavance

leadustin, 2025-04-17:
> Hey
> 
> Qidi X-Max 3 Rev 2 User here
> 
> [image]
> 
> Is it save to do this? I was told that i can click every update that shows up in the GUI.
> 
> Do i have to to something extra? like flash something or so?
> 
> Thx in adavance

I simply updated my Plus3 via Mainsail and did not have to compile a new firmware. However, I do not use FreeDi. But Klipper should be identical. So you shouldn't have any problems either.

martin-badin, 2025-04-18:
Can I flash mainboard and toolhead with `make flash` command?

pmbroth, 2025-04-19:
I have qidi loaded and ran the update to 13 and all works fine.

cptcl, 2025-04-19:
So i did the update and get this error

[image]

[klippy.log](https://github.com/user-attachments/files/19821455/klippy.log)

also here is the klippy log

Can someone help me solve this?

cptcl, 2025-04-19:
Did a complete fresh reinstall.. 
so.. Disconnect every usb devices in the back..
FreeDi image on MMC ... Bootup..its FreeDi 1.30
Update everything
No connection between Moonraker & Klipper.

Startet kiauh, Install klipper, mainsail, cowsnest..
Started FreeDi update.sh and install.sh after redownload from git
No connection between Moonraker & Klipper

Flashed Toolhead & Mainboard by hand..
No connection between Moonraker & Klipper

THIS.. is driving me crazy rn

closed 2025-04-19 by cptcl

pmbroth, 2025-04-19:
glad you got it working...