Phil1988/FreeDi #223: [Feature request]

SmileyMan, 2025-04-18:
**Is your feature request related to a problem? Please describe.**

Screen does not exit "Booting" screen if the tool board replaced with non qidi one

**Describe the feature you'd like**

Option to disable checks and firmware updates for the MKS_THR tool board. I suspect screen is hanging waiting on MKS_THR to respond. 

Fitted Fly SHT36V3 using RS232 communication. Original board failed and wanted the extra features. Klipper sees it as a serial port (RS232 over USB to CH340 serial controller).

SmileyMan, 2025-04-18:
[image]

Phil1988, 2025-04-22:
Can you show me if mainsail gets any errors and also provide a -debug output?