Phil1988/FreeDi #228: [Bug] Canceling print after pausing will bottom out the build plate

thetoadsage, 2025-04-23:
**Describe the bug**
After pausing a print, and then canceling. the build plate will go to the bottom of the chamber. When trying to home after this, the build plate will stall against the bottom

**Expected behavior**
expect the canceled print macro to work as it looks like it should, I can't figure out why it doesn't

**To Reproduce**
Steps to reproduce the behavior:
1. pause the print
2. cancel the print
3. observe

**Screenshots or picture**
If applicable, add screenshots to help explain your problem.

**Console output**
If applicable, add the debug console output here (here is a [how to](https://github.com/Phil1988/FreeDi/wiki/Bugs!))

**Base (please complete the following information):**
 - Printer model: x max 3
 - X3seriesLCD firmware version: current as of today
 - Source: Freedi
 - Mainboard: stock x max 3 mainboard

cptcl, 2025-04-23:
@thetoadsage Hey. Im not a coding professional for gcodes but ChrisInMilwaukee and me have a Beacon probe installed and we had the same issue.

In this discussion
https://github.com/Phil1988/FreeDi/discussions/121#discussion-7882841 
in the first post unter macros.cfg Chris did this

```
[gcode_macro PRINT_END]
gcode:

   ; Compensates for Beacon probe G28 homing causing collision with Z-axis bottom    
   {% set max_z = (printer.configfile.config["stepper_z"]["position_max"]|float) - 19 %} 

```
and at least my Printbed didnt crash into the bottom anymore.  Not while homing and not whe i stoped a printer.
U could give it a try.

Phil1988, 2025-04-23:
I cant reproduce this on my end.

Have you @thetoadsage added something on your side?
If yes, its your responsibility to implement it in a way it works ;)

You likely added a beacon/cartographer and its additional macros.
If so, please adjust the PRINT_END according to @cptcl suggestion.
See also here:
https://github.com/Phil1988/FreeDi/issues/107

I'll wait for your response, but I am pretty positive the [Bug] tag is incorrect 😉

closed 2025-04-23 by thetoadsage

cptcl, 2025-04-23:
> Thank you both!!!! I love the 3d printing community, you guys are so helpful lol. This had me so stumped. I do have a beacon probe installed so [@cptcl](https://github.com/cptcl) solution will most likely fix it! Thank you so much again [@Phil1988](https://github.com/Phil1988) !!

You are welcome. Maybe read our discussion, there could be some helpful tips inside.