Phil1988/FreeDi #230: [Help] Fresh emmc new install worked fine wont print

RustysGarage, 2025-04-24:
i messed up and used max 3 flash on my plus 3, i have 3 of each and a plus 4 and q1pro. anyways i caught what i did and reflashed it with plus 3, worked perfect with freedi till i flashed wrong, i updated to 1.42. i can send print from orca and it heats everything up but sits there never prints,  it shows screen like print is done says reprint on bottom, sometimes i can push that and it will start print lays 1st layer down perfect this is a baby step block or a test cube. then 2nd layer looks great to till about 3/4 way through and then it goes to pure crap, stop print and 2nd layer and 1st layer  can seperate like its not bonded. 1st you cant pull apart at all its smooth like glass. i have reflashed 3 or 4 times same issue every time, so i got a new emmc same thing , i have used print settings from one of my other plus3, i have used their printer.cfg as well same thing, i cannot get past this 2nd layer issue or the sending of orca to printer and showing print is done. could it be the toolhead board? even though its the same one max 3 uses?

RustysGarage, 2025-04-24:
no custom macros or start print end print settings all stock what ever orca puts in i just add ip.

Phil1988, 2025-04-24:
I need detailed information on the current status.
The information flood what happened before etc. is too much.
Can you please summerize the current state of the printer?

Information I need: 
FreeDi version? 
Display firmware?
Is the printer.cfg untouched?
Any custom modifications?
Console output?
Debug output by starting FreeDi manually?
Maybe a video?

RustysGarage, 2025-04-25:
FreeDi version?  v1.42-2-g30c9b57d,   klipper-v0.13.0-0-g61c0c8d2,   mainsail v2.13.2.   moonraker  v0.9.3-0-g71f9e677

Display firmware? not sure where this is
mcu(stm32f401xc)
Version: v0.12.0-289-g08a1c9f1
mcu MKS_THR(rp2040)
Version: v0.12.0-289-g08a1c9f12
Host(aarch64, 64bit)
Version: v0.13.0-0-g61c0c8d2e-dirty
OS: Armbian-unofficial 24.2.0-trunk bookworm
Distro: armbian

Is the printer.cfg untouched? nothing touched at all except what it tellls you to back up from the OG cfg and replace into the freedi per install instructions.

Any custom modifications? no mods no scripts no macros nothing everything on printer is stock

Console output? not sure what ya want

Debug output by starting FreeDi manually?  dont know how to do this

Maybe a video?  this time it started the print 99.9% of the time it does not, and it made it to 3rd layer which it never does.
https://www.dropbox.com/scl/fi/ais4f5ld7rcjm74o45w9j/20250425_145000.mp4?rlkey=kdsg80qd0w10pcghkf722qnd3&st=0b0kw76x&dl=0

Phil1988, 2025-04-26:
Display firmware can be found at the "about" menue or when running the debug console:
https://github.com/Phil1988/FreeDi/wiki/Bugs!
This debug output also give some deeper insights of what it is doing.

About the "it started when I installed the firmware of the X-Max on the X-Plus":
That has nothing to do with it.
Those 2 printers are almost identical on my side - except the printer.cfg and macros.cfg.
So even if it seems to be linked for you, I am sure it is not.

Have you tried different filament/different print temps and checked the slicer profile?

Thanks for the videos btw.: They helped to diagnoze is better.

RustysGarage, 2025-05-06:
ok screen is 1.42 debug i think i did right  not sure where a log is saved though. seen it scrolling but never really said anything

Phil1988, 2025-05-09:
@RustysGarage 
> Have you tried different filament/different print temps and checked the slicer profile?

I'm sure its not a software or FreeDi issue.
It must be something from the Filament/Slicer/Klipper.
It is very hard to say from remote, but I would guess it might be too low nozzle temp/too much cooling/too fast printing or simply some material/filament problems.

Can you do some tests on these parameters and feedback your findings?

RustysGarage, 2025-05-09:
yes i have tried differ filament pla petg and petg-cf  same thing, tried profiles from one of my other x plus 3 in orca same thing, tried the printer cfg from other plus3 same thing, worked perfect till i upgraded now .  to me seems like z offset issue, i can get a perfect 1st layer smooth as glass, 2nd layer smooth as well 3rd layer it balls up if i baby step it down to get it smooth then restart test print 1st and 2nd are way to close  and horrid 1st n 2nd layer ripples, tears and drags. 2nd layer doesnt stick to 1st layer either.

how do i put it back to stock, ill test that see what happens, if it works then ill redo freedi again.

Phil1988, 2025-05-12:
Here are some posts of how to roll back: https://github.com/Phil1988/FreeDi/discussions/73

Will be interessting to see if this changes anything.
Right now, I can not see a reason why it will improve it, but I totally understand that you have to check the cause of it and "start somewhere".

Please share your findings here and I will check this thread frequently to hopefully assist you!

Phil1988, 2025-06-24:
@RustysGarage Any news on this?

RustysGarage, 2025-06-25:
i havent reverted it back yet, i did a total install again  same thing, took it home used all new stock print profiles from orca, i have never encountered anything like this, 1st layer goes down like glass, 2nd layer balls up and never sticks to 1st layer,  its like it raises up to high for 2nd layer if i adjust with baby steps then next print 1st layer is dug into bed and to close, im at a total loss, next step would be a new toolhead board i guess.

Phil1988, 2025-06-26:
That is honestly super weird and I have no idea what it could be and what can solve it.

I dont know why this should be the case, but maybe this machine is doing more z-travel then requested?
As you are having a bunch of them, I should be easy to compare.
Home all, then move eg. 100mm +Z (so that the bed moces down) and measure if its the same on the working machines and on this "special case".

I dont expect this to solve it, but I invite you to install v2.00 from scratch on this printer just to give it another shot.
https://github.com/Phil1988/FreeDi/wiki/Tour-%231:-Requirements

That must be super frustrating mate and I would love to give you "the solution" but there might me something underlaying I'm afraid.

RustysGarage, 2025-06-26:
yeah its a real pain, since i put freedi on all my machines (6 max 3 and plus 3) they print sooooo much better and this one was flawless, and back tracking it was printing great then when i updated it i installed max3 instead of plus 3, so i reinstalled for plus3, ever since then its not printed. 1st layer great 2nd layer trash i made it once to 3rd layer lol. never experienced anything like it.  i will try 2.0 i did just get 2 new emmc yesterday.
thanks

 

https://github.com/Phil1988/FreeDi/wiki/Tour-%231:-Requirements

RustysGarage, 2025-07-08:
Phil i think i accidently found the issue tonight, i got 2.0 installed, new emmc, toolboard, extruder and hotend. did all the cals and started baby step print, bam same issue as above. canceled print about 15 mins later for some reason i grabbed the filmament on top of toolhead and pulled it right out and then pushed it back in without issue. and i noticed every time i start a print the purge line prints  and about a 1/4 of the outside wall doesnt print its clicking like its clogged and then bam it will start printing then the 2nd layer happens,    seems to me the extruder is slipping and doesnt have a grip on filament, how or what would cause this? any fixes for a qidi plus 3? i know it has a pring in it for tension

Phil1988, 2025-07-08:
I'm happy to see you are onto something.
I am a bit confused, because you said you already changed the hotend and extruder?
Normally my suggestion would be to change the extruder gears (that feeds the filament).
Maybe the tension isnt enough or the teeth doesnt bite enough into the filament due to wear and tear or poor manufacturing.
But as far as I understood you changed everything already?

Besides that it can also be that the flow has higher resistance - the nozzle/hotend would be the reason for that, but you changed this also, right?

RustysGarage, 2025-07-08:
yup changed everything  all new was trying to make sure i covered everything. i have a nother new extruder at shop going to bring it home tonight and try it.

RustysGarage, 2025-07-10:
didnt work   grrrr swapped out extruder again last night same thing, i think im going to gut it and put BTT boards and a jazz toolhead on it, im done trying to figure it out   thanks for the help ive changed everything, used printer cfg from my other plus 3 running freedi i measured z steps and its on point, move 55mm and it moves exactly 55,  rotation distance is on point, dont know what else it could be except the main board

pmbroth, 2025-07-10:
Hello on my -x-max 3, I had filament clicking, no clog, I took apart the dual gear extruder and the screws that held the gears in-place to pull the filment were  loose. I tightned them down, and no more clicking.  It occurred again, and the did the same thing.   So in my case the screws kept coming loose.  I changed the extruder and no issue.  Dont know if this is what you are having, but wanted to share my expoerience.

RustysGarage, 2025-07-12:
i got it fixed its printing away,  dont ask me why but everything was replaced except the main board, the hotend was a complete new set hasnt even been used except trying to get it going. well before i gutted the entire rig and was going to mod it out since warranty is over i swapped whole hotend again and bam its printing away.  Thanks Phill for the help   dont know what could be wrong with the new hotend but qidi wanted it back

Phil1988, 2025-07-13:
"but qidi wanted it back" - this means you returned it as it might have been defective on arrival?

So in summery it is/was all down to super hard to get hardware failure, right?

Phil1988, 2025-08-02:
@RustysGarage Any news or an answer for me? 😅
I would like to close this as I think it is solved and was a pretty specific hardware issue for you.

It would be nice, if you could tell the latest status, summerize the solution for others and close this thread 😊

RustysGarage, 2025-08-02:
Yeah I posted the reason just above your post. The new hotend was bad and causing it.

closed 2025-08-02 by RustysGarage

Phil1988, 2025-08-02:
Thanks - so in the end it was "just" a hardware malfunction.
I am glad that it is solved for you now 👍

Happy printing with all your machines mate 😉