Phil1988/FreeDi #231: [Help] Correct steps to flash toolhead

martin-badin, 2025-04-25:
Hey guys. What is the correct process/steps to flash the toolhead?

I have tried this one
1. Turn on the toolhead in boot mode
2. sudo cp ~/klipper/out/klipper.uf2 /dev/sda1
3. sudo service klipper stop
4. sudo service klipper start

This one works until I turn off my printer. After that, my toolhead is not recognized.

cptcl, 2025-04-25:
Open the printer on the backside. Disconnect all USB devices.
than follow your 1,2,3,4. Just..
after 1, type "lsblk" and have a look if the tool head gets /sda1. If it is sda1, go ahead with 2.

After that. Turn off printer. Wait like 30 seconds. And turn it back on.

If your tool head shows anything else than sda1, u have to adapt your cp command.

martin-badin, 2025-04-25:
What does `adapt your cp command` mean?

cptcl, 2025-04-25:
Or.. After u set your tool head in boot mode.. Type "lsblk" and show us the output.

Ps: restarting the Klipper service will not restart your tool head as far as I know.

martin-badin, 2025-04-25:
Thank you. It seems I did everything you described. I follow those steps

https://github.com/Phil1988/FreeDi/blob/master/helpers/klipper_auto_flasher.sh#L29-L36

cptcl, 2025-04-25:
Maybe don't go with the helper sh file..
look at this. https://github.com/Phil1988/FreeDi/wiki/Installation-guide-for-fresh-systems
If u want to do it by hand,.. A bit down u find a spoiler, right above creating the firmware for tool head.
my printer didn't auto flash the tool head so I had to flash it manual, like u trying to do.
hope this helps.
ps: sorry.. I'm in work and typing on phone.. So.. A bit less details in my text here.

martin-badin, 2025-04-25:
@cptcl Please, could you share your firmware for toolhead?

cptcl, 2025-04-25:
@martin-badin if u read my link, u will see what u have to do to build your own firmware. U just have to follow the steps described. 
I run Klipper 0.13 so my tool head firmware will also be 0.13.
Mcu, tool head and host need to be in the same firmware (Klipper Version) to work.
So even if I share.. U would not be able to use it.

Let's take a step back and pls answer this.
How did u set up freedi? From stock system? 
Did u already run freedi and just want to upgrade?

martin-badin, 2025-04-25:
I did set up freedi from stock firmware. And then I want to upgrade to Klipper 0.13.0

cptcl, 2025-04-25:
So u did already run successfully freedi. Ok
Than u hit the update button for Klipper 0.13 and..
What is updated to Klipper 0.13? What does the webgui of the machine show? 
Host?
Mcu?
Toolhead?
I'm asking this cause I also had a problem with Klipper 0.13 upgrade.. And I would need to know what's done and what's not.

martin-badin, 2025-04-25:
Here is screenshot

[image]

martin-badin, 2025-04-25:
but the issue is that toolhead is not recognized after turn off/on the machine.

cptcl, 2025-04-25:
So screenshot is right after flashing tool head, then stop and restart Klipper service right? 
and if u power off and power on the printer it says "mks_thr not recognised"?
Pls check the serial in printer.cgf if this is correct.

Otherwise.. Normaly it should work. 

So now I tell u what my problem was...
after kicking update Klipper 0.13 it told me that is has a problem with "serial" condition in "Serial" and.. The only way I could make it work again is..
flash mm with fresh freedi 1.40 image.., start printer, change serial for printhead in printer.cfg.
Git cloned freedi 1.42, install.sh and update.sh from fresh cloned freedi. restarted printer.
Had to look if everything was recognised correctly.. And than after all other possible updates, updated Klipper, restarted printer again and flashed mcu and tool head by fresh compiled firmware (like described in the link I posted)

Only after that I was able to use my printer again.

Since I see u are on same firmware on all 3 devices.. Normaly it should work. Except.. Serial in printer.cfg is wrong, than it can't recognise it. (normaly)

Itis maybe a bit of work. (and yes I also wanted to avoid a fresh new install on my printer)
but.. If everything fails.. Try it. 
Try it from the beginning.

Ps: my guess is, that I had some troubles back in the Klipper 0.12 time and did some.. Workarounds.. And that is what made my printer stop working. 
pls, try to install it again from scratch, like u did it in the first time u installed freedi.

I'm not an freedi expert, I'm just trying to help and share my experience and what I did to make my printer work again.

martin-badin, 2025-04-26:
@cptcl Thank you for help. Today I changed mainboard of the toolhead and now everything works at first time.

Phil1988, 2025-04-26:
Thanks @cptcl for your support here 👍 

If it is solved for you @martin-badin, please close this thread.

Btw: I saw your pull request, but I will not use it.
I have compiled firmwares already but there are reasons I did not publish them yet.
So just you be aware of it.
Tthanks for your support and will to help :)

closed 2025-04-26 by martin-badin