Phil1988/FreeDi #233: Installation Guide is misleading

KoeSource, 2025-04-27:
Hello

I try to install the FreeDi firmware on my X-Plus3 Printer.
I choose a fresh installation on a 32GB EMMC.
It's really confusing to get the correct Image for the printer.
The first Release assets are only Source-Code, i'm looking for the Image.

So i did whats written here:
[image]

This points me to follow Image:
[image]

This is really old, but i downloaded it and wrote it to my EMMC.
Startet the 3D Printer and i got. "System started abnormally" multiple times.

After watching the tutorial video i found out that there are multiple versions:
[image]

I suggest to improve the written guide.
I even suggest to add the current images always in the assets.

Regards Int3g3r

Phil1988, 2025-04-27:
Hello Koe,

I appreciate your feedback, but I must express my disappointment with your suggestions and the nature of this report.

Developing FreeDi and its documentation has required thousands of hours of my personal time, all provided for free. I don't know how I should feel about someone jumping in and criticizing this project, who hasn't even taken the least amount of time checking the wiki and the release section.

The images and versions are clearly marked in the release section as "(software and **image**)" and a bit more attention could have resolved your concerns.

Here is an example: [FreeDi v1.40 (software **and image**)](https://github.com/Phil1988/FreeDi/releases/tag/v1.40)

[image]

I believe my extensive documentation, including a detailed wiki with (pictured) step-by-step guides on every aspect and a tutorial video, provides the necessary guidance.

Your suggestion implies that I should invest even more time into providing images for every printer model with each update, which also requires some time for creation and testing. 
While I understand the convenience of having everything immediately accessible, I hope you can appreciate the effort that goes into maintaining this project, especially since I manage it entirely on my own.

If you feel that my project does not meet your expectations, you are welcome to explore other options or even create your own project to demonstrate how it can be done differently.

Thank you for your understanding.

KoeSource, 2025-04-28:
Hello Phil1988

You seem to have misunderstood me.
I don't expect you to create a new image for every update.
You could, for example, insert the link from the latest image in each release.
This is just a suggestion.

However, the following instruction in your guide still leads to the wrong image:

> Simply download the image file for your needs.
> You can find the full image at the releases section.
> Search for a *.rar file, download it and unpack it (you will get an *.img file).

If I could I would change it myself, but I don't have the rights:

> Simply download the image file for your needs.
> You can find the full image at the releases section.
> Search for “Images” (CTRL+F) and open the assets.
> The file name should look like this “FreeDi_v1.40_XMax3_Armbian24.2.0_Bookworm6.6.17.rar”, 
> download the correct image for your model and extract it.
> Unpacked you should get an *.img file.

Here I have the Markdown file, could you replace it?
[Installation-guide-for-fresh-systems.md](https://github.com/user-attachments/files/19936005/Installation-guide-for-fresh-systems.md)

That way other users would not have the same problem.

Thanks for your time.
Greetings Koe

Phil1988, 2025-04-28:
> However, the following instruction in your guide still leads to the wrong image:

Can you please provide the link of the page and the faulty URL?

Phil1988, 2025-04-29:
> [...] and this is wrong, at least in my view

The link is forwarding you to the release section with all the releases.
That is correct.

There, you can pick and download the image file you need, just as the text states.

So essentially, you're criticizing that you used `CTRL+F` to search for "rar" and didn't find the latest version?

Well... I have to say, that I expect users to use their brains to some degree ;)

I simply don’t have the time to handle suggestions like this - following that logic, the next iOS user will complain that "`CTRL`" is incorrect and I should write "`Command`" instead.  
You see the point: 
I can't - and frankly don’t want to - invest the time in creating an all-purpose guide for every possible user scenario.
The guide is already very detailed and easy enough to follow.

> Would it not be better if the search would lead the user to follow images?

It would be better if a user understood what they are doing.
I expect users to take the time to engage with the project properly.  
If someone already struggles at this basic step, it's almost guaranteed that there will be further issues down the road - issues that will likely require my time to address.
And to be clear: I’ve done that countless times before that even went down to repairing and configuring users network properly on a 4h remote session to find IP addresses and such.
There’s a limit to how much hand-holding is realistic or fair to expect and I have to set priorities.

closed 2025-04-29 by Phil1988