Phil1988/FreeDi #234: [Help] Beacon H  - need help

dd0184, 2025-05-03:
Freedi has been working wonderfully but as I've gone down the rabbit hole of making my printer better I've run into a problem. I installed a Beacon H and now get this error that reports the mainboard mcu is on too old of a version. I attempted to update it via usb but cannot get the mainboard into boot mode. I also attempted via micro sd with a firmware.bin in the root directory. 

Does anyone know how to make this work?

[image]

cptcl, 2025-05-04:
Compile the firmware by your self. Pls follow this guide.. https://github.com/Phil1988/FreeDi/wiki/Installation-guide-for-fresh-systems#3-you-are-a-hardliner-and-want-to-compile-it-on-your-own

Compile both. firmware for mcu and toolhead. Keep in mind the file for the mcu needs to be renamed.

for getting the toolhead in boot mode.. on toolhead board.. Push Boot button and hold it.. Push reset button and release it.. then release boot button.
Keep in mind, it works consistent if u first unplug all usb devices from the mainboard first.

Check the device with "lsblk" and see if its in boot mode.

dd0184, 2025-05-04:
Problem is I am no longer using the stock qidi toolhead. Does the mainboard also reliably enter boot mode when all USB connections are detached? I had no luck with the booto and reset buttons on the mainboard.

dd0184, 2025-05-04:
I also want to say thank you for the assistance.

Phil1988, 2025-05-04:
@dd0184 without the toolhead you can not get my klipper auto flasher to be invoked.
So you have to manually flash the mainboard mcu using an SD card.
There are other options, but they need more assistance and therefore may need longer.

Please follow this:
https://github.com/Phil1988/FreeDi/wiki/Installation-guide-for-fresh-systems#2-your-mainboard-mcu-and-the-toolhead-are-not-on-klipper-v0120-289---and-you-want-it-the-easy-way

And follow the describtion for "On FreeDi lower than v1.40" to flash the mainboard MCU :)

cptcl, 2025-05-04:
> Problem is I am no longer using the stock qidi toolhead. Does the mainboard also reliably enter boot mode when all USB connections are detached? I had no luck with the booto and reset buttons on the mainboard.

U replaced your toolhead and therefore the toolhead board. U did not mention this anywhere. 
Nobody knows what toolhead board u have now. I assumed u have the stock toolhead.

Sry, but than u have to lookup your toolhead online. Seach for "how to flash it" and what settings u need for the compiler. other than this, its like the same steps as with the stock toolhead guide.
i guess the mainboard (MCU) is the stock one? so u can use the MCU section to flash at least that ootb.
But if u replaced the stepperdriver or the mainoard itself too,.. u have to lookup these as well.

dd0184, 2025-05-04:
Thank you! The problem was the naming of the firmware.bin. It needed to be "X_4.bin"

closed 2025-05-08 by Phil1988