Phil1988/FreeDi #238: [Help] No connection via LAN

Jan-ita, 2025-05-13:
I have installed the full image  Freedi 1.40 on my X-Smart 3 all seems to go like it is described in the manual and at the point when i have to connect with the Printer via LAN by typing the old IP i yust get timeout.
Connection is via Lan cable directly between Laptop Windows 10 and Printer and had workt well for making the backup before.
Is there a new IP i have to use? Ore something I have to do with Putty ore WinSCP?
I am not familliar with these things...

Phil1988, 2025-05-13:
You need check for the new IP address.
Your DHCP will identify it as a new device and give it a new IP.

See also:
https://github.com/Phil1988/FreeDi/issues/156#issuecomment-2673016863

I hope this helps you to find your new IP.

Jan-ita, 2025-05-13:
Thank You!
 Have now connected the Printer via LAN with FritzBox and found the New IP.

closed 2025-05-13 by Jan-ita