Phil1988/FreeDi #246: WiFi Dongal

Nihilus77, 2025-06-07:
I run a TP-Link Nano AC600 USB Wifi Adapter(Archer T2U Nano)- 2.4G/5G Dual Band. When I ran the instal fir the wifi to work, nothing shows up in network area. I switched back to the stock dongal and it works fine, the reason why I use the TP-Link, the stock will lose connection.

leckert123, 2025-06-07:
This is a hardware compatibility issue, not software/firmware. 

Many of the TP Link Wi-Fi dongles use a Realtek or other chipset which is not compatible. Suggest you check out the QIDI Discord and search there for the WiFi dongles that will work.

Nihilus77, 2025-06-07:
> This is a hardware compatibility issue, not software/firmware.
> 
> Many of the TP Link Wi-Fi dongles use a Realtek or other chipset which is not compatible. Suggest you check out the QIDI Discord and search there for the WiFi dongles that will work.

Thanks for the reply, Ill look there. Just odd the TP dongal worked fine for the bone stock X-Max 3 , just after the FreeDi set-up it doesn't.

leckert123, 2025-06-07:
Well, that does seem unusual that the dongle worked prior to FreeDi.  I did not realize you'd used it prior.

Phil1988, 2025-06-08:
It is indeed usual.
QIDI uses a complete different OS with different kernel and drivers.

@Nihilus77 
2 users have reported that this worekd for them (maybe not precisely but in in general):
https://github.com/Phil1988/FreeDi/issues/96#issuecomment-2609027398

I dont have that module and cant test.
Btw.: Its called "dongle" ;)

Please switch over to that thread as its covering thatwifi chip aleady :)

closed 2025-06-08 by Phil1988