Phil1988/FreeDi #252: [Feature request] UI change on display

Pukkahq, 2025-06-22:
On the display the filament extrude length 10, 20, 50mm can this be changed to 20,50,100. I personally do no unload, I just cut the old off and feed the new through behind.

Phil1988, 2025-06-23:
Thanks Paul 😉

I will check this in the future - but to not raise any high expectations: 
This will be a low prio tasks and will probably take several months.

v2.00 took ~3month full throttle and I need to rest and do some real life stuff 😅

pmbroth, 2025-06-25:
Since we have the ability now to change the 4 presets on the main screen, can't one of them be used to do a filament load via a macro?

pmbroth, 2025-06-26:
I figured he can change one of the presets, it is quick,  and done.  Saves you from alot of work.  As people are going to want different features for their varius needs,  and now they have the ability.

pmbroth, 2025-06-26:
agreed, but down and dirty, for now, as he says he does not unload, just cuts and loads it out.

Phil1988, 2026-01-02:
@Pukkahq and @pmbroth 
The latest release includes this new feature ✨
Please update via mainsail to [v2.08](https://github.com/Phil1988/FreeDi/releases/tag/v2.08) and use the [new parameters](https://github.com/Phil1988/FreeDi/blob/eb6c13e3d8010b26cdbc60e8e674f66953f68e0e/config_section/generic/freedi.cfg#L48) to make use of it - and dont forget to restart FreeDi after changing it and then reporting back here if it works as you wished :)