Phil1988/FreeDi #253: FreeDi v2.00

Phil1988, 2025-06-23:
## ⚠️ Heads-up: Upgrading FreeDi from **v1.x ➜ v2.00**
  
A quick notice before the v2.00 release drops later today:

### Why this notice?
Moving from the 1.x line to **v2.00** brings a completely new software base (Armbian 25, Klipper 0.13, fresh screen firmwares, reworked configs, the Wizard, and much more …).  
Although I’ve run several upgrade tests successfully, I **cannot guarantee** a flawless transition on every device.
This major release can cause problems on a normal update.

### 🔒 Please back up first
1. `printer.cfg`, `macros.cfg`, etc..
2. Any custom scripts in your `Klipper` / `Mainsail` config folder  
3. Optional but recommended: Backup your entire EMMC for extra peace of mind  

If something goes wrong you can always flash back or copy your files onto a fresh image.

### Recommended path (safe)
* Wait for the **official v2.00 image** – it will be uploaded later today.  
* Flash it, run the Wizard, restore your preferred sections of your backed-up configs.  

### Experimental path (adventurous)
Can’t wait? Feel free to try an in-place update and **report back**:
Your feedback is crucial for making FreeDi rock-solid.  
If you’re willing to be a guinea pig and can survive a little downtime, your test results are greatly appreciated.

1. Backup everything (see above).  
2. Hit the "UPDATE" button on mainsail
3. If it works: awesome! Tell us in this issue.  
4. If it breaks: also tell us – logs & details help me iron out edge cases.

Happy printing (or debugging 😉),  
**FreeDi-Labs**

Phil1988, 2025-06-23:
placeholder1

Phil1988, 2025-06-23:
> Phil, What will be the procedure to go from beta to production release?

You will get more information about that at the "placeholder2" post.
Let me push the stuff first 😅

alexkipar, 2025-06-23:
Wow, once mine max3 and q1pro finish their 10kg PETG sprint, I will definetely try 2.0 on both (max3 is already on freedi, so some update flow I will check as well)

Muiezz, 2025-06-23:
Where is the image file?

pmbroth, 2025-06-23:
Phil has tp [ut up the final instrucitons, I know there was one typo in the manual instrucitons,  He is suppose to be putting them up later tonight , or tomorrow.  He had an important meeting to attend.

NoNameCola, 2025-06-24:
Is it normal to take more than 4 hours to flash the display? Mine has been going for 4 hours and is at 4%.

Phil1988, 2025-06-24:
@Muiezz 
> Where is the image file?

Its at the assets now. Had to leave yesterday mid releasing because an urgent call.

@Cletis234
Either change your mcu section back to
```
[mcu]
serial: /dev/ttyS0
restart_method: command
baud: 250000
```
or flash the new X_4.bin from `~/FreeDi/mainboard_and_toolhead_firmwares/v0.13.0-154/`.

@NoNameCola 
No, it is stuck.
Switch the printer off an on again, see here:
https://github.com/Phil1988/FreeDi/wiki/Installation-guide-for-fresh-systems#flashing-the-display

It should take about 4min.

MegamanAT, 2025-06-24:
> Tried to update from beta to release, did not work. error with hard recovery. write access denied. So I flashed the release firmware to the emc (x-max3) and now it get this error:
> 
> Klipper reports: ERROR
> 
> mcu 'mcu': Unable to connect Once the underlying issue is corrected, use the "FIRMWARE_RESTART" command to reset the firmware, reload the config, and restart the host software. Error configuring printer
> 
> [image]

I changed the baut to 250000 (was 500000) and now it works

Phil1988, 2025-06-24:
@ScheissQidi 
> Used a fresh emmc, replaced the original. Tried to flash FreeDi V2 to my Plus4. Stuck at 16% for 30 mins now. Any suggestion?

My suggestion is to read the f**** (friendly) manual 😜

[image]
Source: https://github.com/Phil1988/FreeDi/wiki/Installation-guide-for-fresh-systems#flashing-the-display

@MegamanAT 
> So I flashed the release firmware to the emc (x-max3) and now it get this error [...]

You forgot to flash the mainboard MCU or you can leave it like it is and use the legacy 250000 baud :)

@Cletis234 
> ~/FreeDi/mainboard_and_toolhead_firmwares/v0.13.0-154/ did not work.
> Says it is a directory

Yes, ~/FreeDi/mainboard_and_toolhead_firmwares/v0.13.0-154/ is a directory.
I dont know what your are trying to do.

> the Armbian repo is Dirty.

What? Show us!

Phil1988, 2025-06-24:
@ScheissQidi 

Hahah - I'm always thinking the best of you ;)

There is a typo at the file name. It must be **qd_mcu.bin**.
Once inserted you need like 10s (and this is already plenty) to flash the mainboard MCU.

For the screen: Dont worry, this is completely OK.
Just restart the printer. FreeDi will detect that the firmware is not OK and will update the screen.
It sometimes needs a couple of tries.
A few developers already checked this in detail and the cause is not clear yet. 
It is likely a serial overload of that tiny STM32 µC.

@Cletis234 
I was wondering, because there is no "Armbian repo" visible at all on mainsail. 
You mixed the update instructions with the new installation instructions - but its not your fault and rather based on me getting interrupted when releasing and writing the instructions :)
On the release *.rar you had the `X_4.bin` already (its the same as in the `~/FreeDi/mainboard_and_toolhead_firmwares/v0.13.0-154/` dir on your printer).

Now for your error:
1. Have you used the wizard and successfully flashed katapult and klipper to your toolhead?
2. Have you power cycle your printer (turn it off, wait 15s, turn it on)?

ScheissQidi, 2025-06-24:
THX for your best guesses ;-) 
the file was original named as "bin.bin" in the rar-file, I have corrected it to only one ".bin" before using. ((Sometimes I've been so smart ;-)) ))
But I saw no change while booting. 

After my last comment I removed the back cover of the printer, so I could identify hardware and bring some order in this bunch of weired cable... 
BTW, my board is x7 V1.0 
So I turn it on and keep an extinguisher at hand

pmbroth, 2025-06-24:
I ran freedi recover, and get the bewlo

[image]

[image]

pmbroth, 2025-06-24:
I was told not to do the Hall sensor stuff since mine is an x-max 3

pmbroth, 2025-06-24:
stuck on connecting to klipper on x-max 3

Phil1988, 2025-06-24:
@Muiezz something is weird on your system.
Dont continue if you already know something is not right.

If you ahve discord, please head over to get faster help.
If not, please attach pictures here etc. so I can see whats going on :)

NoNameCola, 2025-06-24:
> [@Muiezz](https://github.com/Muiezz)
>
>
> Its at the assets now. Had to leave yesterday mid releasing because an urgent call.
>
> [@Cletis234](https://github.com/Cletis234) Either change your mcu section [...]

[image]

Does this look right? I tried restarting and it's also still taking forever.

adiskovarik, 2025-06-24:
Installed without problem on X-Smart 3. Just one question. Is there a way I can enable "sidefan" toggle on the X-Smart screen ?  There is an empty space below filterfan where I assume other models have the toggle for it.

Phil1988, 2025-06-24:
> Installed without problem on X-Smart 3. Just one question. Is there a way I can enable "sidefan" toggle on the X-Smart screen ? There is an empty space below filterfan where I assume other models have the toggle for it.

Thanks for you feedback.
Very important timing indeed as there is currently a bit madness betweeen Q1 and Smart3 screen firmwares on the auto detection.

For your question:
Unfortunately it is not possible to "enable" it.
The screen is not capable of doing something like this dynamically and everything has to be set up at compiling time.

NoNameCola, 2025-06-24:
@Phil1988 Yes I do, I had previous flashed OpenQ1 and therefore installed the toolhead firmware. Should I disconnect the toolhead and try again? 

[image]

NoNameCola, 2025-06-24:
Ok I think I got it sorted, the issue was I was running klipmi on the display and it was running the service.

Phil1988, 2025-06-24:
ahh... yea... and I was confused.
During the flash process its normal to have the landscape orientation.
If you see this orientated in the right way:

[image]

Please do not continue with the installation.
In both cases, please report back here or join us on [discord](https://github.com/Phil1988/FreeDi/wiki/Community) :)

pmbroth, 2025-06-24:
@Phil1988   I installed Klippaine, and lFreedi screen is stuck  on freedi "Booting"    can access via term and mainsail

chyrio, 2025-06-24:
@Phil1988 my pleasure. Simple issues, simple fixes!

vibraphon, 2025-06-25:
Sorry. I'm either too old, or too stupid, but I don't know how to do the update from 1.42 to  2.0 specifically. 
For the installation of 1.42 there was the step-by-step video I could follow.

Is _**reeDi_v2.00_25.02_6.6.65.rar**_ now the official version, or still the beta?

_“Flash it, run the Wizard, ...”_
Unfortunately I don't know how to flash and run the  Wizard. 

Do I have to update before flashing
- Klipper from v012.0-289 to v013.0-0
- mainsail from v2.13.2 to v2.14.0
- and 6 other system packages as offered?

Thank You

jmpocheau, 2025-06-25:
Tonight, I had a strange issue.
One print stopped unexpectadly after one hour or two. The printer was frozen.
So, I restarted it, and now, the issues I reported are gone.
The temperatures display correctly on the display and while printing, a maintenance icon on the display lead to the the babystepping screen.

 I could not explain what happened, since I already restarted my printer after the update, but now, everything works fine.

Phil1988, 2025-06-25:
@vibraphon 
This is the (stable) release:
https://github.com/Phil1988/FreeDi/releases/tag/v2.00
The beta was clearly marked as "beta".

"Flash and run the wizard" -> is meant for the included image file at the *.rar on the release.

If you want to update I invite you to join discord and ask someone friendly there to guide you.
If the instructions above are unclear to you, changes being high that you cant help yourself on a failed update.

I recommend you use the recommended way and flash the provided image.

For the v2.00 I had no time to to create a video yet :(

Phil1988, 2025-06-25:
@jmpocheau 
> But this looks strange to me to consider the bed from the bottom.

You have to think from the perspective of the screw.
I hope it does make sense for you now :)
So its all correct and follows the logic of klipper for this feature.

Im happy that you like FreeDi :)

zakfromevil, 2025-06-26:
Thanks for your work! On first view all works. On my taste I've added spoolman, timelapse, shaketune, and Fluidd. My AC600 wifi adapter recognized. And few edit start_print macro and printer.cfg for my microProbe.

jmpocheau, 2025-06-26:
@DSCustoms 

> I can't connect to the toolhead board at all. I don't even have the /dev/serial/ folder when I ssh in. What could have happened? I'm going to reflash the emmc and try again tomorrow, but if you have seen this before please lmk.

Hi, I already had this issue. no connection to the toolhead and no /dev/serial.
I encountered that when my extruder leaked and filled the entired toolhead  with a lot of plastic, tearing off the fan wires.
Removing the plastic and reconnecting the wires didn't solved this issue, because the toolhead card was burnt.
I ordered a new one and this solved the problem.
I read that in my case Qidi would have taken this repair under the warranty.
Also some say that this could be due to  defect in the usb C cable.

Good luck

Phil1988, 2025-06-26:
> I can't connect to the toolhead board at all

Already before or after the installation?
I mean: Did you flash the toolhead with the wizard?

Lucianius77, 2025-06-26:
After installation with the wizard. Wizard is saying "found all automatically and will flash mcu and toolhead". Watching on the screen I think it was only flashin one thing - toolhead, and not the board.

I will give you the 3 screens afterwards. A funny thing besides is the restarting screen mit a picture of the x-plus 3 on a x-max 3. ;-)
but it hangs up while "restarting".

[image]

[image]

[image]

DSCustoms, 2025-06-26:
> > I can't connect to the toolhead board at all
> 
> Already before or after the installation? I mean: Did you flash the toolhead with the wizard?

I ran the whole startup wizard. Everything looked great that I could tell. It’s a max3 so your instructions say this is automatic but it just refuses to communicate now that it’s finished. I’m confused why there is no /dev/serial as there is on my other printers. I’m going to try again now, reflash the EMMC and micro sd card and start from new. I’ll take a pic of the toolhead install screen this time. 

I even tried booting the toolhead board with the button like the old instructions but nothing ever happened from that. 

Seems like the exact same issue that ^^ is having. 

Other Qs.
The sd card is just to run on the first boot right?

Lucianius77, 2025-06-26:
@DSCustoms I think we have the same or nearly the same problem.

DSCustoms, 2025-06-26:
> [@DSCustoms](https://github.com/DSCustoms)
>
>
> Hi, I already had this issue. no connection to the toolhead and no /dev/serial. I encountered that when my extruder leaked and filled the entired [...]

This is a printer that worked right before installation, no problem with the toolhead board (v1.1) or usb cable

Phil1988, 2025-06-26:
Have you (both of you) flashed the mainboard mcu with the X4.bin and it was renamed to *.CUR?

Lucianius77, 2025-06-26:
no, I understood it was only necessary for beta testers. Please tell me how to flash the x4.bin

DSCustoms, 2025-06-26:
> Have you (both of you) flashed the mainboard mcu with the X4.bin and it was renamed to *.CUR?

I have the X_4.bin file on the microsd card still. Should that change? I'm doing it again now, last time I did them at the same time, should I try it with just the bin first?

Phil1988, 2025-06-26:
Both of you:
Yes please [flash the new firmware](https://github.com/Phil1988/FreeDi/wiki/Tour-%231:-Installing-FreeDi#flashing-the-mainboard-mcu) to the mainboard mcu.
I didnt had to in some cases depending on the klipper version I tested, but changes being high that your mainboard needs to be updated.
It _must_ be renamed automatically to *.CUR when its finished and successful.

DSCustoms, 2025-06-26:
Trying to start the printer with just the sd card, no EMMC. Sitting on freedi boot screen, progress bar says booting… and got to 100% then hangs. I’ll try a few restarts.

DSCustoms, 2025-06-26:
Oh. Binary file was renamed to .cur after that. Even though it was hung. I’ll go try the EMMC only now

chrashing, 2025-06-26:
Have updated from v1.42 to v2.00 on my Xmax3. Thank you for another great enhancement.
Maybe, to fix the only problem I had, suggest that instruction "Flash it, run the Wizard, restore your preferred sections of your backed-up configs." be expanded some way to mention flash the emmc for the mainboard, and copy to the SD card for the MCU.   It seemed obvious that the emmc needs to be flashed, but then I read it as the wizard would take care of the MCU update, which didn't happen.

Phil1988, 2025-06-26:
> Ok maybe remove the instruction to try both at once? I just did microsd card without the EMMC. Turned it off after a few minutes of it hanging at the booting progress, removed micro sd and added EMMC, and now it loaded without an error.

@DSCustoms 
No I wont remove the hing as it is definitly not connected and I want the users to save time :)
I flashed the EMMC and the sd card simultaniously a couple hundred times already on the Max3/Plus4.
I am sure that it works. 
And as the instruction says - it needs only a few seconds to flash the mainboard mcu (really like 2s - but some gap to be on the save side).

Anyways: I am happy that if finally worked for you

Phil1988, 2025-06-26:
> Looks fantastic so far on this one.

Thanks - I like that you love it ☺️

I dont know if you did it a few times and formated the card in between etc..
As you now know about how to check if it workes, you can simply do it on all of your printers.
Good thing that it is super simple and fast if you keep the back cover open.

Dont forget to put the *.bin on the sdcard after every flash process (as the renamed *.CUR will _not_ flash the next printers mainboard mcu).

I hope FreeDi does not only lokk fantastic but also feels better, faster on your daily uses and does print faster with improved quality.
For the X3 series the difference is quiet noticable in every aspect :)

MrPaulAR, 2025-06-26:
>Experimental path (adventurous)
>
>Backup everything (see above).
Hit the "UPDATE" button on mainsail
If it works: awesome! Tell us in this issue.
If it breaks: also tell us – logs & details help me iron out edge cases.

I'm willing to break it for science.  Currently running 1.42 and looking to go the "adventurous" route.   However, when I click the UPDATE button in mainsail it doesn't find any new releases.  I have "channel: stable" listed in printer.cfg.  Also, should I upgrade armbian via the ui or cli (apt)?

[image]

ScheissQidi, 2025-06-27:
Hi Phil, 
FreeDi 2.0 aint good work... it's a GREAT WORK! 
THX !

I encountered so far 2 issues:
one minor, related to the guide flashing toolhead:
During flashing the toolhead on my PLUS4, the wizzard ordered to press the "reset button twice" on the toolhead board. Found right bottom corner. Flashing was done, klipper 0.13.0-154-g9346ad19 running on toolhead and mcu

major issue: "Bed leveling" macro not working, "Macro LEVEL_BED called recursively" is the only response
console G28 SCREWS_TILT_CALCULATE move the toolhead to 153,5/285/20 and stop there, holding current to 1.07A, no other response

Phil1988, 2025-06-27:
@MrPaulAR 
Please have a look at this and show the results:
```
git remote -v
git remote show origin
git config --get-all remote.origin.fetch
git rev-parse --abbrev-ref --symbolic-full-name @{u}
```

MrPaulAR, 2025-06-27:
The remote doesn't look right but maybe it is because it's showing the proper tags.  
```
mks@mkspi:~/FreeDi$ git remote -v
originCrawFish-B18https://github.com/Phil1988/FreeDi (fetch)
originCrawFish-B18https://github.com/Phil1988/FreeDi (push)
mks@mkspi:~/FreeDi$ git remote show origin
* remote origin
  Fetch URL: https://github.com/Phil1988/FreeDi
  Push  URL: https://github.com/Phil1988/FreeDi
  HEAD branch: master
  Remote branches:
    refs/tags/v1.00  tracked
    refs/tags/v1.02  tracked
    refs/tags/v1.03  tracked
    refs/tags/v1.03b tracked
    refs/tags/v1.20  tracked
    refs/tags/v1.21  tracked
    refs/tags/v1.30  tracked
    refs/tags/v1.40  tracked
    refs/tags/v1.41  tracked
    refs/tags/v1.42  tracked
    refs/tags/v2.00  tracked
  Local branch configured for 'git pull':
    master merges with remote master
  Local ref configured for 'git push':
    master pushes to master (local out of date)
mks@mkspi:~/FreeDi$ git config --get-all remote.origin.fetch
+refs/tags/*:refs/tags/*
mks@mkspi:~/FreeDi$ git rev-parse --abbrev-ref --symbolic-full-name @{u}
fatal: upstream branch 'refs/heads/master' not stored as a remote-tracking branch
mks@mkspi:~/FreeDi$ 
```

Phil1988, 2025-06-27:
@ScheissQidi 

> FreeDi 2.0 aint good work... it's a GREAT WORK!
> THX !

😄
Thanks a lot 🙏

I dont understand the first "minor" issue:
The wizard ordered to press the reset button twice, you did and it worked?!
Whats the problem?  😅

The "major issue":
Please change the both macros in the macros.cfg to:
```
[gcode_macro SCREWS_TILT_CALCULATE]
rename_existing: BASE_SCREWS_TILT_CALCULATE_BASE
gcode:
    { action_respond_info("starting screw rotation calculation...") }
    M141 S0 # disable chamber heater (see https://github.com/qidi-community/Plus4-Wiki/tree/main/content/chamber-heater-issue)
    BASE_SCREWS_TILT_CALCULATE_BASE

[gcode_macro LEVEL_BED]
description: "Measure bed and calculate screw adjustments for manual bed leveling"
gcode:
    G28
    SCREWS_TILT_CALCULATE   

```

Phil1988, 2025-06-27:
@MrPaulAR 

I dont know why, but your remote-fetch-refspec is not checking the remote origin.
Please do:
```
git config --add remote.origin.fetch "+refs/heads/*:refs/remotes/origin/*"
git config --add remote.origin.fetch "+refs/tags/*:refs/tags/*"
git fetch origin
```

Then update via the Update Manager :)

MrPaulAR, 2025-06-27:
Those git settings fixed the web-ui update.  After updating everything I'm still on armbian 24.  Should I upgrade to armbia 25 manually via cli (guessing `sudo do-release-upgrade` or similar but I'll figure that out if the answer is yes.

Phil1988, 2025-06-27:
@ScheissQidi 
No they are not. 
I just fixed it, so please compare yours carefully with what I have posted -> save and restart -> your bed leveling will work.

@MrPaulAR 
Good!
Have fun with it :) 
(About the Armbian upgrade: I do not recommend that as error potential is high - So maybe test it in this state. A major OS upgrade is out of scope for me to support :D )

To both of you:
To not spam this thread, I invite you to join discord, open discussions or a issue report on real issues.
It is not a "multi-purpose-post-all-issues-with-v2.00" thread :)

DSCustoms, 2025-06-27:
@Phil1988 This is partially an installation of 2.0 question. How do I flash a new toolhead board (adapter board) when I have to replace it? Does that old instruction about holding the boot button work?

tekknix, 2025-06-28:
Many thanks for making FreeDi ready for the Plus 4. Really appreciate your work. Unbeliveable how this private Project has grown so fast over the time. Many, many Thanks.

I've used a brand new 32GB eMMC and a fresh Micro-SD-Card to flash it to my Plus 4. Only the Flashing of the Toolhead has made me crazy (due to the really bad contact if Dupont was sitting not correct in the Pin-Holes). But i've finally made it and everything runs as expected.

Only 1 Question so far is open: I've extended (or better tried) to expand my eMMC Partition to make use of the whole 32GB with 

`sudo systemctl enable armbian-resize-filesystem`

followed by 

`sudo reboot`

On the Display i can see (zzzz, System wird neu gestartet. Bitte warten...". 

How long this maky take? 

It seems to has restarted (cause i lost Network-Access) but i'm not sure how long this may take to resize and start up again. I don't like to restart and been left with a scrambled eMMC.

Any help will be appreciated.

Regards
MoS-tekknix

tekknix, 2025-06-28:
Okay, seems that you need to PowerOff and PowerOn after it. Now the eMMC is showing up with 24G Free Space. Maybe a good idea to modify the Instructions?

[](https://github.com/Phil1988/FreeDi/wiki/Expanding-the-filesystem-(for-larger-storage-devices)

Cheers
MoS-tekknix

vibraphon, 2025-06-28:
X3-Plus: Following the installation guide everything worked fine.
When I am doing z-leveling and input shaping I always get the message:
**_Klipper is not connected
Klipper service has been stopped or manually terminated. Please use the "RESTART" command to restart it._**

After restarting Klipper I always get the same result.

tekknix, 2025-06-29:
Sorry, my fault

rodjerhell, 2025-07-05:
qidi x-plus 3.  после обновления до версии 2,0 принтер останавливается через 2 часа печати с теплой камерой. камера 60 градусов.  need help!!!

Phil1988, 2025-07-05:
@rodjerhell
This thread is for FreeDi v2.00 announcements and important/specific information only.
It already got way over that and I need to stop this.

Please ask for help on discord or a seperate thread with more information.
"it doesnt work" doesnt work to help you ;)

ChrisInMilwaukee, 2025-07-07:
Hi Phil,

Per your recommendation in the "Experimental Path", I let the Update button in Mainsail do its thing.  It appears to have updated the server-side components, but it is not reflected in the display.  TBH, I haven't read through all of the Issues posts or other resources, and I do intend to reflash the machine per normal instructions.  But wanted to let you know before I did that if there's something you'd like me to check.

Thanks for your continued efforts!
~Chris

ChrisInMilwaukee, 2025-07-07:
Hi Phil,

No, that file is not on the file system.  The machine seems to think it's on v2.0 from what I can see on the web interface, but perhaps not.  I don't remember touching the Update button for it.

[image]

Short snippet from restart script:

```
cd ~/FreeDi/FreeDiLCD && sudo ~/klippy-env/bin/python3 start.py -debug
[sudo] password for mks:
Connecting to Moonraker at ws://127.0.0.1:80/websocket
Using selector: EpollSelector
TFT baudrate detection...Websocket connected
Moonraker connection opened at ws://127.0.0.1:80/websocket
G-Code file path is: ~/printer_data/gcodes
Found z_offset = 1.929 in variable "z_offset" in section "probe"
Detected baudrate: 921600
Display response: comok 1,1281-0,TJC8048X250_011R,165,10201,1BD83401AEB0A05A,33554432-0
Firmware update failed: SerialBase.read_until() got an unexpected keyword argument 'expected'
Printer model: x-max3
Printer volume: 328x326x322
Klipper version: v0.13.0-0-g61c0c8d2-dirty
Software version: v1.42 stable
FreeDiLCD start
FreeDi sw-version sent to screen!
pushing data to tft. Last update: (0.38 sec)
data.klipper_state != self.printer.klipper_state
klippy ready
Printer state: standby
Printer in standby
Forwarded to main page
_BootPage: len: 1 data[0]: ee
ActiveScreen: = BootPage
pushing data to tft. Last update: (0.21 sec)
_NewMainPage: len: 1 data[0]: ee
ActiveScreen: = MainPage
pushing data to tft. Last update: (0.4 sec)
pushing data to tft. Last update: (0.4 sec)
...
...
```

I'll let the script go and see if it finishes, but it looks to me like it's still actually using 1.42.

[11 more replies]