Phil1988/FreeDi #260: FreeDi 2.00 Qidi Q1 Pro display does not update with changes when said changes are made thru the webUI rather than on the display. Does not update when print started thru webUI.

NoNameCola, 2025-06-28:
The Q1 Pro display does not update it's temps, or even show the print status screen if you start a print thru the webUI instead of via the display. 

It should update to show current temps, and show the print progress screen regardless of how I start the print.

Steps to reproduce the behavior:
1. Start print using WebUI
2. Display does not update with temp values or show print progress screen. 

 - Q1 Pro
 - FreeDiLCD 2.00
 - Source: FreeDi 2.00 image
 - Mainboard: Stock Q1 Pro Mainboard.

Phil1988, 2025-06-28:
It does work perfectly fine for me and many others.

Can you please provide a debug log?
https://github.com/Phil1988/FreeDi/wiki/Bugs!#are-there-any-bugs

tekknix, 2025-06-28:
Interessting, i see the same behaviour on X PLUS 4, even if i do Filament-Unload / Load or change the Nozzle/Bed-Temp, it will not shown or updated on the Display. 

Even after a complete restart (Flashed eMMC, SD-Card, Toolhead...), the issue will be the same. Will try to collect a Debug-Log

Regards
MoS-tekknix

tekknix, 2025-06-29:
Mhh, today in works without problems...

Phil1988, 2025-06-29:
@NoNameCola 

Do you have addintional software installed like shaketune?
I have seen 5 reports where this was installed and slowed moonraker start to much.

Please try this:
```
sudo nano /etc/systemd/system/FreeDi.service
```
Change the delay to 10 (or if this is not enough to 15) at the following line:
```
ExecStartPre=/bin/sleep 10
```

tekknix, 2025-06-30:
Have indeed set the Startup-Delay to 12Seconds and now regardless how often i'll start the Printer from Off-State, Firmware-Restart oder Klipper-Restart, Display is always working and in Sync with the WebGUI now.

tekknix, 2025-07-01:
Maybe due to different Speed of the used eMMC? I've not done a speedtest and from my thoughts it seems to be more as sufficient:

mks@mkspi:~$ dd if=/dev/zero of=speedtest bs=64k count=10000 conv=fsync
10000+0 records in
10000+0 records out
655360000 bytes (655 MB, 625 MiB) copied, 5.84632 s, 112 MB/s

mks@mkspi:~$ dd if=speedtest of=/dev/null bs=64k count=110000000
10000+0 records in
10000+0 records out
655360000 bytes (655 MB, 625 MiB) copied, 5.63418 s, 116 MB/s

Anyhow, it's not really a problem to set it to 12s instead of 5s, cause normally i will start the printer only 1 or 2 times a day. 

So nothing to worry about.

narfviper, 2025-07-05:
> [@NoNameCola](https://github.com/NoNameCola)
>
> Do you have addintional software installed like shaketune? I have seen 5 reports where this was installed and slowed moonraker start to much.
>
> Please [...]

Hi all,

I had the same issue on my Qidi X-Max3 with shaketune installed.
To increase the time like described above fixed it for me.

Thank you!

Phil1988, 2025-07-07:
Can someone with this problem (preferably with shaketune installed) test this and report if this works (this is one line):

```
ExecStartPre=/usr/bin/bash -c 'PORT=$(awk -F": " "/moonraker_port/ {print \$2}" /home/mks/printer_data/config/freedi.cfg); PORT=${PORT:-7125}; until nc -z 127.0.0.1 "$PORT"; do sleep 1; done'
````

kingtricky, 2025-07-12:
I would just like to add ive got the same issue. I dont have shaketune installed. Only difference i have is beacon. Sometimes power cycling the printer helps. Ive tried the above fix, ill see how it goes.

pmbroth, 2025-07-18:
Hello Phil, I noted in discord the same, that  part fan and extruder temp, and chamber temps were not updating, and with me.  Also  after about an hour the chamber heater shut down.

MicroSur, 2025-07-18:
> I cant help without debug outputs :)

I apologize. You can't just take and replace the names of the coolers, making them like in the usual stock configuration )))

MicroSur, 2025-07-19:
> [@MicroSur](https://github.com/MicroSur) So you renamed the fans to the weird stock "fan1 , "fan2" names? 😄 That does not work and the names are "suboptimal".

freedi				original & openq1

fan					cooling_fan
sidefan 			auxiliary_cooling_fan
chamber_heater_fan	chamber_fan
hotend_fan			hotend_fan
filterfan			chamber_circulation_fan
hotend_fan2			hotend_fan2
Mainboard_fan		board_fan

Phil1988, 2025-07-21:
@MicroSur 
It doesnt matter that much how QIDI or anyone else named different objects.
FreeDi has to identify the parts and as QIDI is not consistent and not clear on names over the time, you have to stick with my naming convention or use a stock system.

Phil1988, 2025-09-14:
@NoNameCola  @tekknix @chrashing @Potatonater5001 
Can you confirm weather or not it is still existent with V2.01?

narfviper, 2025-09-14:
> [@NoNameCola](https://github.com/NoNameCola) [@tekknix](https://github.com/tekknix) [@narfviper](https://github.com/narfviper) Can someone with this problem (preferably with shaketune installed) test this and report if this works (this is one line that should replace the `ExecStartPre=/bin/sleep 10` line ):
> 
> Open the file via nano:
> 
> ```
> sudo nano /etc/systemd/system/FreeDi.service
> ```
> 
> then find the line starting with
> 
> `ExecStartPre=`
> 
> and replace it with:
> 
> ```
> ExecStartPre=/usr/bin/bash -c 'PORT=$(awk -F": " "/moonraker_port/ {print \$2}" /home/mks/printer_data/config/freedi.cfg); PORT=${PORT:-7125}; while ! echo > /dev/tcp/127.0.0.1/$PORT 2>/dev/null; do sleep 1; done'
> ```

Sorry I just saw your post.

I was updating to FreeDi 2.01 2 days ago, had no problem after update.
Yesterday I installed shaketune again, without any problem so far.

As far as I can see you have already entered the line above in the FreeDi.service config.

I have done a few powercycles of the printer and never had an issue with the display so far.
I will continue to monitor the situation.

Phil1988, 2025-09-15:
Yes I included all the fixes on v2.01 already.

Very nice to hear that it all wors flawless for you even after installing S&T 👍

Phil1988, 2025-09-16:
Because of low activity of the OP and the reports that this is not existent anymore on v2.01, I am closing this :)

closed 2025-09-16 by Phil1988