Phil1988/FreeDi #261: [Help] Why the Chamber-Fan is on even if there is no Chamber Heating?

tekknix, 2025-06-29:
I'm running FreeDi 2.0 on an more or less unmodified X Plus 4. So far everything seems to went well.

Anyhow, i've two Problems.

First: 

Why the ChamberHeater-Fan get on even if there is no Temperature set for it?

[image]

[image]

tekknix, 2025-06-29:
I've seen a similar behaviour in Bambu-Lab's Configuration, where the Chamber-Heater-Fan is used (without the ChamberHeater is heating) to cool down the Motors. Is this similar to that?

3DWoodworks, 2025-06-29:
Howdy do i get unscripted. I dont want anymore emails and notifications. I
don't even run this on my xmax3 its stock

Phil1988, 2025-06-29:
> Why the ChamberHeater-Fan get on even if there is no Temperature set for it?

I included this as safety feature to cool the heater down and decrease the risk of catching fire ;)
Otherwise the heating coil would stay hot and cause problems.
If you prefer to not use it, change the heater_fan to fan_generic but I do not recommend it.

Phil1988, 2025-06-29:
@3DWoodkorks
I can subscribe or unsubscribe you.
You have to do this via the git settings or maybe by hitting "unsubscribe" on the mail you get.

Phil1988, 2025-06-29:
Well... I spend days on the printer config and macros.
A LOT has been "changed" (created from scratch) - so the machine is pretty different.
If we go through all the changes comparte to the original firmware that would take some time :D