Phil1988/FreeDi #263: [Feature request] Support QIDI Box

jaydenseric, 2025-07-01:
Add support for the [QIDI Box](https://au.qidi3d.com/products/qidi-box).

What is involved in adding support? How long is that effort likely to take?

jmpocheau, 2025-11-23:
I have a similar experience with my X-MAX3.
I was expecting some MMU to be available for my printer, but it looks that it will never happen.
So I built a boxturtle, and, thanks to Freedi, I have been able to make it work.
The integration work is still in progress.
If somebody is interested, I also can share my experiences.

jmpocheau, 2025-12-26:
> > If somebody is interested, I also can share my experiences.
> 
> Sure mate! Please check out this and if suitable post it there or open a discussion for the BoxTurtle <3

Hi, Phil, at last my BoxTurtle is working well, but the integration has been a lot of work.

I have opened a discussion in order to share my experience with anybody who could be interested.

Minzdraw, 2026-03-25:
@Wazzup77 Hi. I'm very interested in developing compatible software for the Qidi Plus4 and Qidi Box. I can't send you my Qidi Box, as I'm most likely located on another continent (Russia). But I can be your partner, provide access to your computer, and perform the necessary tasks to assist with software development.