Phil1988/FreeDi #266: Z offset won't go over 1.549

gptech2444, 2025-07-04:
I've tried using the Z offset setup but it sets the height too high. 1.349 
I need it to be 1.649 but the config file won't save any changes above 1.549.

Phil1988, 2025-07-05:
Thanks for sharing your experience with the Z offset setup.

To help you and others better, could you please update the issue title with relevant tags and include your printer model in the message (also for the next time)? 
This information is important to diagnose the issue more effectively.

From what you describe, it sounds like this behavior might be related to how Klipper or Mainsail handle configuration saving, rather than something specific to FreeDi. 
FreeDi itself does not manage the config file saving mechanism.

If you haven’t already, I recommend checking the Klipper and Mainsail documentation or forums regarding config saving and babystepping behavior, as there are known quirks depending on firmware versions.

If you can provide more details or logs, I’d be happy to take a closer look and see if there’s anything on the FreeDi side that can be improved.

gptech2444, 2025-07-05:
No problem, I was just getting frustrated that it wasn't going smoothly. 
Maybe there was an update installing in the background I wasn't aware of and that was stopping the config file from saving.
I've installed klipper on previous printers I've owned and haven't had this issue.
So far the QIDI is printing great, as good or better than before.

Phil1988, 2025-07-08:
I changed the title for you now and I had to search and look up your printer here:
https://github.com/Phil1988/FreeDi/issues/265
**QIDI X-Plus3**

For the sake to save my time, please do these things on your side next time please 🙏

closed 2025-07-08 by Phil1988