Phil1988/FreeDi #270: [Help]

Dxjpusa, 2025-07-05:
Hi
I tried to flash my X-plus 3 with the following directions with the new software. 
https://github.com/Phil1988/FreeDi/wiki/Installation-guide-for-fresh-systems#flashing-the-display

Using a new 32g emmc.  I backed up all my files and kept the original emmc out of the printer.

 It was a fresh install over the stock MCU.  I wasnt able to get it to connect in klipper.  I updated the MCU serial as described ( the config doesn't show (#[mcu MKS_THR]) as described.  The config shows it as ( [mcu Toolhead]).  I checked in the devices and only the host is working.  I tried to flash the mother board with the bin file noted in this section.

 https://github.com/Phil1988/FreeDi/wiki/Tour-%231:-Installing-FreeDi

The .bin file never changes to .cur, and the printer is idle with the mcu connection error.

Klipper reports: ERROR

mcu 'mcu': Unable to connect
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Error configuring printer

Is there a procedure to reflash the unit.  I tried both 32g and 156mb SD cards as some motherboards wont work with a large card.

Any help would be greatly appreciated.  Thank you

[klippy (4).log](https://github.com/user-attachments/files/21072164/klippy.4.log)

Phil1988, 2025-07-05:
Hey @Dxjpusa 

welcome on board.

First off, please give your topics clear and self explainings names - "[Help]" only will make noone to find anything 😉

It looks like you switched betweeen the v1.xx and v2.xx install guide.
The installation process is entirely different and I recommend you are using the v2.00 only.

You can see it on the right side at the wiki:
[image]

For your problem: Please make sure you have formatted the microSD correctly and try again.
You can see how at this section:
https://github.com/Phil1988/FreeDi/wiki/Tour-%231:-Requirements#2-microsd-card

I strongly recommend you are reading the whole install guide for v2.00 again just to make sure you havent missed any additional steps or simply doing it again from scratch :)

I hope this clears things out for you and the installation does work perfectly now 😊

Phil1988, 2025-07-08:
@Dxjpusa Is this still a thing or is it solved for you?

Dxjpusa, 2025-07-08:
I was still having problems after reflashing the emmc.  I wasn't able to get the serial from ssh.  I put the .bin file on  a sd card and put it in the MCU turned the machine on and it loaded up.  I noticed when going through the menu the first time it would show the tool head installing the software and it showed the serial, but it never showed the MCU flashing.  Im able to get the serial now.

Its all working now, and prints.  I'm still working through a few quirks on the screen.  If I start a print from the machine then everything works on the screen.  If I send a fie from mainsail it will print but then it shows the last file thumbnail on the screen.  

The freedi repro is dirty but I had run out of time.  Ill get a backup of everything and update it.  Moonraker needs an update, but everything else seems fine.

Thank you

Phil1988, 2025-07-09:
"Moonraker needs an update" - the update manaer only shows you that an update exists. It doesnt **need** that.
You dont have to be always on the latest version as it quite often even introduces issues.

I think this issue is solved for you now, isnt it?

For other things, please open a new thread and see if there might be a matching one already open.
Not sure, but maybe this is related?
https://github.com/Phil1988/FreeDi/issues/260

Just as it is already hard for me to follow the sometimes mixed questions/issues ;)

closed 2025-07-09 by Phil1988