Phil1988/FreeDi #272: [Help] Wifi Connection screen displays no available connections

esheffield3, 2025-07-05:
I followed the installation guide to [Step 5](https://github.com/Phil1988/FreeDi/wiki/Tour-%231:-Installing-FreeDi#5---meet-the-wizard-) and am hitting a snag when it gets to the Wifi connection display. Currently it's not showing any available connections but I can tap on the space where they should be and a password screen shows up. I'm not sure if it thinks they're being listed and I'm actually tapping on invisible networks or what is going on. 
I had tried an earlier version of FreeDi a month ago but I discovered my ethernet connection just doesn't work (we had a bad storm and I had a faulty surge protector that ended up frying several components to the printer, the ethernet seems to be one of them), I ended up reinstalling the original firmware. So I was eager to see that I could download this version using wifi. The wifi was previously working as I was just uploading and printing wirelessly last night.

[image]
[image]

Phil1988, 2025-07-06:
Stock wifi dongle?

If its not listed and you click there, you will be forwarded to input a password for a wifi that doesnt exist :D
So its out of function.

You can try to click the refresh button.
If it doesnt work you may need to skip it.
1-2 guy reported, that the real network screen worked, while the wizard menu doesnt.

esheffield3, 2025-07-06:
It is the stock wifi dongle.
Same issue, I've tried refreshing and nothing appears to happen.

[image]

> Stock wifi dongle?
>
> If its not listed and you click there, you will be forwarded to input a password for a wifi that doesnt exist :D So its out of function.
>
> You can try to click the refresh button. [...]

Phil1988, 2025-07-06:
@esheffield3 
Did you install from scratch or update?
I am asking because my image has the driver for your module installed - on the other hand on an update the driver shouldn't have changed.

By the way, the Tenda Model: W311MI is equipped with at least 2 different wifi modules. Source: The both modules I have on my desk ;)

esheffield3, 2025-07-06:
(I'm a real novice at github so I didn't intend to close it, my bad.)

> So I was able to successfully connect the ethernet and that works (I guess Qidi's firmware was doing something weird before). The wifi still doesn't work as it should however but I can at least connect to it now.

esheffield3, 2025-07-07:
No, the wired connection works but the wifi does not. 

It's not necessarily as big of a deal now since the printer is operational, I just can't really move it in the future if the wifi doesn't work.

esheffield3, 2025-07-07:
[wifi-log.txt](https://github.com/user-attachments/files/21111513/wifi-log.txt)

Phil1988, 2025-07-08:
Can you please check:
```
ip a
```
and
```
iwconfig
```
and
```
lsmod | grep rtl8xxxu
```

esheffield3, 2025-07-08:
[wifi-log-2.txt](https://github.com/user-attachments/files/21132205/wifi-log-2.txt)

esheffield3, 2025-07-10:
Sorry, I tried to clean it up a bit. I'm not really sure why it's exporting it weird, I'm not super knowledgeable on SSH but hopefully this is better.

[putty3.log](https://github.com/user-attachments/files/21153920/putty3.log)

If this doesn't work out, I do still appreciate you helping out. I do really enjoy the firmware otherwise! I don't really see myself moving it anytime soon so if it needs to have a wired connection it's not really a big deal.

esheffield3, 2025-07-17:
@Cletis234 yeah I press it several times and it doesn't seem to change anything.

zuvyy, 2025-08-17:
@Phil1988 no worries, WIFI isn't the primary necessity in 3d printing ;-)
you are right - I do use a complicated 16 char password including a backslash :-\
I investigated on that already as I read about problems other linux users have with special characters in passwords. the provided solution was to save the passphrase in a cfg file which I tried without success. I didn't go so far to change my password as there are a lot of devices connected. 
unfortunetely I m on a business trip for 2 weeks now and can't play around with my good friends QIDI and klipper and freedi,
so once I'm back I'll change that password and give you more debug info. btw I suspect a few users here aren't providing good debug info because the link that you're providing is broken. it's due to the ! exclamation mark which is always outside of the link but seems to be part of the URL - so I'm gonna remove the backslah from my wifi password and maybe you wanna remove the exclamation mark from your link ;-)  --> https://github.com/Phil1988/FreeDi/wiki/Bugs! <--

and one thing that was bugging me, I couldn't find a way to deactivate wifi at all. so just in case my research fails, I'd love to know how to deactivate wifi networks, permanently disconnect from a certain SSID or whatever changes that behaviour of the network manager constantly trying to connect. 

thx so far and keep up the good work, cheers

zuvyy, 2025-09-06:
I finally found the time to jump on the WiFi issue and solved it after 5min. I'm feeling stupid to tell... I just entered the wrong password :-D so sorry for all the fuzz at a moment where you needed nothing as stupid as that. Well I learned a lot and I won't confuse 0 with O in the future.
thx for all the effort again!