Phil1988/FreeDi #273: [Bug] Previous MCU 'mcu' shutdown: Missed scheduling of next digital out event

breathless19, 2025-07-05:
**Describe the bug**
Mid print, I get the following error message, and the print fails: 

**Klipper reports: SHUTDOWN

Previous MCU 'mcu' shutdown: Missed scheduling of next digital out event
This is generally indicative of an intermittent
communication failure between micro-controller and host.
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown**

I have previously (with the new 2.00 firmware) completed the same exact print, so it is apparently intermittent.   Upon rebooting, I ran the debug (below)

The only thing I have installed on this printer is a webcam, and a BTT 2.0 filament runout sensor (both confirmed working). 

**Screenshots or picture**

[image]

**Console output**

[image]

 - Printer model: X-Max 3
 - X3seriesLCD firmware version: latest
 - Source: FreeDi image
 - Mainboard: stock

Phil1988, 2025-07-06:
Thanks @breathless19 for the detailed report.
Can you please check the klipper versions of the subsystem on the MACHINE tab?

Have you printed higher temp material like ABS/ASA/PA etc?

What you are seeing is caused by communication error between host and mainboard MCU.
As they are hardwired it can only be 2 things that I can think of:
1. Stability issues by the host (CPU/RAM - overload or high temps) - this can be checked via htop (CPU/RAM) and checking the graph on Mainsail (temp for CPU (and stepper drivers)).

2. Klipper: Some klipper changes results in issues that cause stability issues like this.

The klipper shutdown is never linked to FreeDi per se.
Maybe the klippy.log does also show the cause. 
As I am no expert with this, folks at mainsail/klipper discord will be a way better to help you out there.

Lets stay in touch while finding the cause for this.

Phil1988, 2025-07-07:
The klippy.log looks good from my side.
Was it the one from the date where the error happened? 

Only a minor thing, but you have untracked klipepr files:
Untracked files: klippy/extras/auto_z_offset.py, klippy/extras/freedi.py, klippy/extras/freedi_hall_filament_width_sensor.py
Not sure how you managed to do that, but you can run the install script that should clear this out:
```
cd ~/FreeDi && ./install.sh
```

Except of this I can not see anything wrong.
But I am also no expert on klipper logs 😅

Phil1988, 2025-07-08:
Looks good!
Check 
```
cd ~/klipper
git status
```
- it should be clear now.

Besides that, I can only ask you to check keep an eye open for this to happen again - and if it does, ask the klipper guys as this is not a FreeDi bug.

I would love to help you better, but unfortunately I cant.

I will close this issue, but it would be nice if you can update this if you found a solution so others that might have the same issue knows the solution :)

breathless19, 2025-07-10:
Just tried the same print again, and got a different message, again about the MCU, but this time about the toolhead: 

[image]

I checked and my baudrate is already at 500000.   Is it feasible that a further increase of the baudrate could "fix" this?   I noticed that this was mentioned in your 2.00 release info about for some people needing to go from 250k to 500k....