Phil1988/FreeDi #290: [Announcement] Warning about a fundamental issue on FreeDi

Phil1988, 2025-07-16:
Hello everyone!

There are currently a few reports about affected X3 series owners with freezing systems mid print.
(See here https://github.com/Phil1988/FreeDi/issues/288).

I didnt had that problem on my tests before releasing v2.00 on my X-Max3 - but there definetly is "something" not working for some users.
Unfortunately I am gone for a while (see https://github.com/Phil1988/FreeDi/issues/283) and I cant track this down.

For now, I can only give all of you a warning:
Do not update or install FreeDi v2.00 if you need to run your printer for important projects!

As soon as the stress reliefs I will jump into that and search for a solution.
Until then I can only ask that the experienced users to check for the reason and help others by your findings.

Sorry for the inconvenience but I want do make sure you are informed as best as possible about this!

jawzx2, 2025-07-20:
I have had a random freeze with my X-Plus-3 at least twice since updating to 2.00. I had attributed it to possibly "flakey" toolhead firmware, as I am also one of the ones that experienced issues flashing my toolhead and had to use the flashnuke method and the special firmware... However, I have *NOT* used my printer extensively since the update and don't have much info to contribute. I have made several successful prints, one of them longer than 5 hours, and one of the freezes occurred when the printer was sitting idle, and not while printing, if that info helps.

ScheissQidi, 2025-07-22:
X MAX 3, setup worked flawless, PID tuning, input shaping, bed leveling, bed PID, z offset, but no chance to get my logitec cam working. 
1st print, orca PA test, screen freeze, no alteration of values, in moonraker al looks fine, print starts and after a few lines, printer stopped, klippy not connected. Booting up after an hour, printer shut down, klippy not connected.
So, this was the last chance for this creepy qidi hardware, I ll change mainboard to a fly or original mks

xTremePower, 2025-07-30:
Hello all, 

I had this problem even before FreeDi on my xmax3, the problem was a faulty USB C cable going from the motherboard to the head, Be careful it's not a basic USB C cable, don't try Amazon or other to find one and don't use other cables, Qidi sent me one new in warranty (of course don't say you flashed the firmware)

But it was also caused on my first xmax3 by the lack of mini fan inside the toolhead, 

But as sad by @zjuergen66 the temp sensor often report crazy temps without any reason ....
I will see how to push this temperature up to 150 of something, I don't think even at 150 C it can't cause fire or something ... but I will not share the method, it's my own test, and I take the risk myself (I have fire alarms in case of)

jawzx2, 2025-07-31:
I've successfully made maybe 20 prints under 2.5 hours each, sometimes one right after another since the upgrade to 2.00 on my X-Smart-3. I even successfully completed one 6+ hour print. Today I tried a 4 hour print, somewhere around the 3 hour mark I got the mid-print-freeze and black screen again. :( I guess it really does still exist.

Potatonater5001, 2025-07-31:
I believe @zjuergen66 may be onto something here with toolhead temps.

The RPI foundation somewhat recently approved for the rp2040 to officially support 200MHz, and it seems klipper v13 actually implemented this and runs rp2040s at 200MHz.

I am seeing very high temps just barely below the datasheet specified maximum of 85C for the rp2040 on my toolhead while printing.

Perhaps this issue is only appearing on v2.0 because of the fact these newer klipper versions are running the rp2040 at 200MHz, which is pushing many of our X3 machines (with awful toolhead board cooling) just past this thermal limit and causing the emergency shutdown.

I have drilled a bunch of holes in the rear toolhead board cover on my xplus3 for cooling, so far temps are lower and I will report if the freezes continue after this "mod".

Also, thank you @Phil1988 for this project, it has made my xplus3 so much better and finally hackable. Your work is great and appriciated :)

pmbroth, 2025-08-01:
> > My x-max 3 has the fan in the back cover..
> 
> Do you have the crashing issue?
> 
> If so, try printing abs or asa with chamber heated and watch the toolhead temp in mainsail. Maybe the little fan they added on your model still just isnt enough for 200MHz.
> 
> Wouldnt shock me considering I am still seeing around 80C even with big holes drilled in the cover AND an actual heatsink stuck on the rp2040.

Hello, I will have to print asa again next week when i’m back in town,.  The last time, I printed asa, my system did crash, and the chamber heater shuts down after about an hour.  I have run long prints, but with pla, and it worked fine.  Brian

MidbrainMan, 2025-08-01:
I'm not sure it's the RP2040...just had another shutdown but this time about 5 minutes into my print. They seem to be coming more and more frequently as time goes on. I'm thinking less and less that it's a thermal issue. Attached the log from my most recent shutdown, looked like my RP2040 stayed below 70C the whole time, and that's without modifying for any cooling/heat sink (just stock setup).

[klippy(1).log](https://github.com/user-attachments/files/21540095/klippy.1.log)

Potatonater5001, 2025-08-01:
> I'm not sure it's the RP2040...just had another shutdown but this time about 5 minutes into my print. They seem to be coming more and more frequently as time goes on. I'm thinking less and less that [...]

hmmm... yeah none of your temps are anywhere near danger zone.

Phil1988, 2025-08-01:
Hello,

as there are quite a few users affected, I am pretty sure this not related on a defective toolhead cable.

I also dont think it is related the Mainboard_STM32F402 temp readings.
Sure it could in theory invoke an emergency stop, but no system freeze.

A totally different Armbian OS version has this error too.

I think @Potatonater5001 has put up something very important to the focus (good job):
**The different integration of the RP2040.**
I know for sure, that extensive I/O usage can freeze the system (that happens occationally when updating the screen firmware).
Maybe the higher clock speed of klipper v0.13 results in something similar to klipper/the sytem.

**So far I am not sure if any Plus4 owners are affected - is there anyone?**
If no Plus4 owner has this problem, it must be something related to the toolhead because this is the relevant intersection between those machines affected and not.

Thanks for your ideas and thoughts about possible causes @xTremePower @zjuergen66 

@PaleoW We have a discord subchannel for FreeDi here:
https://github.com/Phil1988/FreeDi/wiki/Community#-join-us-on-discord

Phil1988, 2025-08-02:
@zjuergen66 
Emergency stop: The printer stops immediately, klipper needs a restart to make it operational again, system stays responsive.
System freeze: The printer stops immediately and the system is not responsive at all - no WebUI/ssh access any more. Reboot required.

MidbrainMan, 2025-08-02:
@Phil1988 See my post in issue #288. I've had success (no failures for x2 days now) by changing the step_pulse_duration from the stepper motor section of the printer config (for steppers X, Y, Z & Z1) to 0.000002 from the current 0.0000001. Big caveat here, this was a solution proposed by AI, so take it with a grain of salt in terms of it being the root of the issue vs a temporary "band-aid" solution. I'm not savvy enough to discern if the solution is a good one, but so far the results seem positive...

jawzx2, 2025-08-02:
@MidbrainMan I just went to check my configs on my X-Plus-3 and all my stepper pulse durations were already set to 0.000002. So that unfortunately counts it out as a solution to my crashes. I have been runing a very skeletal toolhead back panel I found on Printables, or maybe it was Makerworld? In any case, I'm adding a heatsync to the rpi on the tool head, I had one crash happen when the printer was sitting and not even printing, but the others I have had have all been longer prints with PETG or ASA. There's actually a version of the back cover I'm using that has a mount for a 30mm fan, but I chose not to use it on the idea of keeping the moving mass as low as possible... Maybe I'll print one up and give it a try.

MidbrainMan, 2025-08-05:
@nefelim4ag @Phil1988 Thanks for the input, this thread seems to have split the conversation slightly as it seems we're discussing the same issue as in bug #288 (that's my fault too). I'll respond here though as it seems most appropriate.
I've run nefelim4ag's suggestions by the LLM again and it remains convinced that the step_pulse_duration is the root of the problem. Again, my results have been positive (although anecdotal) with x2 successful 12 hour prints which had previously been unattainable on this printer (Q1 Pro). This is the counter point it provided:

[image]

[image]

I want to reiterate that I feel strange to be pitting an LLM against capable klipper dev's opinion, especially since I don't have any helpful knowledge in this area...however I'm doing it anyway because my results have been promising and I think potentially dismissing this as a solution may delay our ability to resolve the issue. 
I will update at the first sign I see of my printers getting the black screen issue again even with the modifications to the step_pulse_duration.

Paradisebaker, 2025-08-05:
> 
> **So far I am not sure if any Plus4 owners are affected - is there anyone?** If no Plus4 owner has this problem, it must be something related to the toolhead because this is the relevant intersection between those machines affected and not.
> 

I am a Plus4 owner, flashed Freedi last weekend, everthing seemed fine, but this morning I found that my printer was dark (leds off), the screen unresponsive and dark, the lan switch port said "connected" but there was no mac address visible, could not reach the device at all. Powerswitch was on and the integrated light on.
Too bad this happened mid-print.

So, to answer your question, it seems that the Plus4 is also affected.
Let me know if I can help somehow with troubleshooting.

BR
Markus

Phil1988, 2025-08-05:
@MidbrainMan and anyone else:
To help keep the announcement thread clear and focused on important updates, please post any error reports, detailed discussions, or troubleshooting about this issue in the dedicated [bug report thread](https://github.com/Phil1988/FreeDi/issues/288#issuecomment-3148656294). 
This announcement thread is meant only for high-level information and official updates, so everyone can quickly find the latest news.

Thank you for your understanding and cooperation 😊

akschu, 2025-08-25:
Things back to normal again?  I'm thinking about diving in!

akschu, 2025-08-25:
Outstanding, thanks!

LifeDay, 2025-09-03:
I just had a new Plus 4 delivered yesterday. If you're looking for a fresh machine to try it out on, let me know. Otherwise, I'll hang tight for the main release.

Thanks for keeping this project running!

jawzx2, 2025-09-13:
Thanks to Phil and all the testers for finding a solution! What's the recommended upgrade path from 2.00? Full flash of the emmc or can it be performed in place with either the moonraker console or system cli?

akschu, 2025-09-14:
@Phil1988  Just wanted to report back with my experience.  I installed FreeDi using the instructions at https://github.com/Phil1988/FreeDi/wiki/Introduction and noted a few things:

1.  The first boot hung on the boot screen, but I figured out how to ssh in and restart.
2.  My toolhead didn't update in the wizard.  (X-Max3) Thankfully, I found found the procedure to update it at https://github.com/Phil1988/FreeDi/wiki/Installation-guide-for-fresh-systems#on-freedi-v140-and-later and after I updated it, it worked.

Since you already break out toolhead updates for the other printer models, I wonder if it doesn't make sense to remove the x-max updater from the wizard and just make everybody update the toolhead manually.

Anyway, printer is up and running, and initial calibration prints look great.  This is truly great software that I'm happy to support.  50 euros your way.

jawzx2, 2025-09-15:
I too had difficulty getting my toolhead up and running on the change to 2.01 from 2.00. After a lot of fiddling with manual flash nuke and manual firmware install, including clkdiv4 and clkdiv8 versions from the testing thread , with no success, I remembered that I had used a "SPECIAL_klipper_v0.13.0-154.uf2" from issue #265 previously.

Manually flashing that toolhead firmware solved my issue, and my xplus3 is up and running now. I am a little confused as to what exactly the compile options that made that firmware work were, and do not entirely understand if my toolhead will require special firmware going forward, or if this is purely an issue with the version of Klipper FreeDi is currently using.

I would have expected that choosing "skip, next" on the wizard would leave the toolhead firmware unchanged, but it appears that *something* changed to cause the toolhead to fritz out on the update to 2.01. After the wizard, in which I chose skip and next at every step after language selection, the toolhead couldn't be found and was not appearing as a serial device when using <ls /dev/serial/by-id/*> Putting it in boot mode with the buttons worked reliably, but only flashing the "SPECIAL" firmware returned it to a functional state. I *am* comfortable with ssh and the terminal, but I'm not an experienced coder and pretty much all my microcontroller experience is in 8 bit mcus in the Arduino IDE, so this is a bit out of my depth.