Phil1988/FreeDi #296: [Bug] Wrong display model

timniklas, 2025-07-23:
**Describe the bug**
I own a X Plus 3, when following the tutorial everything except for the display works fine.
Flashing the display manualy via USART HMI seems to reveal the issue:

Expected display model: TJC8048X250_011
My display model: TJC8048X570_011

Is somebody else having this problem?

**Expected behavior**
If not self self-explanatory, add a clear and concise description of what you expected to happen.

**To Reproduce**
Steps to reproduce the behavior:
1. Download USART HMI
2. Click on green upload button
3. Select tft file
4. See error

**Screenshots or picture**
[image]

**Base:**
 - Printer model: X-Plus 3
 - X3seriesLCD firmware version: none
 - Mainboard: stock

timniklas, 2025-07-25:
It seems that I've bricked my screen, I am unable to flash the original firmware aswell.
Im just ordering a new one...

Phil1988, 2025-07-25:
Bricking is pretty hard - if not hardware damaged by a short etc..

X5 and ~13cm seems to be the TJC8048X550_011 - a 5" screen.
And this came in your stock (and untouched) Plus3?

I guess its still usable/flashable with the right firmware and via SD card.

Phil1988, 2025-07-25:
Did you try to flash the stock image?
https://github.com/QIDITECH/QIDI_PLUS3/blob/main/UI/800_480_PLUS.tft

Make sure to have a FAT32 formatted SD card with no hidden partition.
Here is a how-to:
https://github.com/Phil1988/FreeDi/wiki/Tour-%231:-Requirements#2-microsd-card

timniklas, 2025-07-26:
Already tried that, yes.

Phil1988, 2025-07-26:
Have you contacted the QIDI support if this screen model is intentional?

timniklas, 2025-08-05:
The qidi support sent me a lik to a new display which arrived today.
Flashing went fine this time, but I cant click anything in the installation wizard.
Im stuck with "Next", but Im unable to click it.

[image]

timniklas, 2025-08-06:
Thanks for you hard work, its working fine now! 🥇 
I was able to fix the problem by reinstalling the display. The padding on the case was pressing against the display and blocking the touchscreen.