Phil1988/FreeDi #298: [Bug] Xmax3 Chamber Current and Target temperatures display 0s instead of actual values.

chrashing, 2025-07-29:
Issue: 
For the printer display: Chamber Current Temperature is always 0. Also the Chamber Target Temperature always shows 0 unless the target value is updated through the printer display screen. The Chamber target temperature display does not update when the value is changed in moonraker. The printer/klipper knows the chamber current temperature and functions appropriately.

Attached is a picture to show that Chamber Current and Target temperatures values are shown on the moonraker display while the printer display 0s for the values.    

[image]

Attached is the debug data from the debug script.

[ChamberTargetNotDisplayed.txt](https://github.com/user-attachments/files/21478868/ChamberTargetNotDisplayed.txt)

 Case: Chamber Target temperature updated through the Printer Display: 
   
    Chamber Target  temperature on printer display updates correctly. 
    Chamber Current  temperature on printer display remains 0 in ERROR.  
    Chamber fan/heater functions as requested by the target update.            
    Moonraker shows updated target, and current temperature properly

 Case: Chamber Target temperature updated through Moonraker application:

    Chamber Target  temperature on printer display remains 0 in ERROR
    Chamber Current temperature on printer display remains 0 in ERROR.
    Chamber fan/heater function as requested by the target update. 
    Moonraker shows updated target, and current temperature properly

 So on my Xmax3, 
    The Chamber Current temperature on the printer display is always 0.  
    The printer displayed Chamber Target temperature is only ever updated from 0 when the value is entered through the printer display.
    Moonraker Chamber changes are NOT propagated to the printer display.    
    Changes to the Chamber Target temperature from gcode commands during printing are NOT propagated to the printer display.  
    Heat Bed and Nozzle Target and Current Temperatures are always correctly displayed regardless if they are entered from gcode, moonraker, or the printer display. 

Let me know if anything more info is desired to resolve this. Meanwhile the printer is perfectly usable.

chrashing, 2025-07-29:
**Additional infor about my Xmax3**

My Xmax 3 is stock and the config files have limited minor tweaks. The only hardware change is my filament sensor signal is inverted from the original. 

 **filterfan definition from printer.cfg .**  

## Activated charcoal blowing fan
[fan_generic filterfan]
pin: PC9
max_power: 1.0
cycle_time: 0.0100
hardware_pwm: false
kick_start_time: 0.200
off_below: 0.3
shutdown_speed: 0

**Below list all changes from stock freeDI v2.0** 

FreeDI V2.00 Changes 
---------------------

sudo nano /etc/systemd/system/FreeDi.service
ExecStartPre=/bin/sleep 10       # Increase from 5 to 10, provides more time for moonraker to start before freeDi.
                                 # I do not know that is was necessary, Saw this increase being recommended on
                                 # Discord for problems on other machine types, and thought as insurance,
                                 # it would not hurt.  
-----
printer.cfg set heater_bed_pwm_cycle_time to 0.016 for 60hz

printer.cfg set stepper_z position_min from -2 to -3 to allow Z offset to be set on my machine.

printer.cfg Inverted filament sensor from !PCi to PC1 because my detector is inverted.

printer.cfg Changed mainboard fan target_temp to 20 so fan is always on.

macro.cfg  CHAMBER_TEMP_CONTROLLER macro
           changed M106 P3 S0 to  M106 P3 S127  to set min filter fan speed 50% because I use a Filter.
                                               
macro.cfg  {% set home_all_final_position_z = 6 %} ; Changed from 20 to 6 to better heat the Z probe before printing.

macro.cfg  LOAD_FILAMENT macro
           Changed G1 E40 F600  to G1 E60 F300 to 1) stop extruder from slipping (clicking) and 2)extrude a little further.
           M104 S0      ; Added to end of Load_Filament to have extruder cool after loading. 
--------------------------------------------------

Phil1988, 2025-08-02:
Hey @chrashing 
could you check if this is related to this? https://github.com/Phil1988/FreeDi/issues/260 

I am back again and just checked the reports.
For me it seems as these issues are related.

I suggest you use this instead the fixed 10 seconds on the FreeDi.service:
https://github.com/Phil1988/FreeDi/issues/260#issuecomment-3045557968

Regarding your debug output:
Thanks that you have attached that and collected all the data for me.
This helps a lot.

Can you please leave the debug console open a bit longer and use the printer for the reported bug?

Eg: Change the values at Mainsail and leave the debug console open so we can see if it recognized the changes (and tell me if the changes have been shown or not on the screen).
Then do it from the screen and check if the values have been forwarded to mainsail.
Then copy and attach the whole debug output here so we can follow what is working and what not :)

closed 2025-08-03 by Phil1988

V3DPrinting, 2026-01-07:
I have the issue on 3 X Max 3 running FreeDi 2.07 and 2.08 on custom OS : Actual temperature of the chamber is not displayed, target is.

FreeDi is only used for the touchscreen. Klipper is Vanilla.

Naming is matching (and was working fine with FreeDi 1.x)

OS : Armbian-unofficial 24.11.1 bookworm
Klipper : v0.13.0-399-gedaa61471 or v0.13.0-375-gba79d72f or v0.13.0-374-g99c0bfca4
Packages installed

[image]

[image]

KlipperScreen is not active.