Phil1988/FreeDi #299: Qidi 1 Pro FreeDi 2.0 Probe triggered prior to movement - SDCARD_RESET_FILE cannot be run from the sdcard

AxelSat, 2025-07-31:
Hi,
everything bigger than 80X80 mm ends with Probe triggered prior to movement - SDCARD_RESET_FILE cannot be run from the sdcard and the print job stop.

11:19
SDCARD_RESET_FILE cannot be run from the sdcard
11:19
SDCARD_RESET_FILE cannot be run from the sdcard
11:19
Probe triggered prior to movement
11:19
Probe triggered prior to movement
11:19
Adapted probe count: (8,8)
11:19
Ratio: (0.9523809523809523, 0.9302325581395349)
11:19
Adapted mesh bounds: ([22.5, 22.5],[222.5, 222.5])
11:19
Original probe count: (8,8)
11:19
Original mesh bounds: ((20.0, 15.0),(230.0, 230.0))
11:19
Found 1 objects

[Qidi_1_Pro_Debug_Log_31-07-2025.txt](https://github.com/user-attachments/files/21527110/Qidi_1_Pro_Debug_Log_31-07-2025.txt)

nimon, 2025-07-31:
Change sample_retract_dist in printer.cfg to a value of 5(which is stock), it fixes the probe fault

Phil1988, 2025-08-03:
@AxelSat the suggestion of @nimon is more a less a workaround that I do not recommend.
If your probe is triggered prior to movement it means that either your probe offset is not on point or your bed is not leveled/flat.

I recommend doing my recommended "**First Steps**" (in that order) that you find at the wiki.

Reason is that with this workaround the bed meshing time increases and the print quality is a bit worse with an unleveled bed.

AxelSat, 2025-08-03:
> [@AxelSat](https://github.com/AxelSat) the suggestion of [@nimon](https://github.com/nimon) is more a less a workaround that I do not recommend. If your probe is triggered prior to movement it means that either your probe offset is not on point or your bed is not leveled/flat.

Ok thanks' for the hint; 

> I recommend doing my recommended "**First Steps**" (in that order) that you find at the wiki.

Actually I've done that after the installation, went thru all the steps, but end up with the problem shown on topic, surely a mistake i've made was, levelling without heating it up the bed first, releveling afterwards but at 60 °c, but that didn't solve the problem.
After reading Nimon Hint's, I went thru the pinter.cfg and found out that one problem was the horizontal_move_z:  under the besh mesh section.
Your value was 2.6 but my z_offset is 2.7
So I've raise horizontal_move_z: 3 and that solved the problem.

I think I have a very bad Bed
These are the values after leveling the bed

[image]

and this i the result of the bed mesh

[image]

> Reason is that with this workaround the bed meshing time increases and the print quality is a bit worse with an unleveled bed.

Just a question Phill if I may.
Since the mesh is set to 230X230 it would be impossible to print something 240 x240 right, or even here I'm mistaking.
as an example I wanted to print this to check my first layer quality

[image]

the result is this again 

15:59
SDCARD_RESET_FILE cannot be run from the sdcard
15:59
SDCARD_RESET_FILE cannot be run from the sdcard
15:59
Probe triggered prior to movement
15:59
Probe triggered prior to movement
15:59
probe at 222.340,226.129 is z=2.878750
15:59
probe at 222.340,226.129 is z=2.886250
15:59
probe at 222.340,195.340 is z=2.751250
15:59
probe at 222.340,195.340 is z=2.751250
15:59
probe at 190.922,195.340 is z=2.911250
15:59
probe at 190.922,195.340 is z=2.906250

I've remove some line to shorten the list..........

15:56
probe at 33.820,10.602 is z=2.890000
15:56
probe at 2.399,10.599 is z=2.855000
15:56
probe at 2.399,10.599 is z=2.873750
15:56
Adapted probe count: (8,8)
15:56
Ratio: (1.0, 0.9579911111111111)
15:56
Adapted mesh bounds: ([20.0, 15.0],[240.0, 230.548])
15:56
Original probe count: (8,8)
15:56
Original mesh bounds: ((20.0, 15.0),(240.0, 240.0))
15:56
Found 5 objects

So the only way i found to print that, was to move the outside squares living more space on the outside.
I'm i still missing something here ? i know that most of the time I wont print big things or filling up the plate, 
but shouldn't we have the ability to print a size of 245x245 ?

Thank you Phill for you time a patience.

Phil1988, 2025-08-03:
I see.
The z_offset of 2.7 is quite high compared to my ~1.5 on the Max3 and Plus4.
Plus you are having some respectable bending of 0.3mm.

So for you you need to make this slight adjustment.

For your build volume:
Your bed is very tilted on the front edge and the back right.
When the probe tries to mesh it, the bed is too high and the probe is already triggered before it would go down and measure it.
So it aborts.

You can increase the horizontal_move_z a bit more to 3.3 and see if this is enough.
You can not do much about such a bend bed:
Either adjust the height based on "the best compromize" do make it as even if possible even it its then tilted a bit (meaning to raise the left backside a bit in your case - even if its not the perfect calculated bed level)
OR 
you have to flatten your bed (some folks use adhesive tape etc. to adjust it)
OR
you need to buy a new bed.

So the easiest way is indeed to do the software workaound - but I wanted to give you some more information :)

Btw: My name is Phil (with one "L") 😉

AxelSat, 2025-08-03:
> Btw: My name is Phil (with one "L") 😉
Forgive me Phil.....

Thanks' again for your quick reply and hints Phil, now things are more clear, very kind of you given me all the alternatives on how to 
approach and solve the problem.

Best regards.