Phil1988/FreeDi #309: [Bug]

robonxt, 2025-08-14:
**Describe the bug**
After updating the system and other parts such as moonraker, klipper, etc, I am no longer able to use the input shaper on my XSmart

**Expected behavior**
Input shaper should be working on updated things.

**To Reproduce**
Update to latest everything, run Input Shaper from display.

**Extra Information**
I recall seeing in changelogs that moonraker, kipper, mainsail have major updates since the release of FreeDI v2.00

**Console output**
If applicable, add the debug console output here (here is a [how to](https://github.com/Phil1988/FreeDi/wiki/Bugs!))

**Base (please complete the following information):**
 - Printer model: X-Smart 2
 - X3seriesLCD firmware version: v2.00-0-g0663905e
 - Source: FreeDi image
 - Mainboard: Stock firmware

robonxt, 2025-08-15:
Huh, seems like it's working now after I loaded the default macros. Weird, it was not working yesterday. Perhaps it needed a full system restart. 

As for the thumbnail, I cannot test it due to #310 preventing me from printing.

Phil1988, 2025-08-17:
I am sure your macros.cfg was pretty faulty and also the reason for this.
Now as https://github.com/Phil1988/FreeDi/issues/310 is solved I hope this is also gone.

Please check and verify this.

closed 2025-08-19 by Phil1988

robonxt, 2025-10-03:
Just an update: 

input shaper is working now. Config was probably not working like you suggested.  I've pulled from the latest update and it's not bugging me anymore.

Thumbnails are also working at the moment, most likely a slicer issue.