Phil1988/FreeDi #310: [Bug] Default X-Smart3 config gives error for fans

robonxt, 2025-08-14:
**Describe the bug**
I'm getting the error `The value 'sidefan' is not valid for FAN` error with my X-Smart 3. This happens when using the default macro files from the repository. If I comment out the sidefan lines, I then get `The value 'fan2' is not valid for FAN`. I broke my working `macros.cfg` file when trying to fix #309, so I'm trying to restore it. 

**Expected behavior**
X-Smart 3 configs from repo should work without errors. 

**To Reproduce**
1. Use my [`printer.cfg`](https://github.com/user-attachments/files/21781572/robonxt.s.printer.cfg.txt) (it's uploaded as a txt file, just remove the `.txt`
2. Use latest X-Smart 3 macros from repo.

**Base (please complete the following information):**
 - Printer model: X-Smart 3
 - X3seriesLCD firmware version: v2.00-0-g0663905e
 - Source: FreeDi Image
 - Mainboard: Stock

**Extra Information**
I forgot how I fixed it back then when I migrated from v1.2X to v2.00. 
Is the main issue the `macros.cfg` file, or my `printer.cfg`?

robonxt, 2025-08-15:
Sure. I'm getting this error:

[image]

I am using this [`macro.cfg`](https://github.com/user-attachments/files/21803421/robonxt.s.macros.cfg.txt) (The only difference between this macro file and the one in the repo is the lines that control the sidefan have been commented out)

robonxt, 2025-08-16:
After a fresh system restart, I no longer get that sidefan error, but it has to do with another undefined fan macro.  

The reason is that the macro file in the repo had extra configuration for the other printers (chamber temp fan), when the X-Smart 3 does not have it. After removing the offending `CHAMBER_TEMP_CONTROLLER` macro and updating `SET_PRINT_STATS_INFO` macro, it's printing. Will report back after a few days of testing.

closed 2025-08-17 by Phil1988

robonxt-ai, 2025-08-17:
So far it's working fine. And yes, it looks like the only issue is the chamber fan, the side fan seems to be some leftover macro from past installations.

robonxt, 2025-08-17:
So far it's working fine. And yes, it looks like the only issue is the chamber fan, the side fan seems to be some leftover macro from past installations.