Phil1988/FreeDi #313: [Bug] The screen wants me to go through the Wizard every time the system is switched on.

okunal, 2025-08-16:
**Describe the bug**
The screen wants me to go through the Wizard every time the system is switched on.

**Expected behavior**
Wizard should only run at first install

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Screenshots or picture**
If applicable, add screenshots to help explain your problem.

**Console output**
If applicable, add the debug console output here (here is a [how to](https://github.com/Phil1988/FreeDi/wiki/Bugs!))

**Base (please complete the following information):**
 - Q1 Pro
 - LCD firmware version - whatever came with FreeDi
 - FreeDi image
 - Mainboard: Stock Qq Pro

andreykurichev, 2025-08-16:
@okunal 
In the freedi.cfg file, make it wizard: False and the wizard will not start when the system starts.

Phil1988, 2025-08-17:
@okunal 
The normal process is to finish the wizard once.
If you skip it, then it will turn back on the next boot.

If you dont want to use (abort) it, you need to change it in the freedi.cfg
`wizard: False`

Phil1988, 2025-08-19:
I wonder why it didnt change it automatically.

I am reopening this issue.

Can you please change it back to 
`wizard: True`

and capture the debug output while going through/finishing the wizard?
https://github.com/Phil1988/FreeDi/wiki/How-to-report-bugs

okunal, 2025-08-19:
Thanks Phil. Here is the result:

[MainThread] Section [freedi] not found in the printer.cfg.
[MainThread] Using freedi.cfg instead of printer.cfg.
[Thread-1 (monitor_updates)] Connecting to Moonraker at ws://127.0.0.1:7125/websocket
[MainThread] self.user_name: mks
[MainThread] self.user_group: mks
[MainThread] TFT baudrate detection...
[Thread-1 (monitor_updates)] Websocket connected
[Thread-1 (monitor_updates)] Moonraker connection opened at ws://127.0.0.1:7125/websocket
[Thread-1 (monitor_updates)] File name was reset
[Thread-1 (monitor_updates)] Updating print stats with state: standby
[MainThread] Detected baudrate: 921600
[Thread-1 (monitor_updates)] Getting metadata with state: standby, current printer state: None
[MainThread] Detected display model: TJC4827X243_011R
[Thread-1 (monitor_updates)] Printer state updated: standby
[Thread-1 (monitor_updates)] G-Code file path is: ~/printer_data/gcodes
[MainThread] Display firmware: v2.00
[MainThread] LCD firmware is OK
[MainThread] Waking up LCD...
[MainThread] Monitoring is already running.
Software version: v2.00 stable
[MainThread] Printer model: unknown
[MainThread] Printer volume: 245x258x248
[MainThread] Klipper version: v0.13.0-213-gd34d3b05-dirty
[MainThread] FreeDiLCD start
[MainThread] FreeDi sw-version sent to screen!
[Thread-6 (dynamic_lcd_update)] Wizard page not initiated, forwarding to wizard landing page
[Thread-6 (dynamic_lcd_update)] Forwarding to wz_01
[ThreadPoolExecutor-0_0] _WizardPage_01_install_start: len: 1 data[0]: ee
[ThreadPoolExecutor-0_0] ActiveScreen: = WizardPage_01

Phil1988, 2025-08-19:
This is the start, but not the whole output.

You must run it completeley while you are using the printer - it logs the whole interaction and I need that.

So it would be nice if you can do it again, leave it open for the whole time until you hit the "finish wizard"-button and then copy and paste it here 🙏