Phil1988/FreeDi #344: [Bug] WiFi Password Char Limit

my2kta, 2025-09-19:
Issue:
During the initial setup, when attempting to enter my Wifi password (16 char) the field appears to be limited to 14 chars.  When pressing any key beyond the 14th entry, the screen does not update.   Unsure if special characters are to blame, ex:  Aaaaaa.1111:Bbbb

Complete fresh install and begin Wizard.
1.  When at Wifi network screen, select SSID.
2.  Begin typing characters or string that is 16 char or longer.
3.  Observe if screen continues to update past the 14th character.

Q1 Pro.

I prefer to use the ethernet connection so this is not a blocker for me, but I noticed this behavior in both v2.0 and v2.0.1.  I have not attempted to connect to the WiFi post setup wizard.

wildy, 2025-10-07:
I just flashed FreeDI onto my XMAX 3, and wasn't able to connect at all -- turns out that the string that ends up in `/etc/NetworkManager/system-connections/*.nmconnection` gets truncated. Had to choose another network, SSH into the printer and edit the nmconnection myself.

wildy, 2025-10-07:
> [@Scrumplex](https://github.com/Scrumplex) you are right. I will add it and include it with the next release.
> 
> [@wildy](https://github.com/wildy) Is yours different from the OPs report?

Yes, I think this is a different problem. The password in the connection file gets truncated.

Phil1988, 2025-10-07:
> Yes, I think this is a different problem. The password in the connection file gets truncated.

tuncated to how many signs?

wildy, 2025-10-07:
> > Yes, I think this is a different problem. The password in the connection file gets truncated.
> 
> How many characters will it be truncated to?

I would have to replicate this again, since I got it to work successfully by editing the `.nmconnection` file directly. Expect this to be done tomorrow -- I will be near the printer.

Phil1988, 2025-11-25:
Thanks mate - I do my best :)
It would be nice if you can ask questions like this preferebly via discord (faster responses from users and me) or via a seperate thread just to keep this clean ;)

For your question, please use v2.01 (it has a full image) as basis and update from there to v2.02.
And welcome to the family :)

Phil1988, 2025-12-02:
I am closing this due to inactivity (and because I am positive that this is fixed).